shadow-1/yi-hack-v3 #1: rootfs_h20 and home_h20

add210, 2017-04-06:
I am very new to this. How do I hack my Yi Dome camera step by step. I cant figure out what the rootfs_h20 and home_h20 are.

Thank you,
Ed

add210, 2017-04-06:
Never mind. Found them.

However, I still get "This camera can only be used within China" alert.

My start_reset log is as follow:

[./dispatch][1/1/0:0:11:145]: hw_type(2)
[./dispatch][1/1/0:0:13:945]: got sn(5vcrZ1QsxYoBu2RLXE6GQxw4xRSFhzdc)[./dispatch][1/1/0:0:13:946]: got pwd()[./dispatch][1/1/0:0:13:947]: got ssid()[./dispatch][1/1/0:0:13:948]: got tnp_init_string()[./dispatch][1/1/0:0:13:949]: reset because no wifi config[./dispatch][1/1/0:0:13:951]: send_saveconfig_msg msg snd success[./dispatch][1/1/0:0:13:951]: get_config save conf
[./dispatch][1/1/0:0:14:24]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-h19, dlproto = qiniu
[./dispatch][1/1/0:0:14:24]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-h19, dlproto = qiniu
[./dispatch][1/1/0:0:14:45]: init ok, cost time(2899 ms)[./dispatch][1/1/0:0:14:58]: got pwd()()[./dispatch][1/1/0:0:14:68]: got ssid()()[./dispatch][1/1/0:0:14:69]: config save finish 1[./rmm][1/1/0:0:14:158]:  0x97
[./dispatch][1/1/0:0:14:393]: save_config[./rmm][1/1/0:0:14:796]: ai init ok!
[./rmm][1/1/0:0:14:797]: HI_MPI_AENC_AacInitHI_MPI_AENC_AacInit end
[./rmm][1/1/0:0:14:867]: rmm_init_aenc init ok!
[./rmm][1/1/0:0:14:941]: rmm_init_adec init ok!
[./rmm][1/1/0:0:14:942]: 
system call back libsns_ov9732.so_____________________
[./rmm][1/1/0:0:15:66]: sns_call_backsns_call_back
[./rmm][1/1/0:0:15:68]: HI_MPI_ISP_3A SUCCESS
[./rmm][1/1/0:0:15:234]: 
_____Gamma test for table 4__________
[./rmm][1/1/0:0:15:257]: 
---------Rock_COMM_VENC_BindVpss VeChn = 0, VpssGrp = 0, VpssChn = 0
[./rmm][1/1/0:0:15:260]: rc set success! venc(0)
[./rmm][1/1/0:0:15:276]: 
---------Rock_COMM_VENC_BindVpss VeChn = 1, VpssGrp = 0, VpssChn = 1
[./rmm][1/1/0:0:15:277]: rc set success! venc(1)
[./rmm][1/1/0:0:15:280]: HI_MPI_RGN_Create 2 success![./rmm][1/1/0:0:15:281]: rmm_vpp_create 2 success![./rmm][1/1/0:0:15:282]: rmm_init success[./rmm][1/1/0:0:15:295]: font:/home/base/iso8859-1-16x32.bin
[./rmm][1/1/0:0:15:297]: filesize=16384
[./rmm][1/1/0:0:15:316]: init font /home/base/iso8859-1-16x32.bin ptr size(4)[./rmm][1/1/0:0:15:317]: font:/home/base/iso8859-1-8x16.bin
[./rmm][1/1/0:0:15:318]: filesize=4096
[./rmm][1/1/0:0:15:336]: init font /home/base/iso8859-1-8x16.bin ptr size(4)[./rmm][1/1/0:0:15:338]: load_font success[./rmm][1/1/0:0:15:351]: msg_proc success[./rmm][1/1/0:0:15:354]: 
_____Gamma test for table 1__________
[./rmm][1/1/0:0:15:386]: pid[956][./rmm][1/1/0:0:15:387]: pid[954][./rmm][1/1/0:0:15:393]: pid[955][./dispatch][1/1/0:0:15:455]: invalid msg 0x1002
[./rmm][1/1/0:0:15:459]: got RMM_SPEAK_WAIT[./rmm][1/1/0:0:15:521]: msg snd success[./cloud][1/1/0:0:15:836]: open share mem ok
[./p2p_tnp][1/1/0:0:16:607]: p2p_set_tnp_init_status 1 send_msg ok!
[./dispatch][1/19/6:41:36:103]: DISPATCH_SET_DEFAULT_TIME 1453185696[./watch_process][1/19/6:41:36:604]: check_interval=10
[./watch_process][1/19/6:41:36:840]: process=dispatch;cmd=cd /home/app;./dispatch &;
[./watch_process][1/19/6:41:36:840]: process=cloud;cmd=cd /home/app;./cloud &;
[./watch_process][1/19/6:41:36:840]: process=rmm;cmd=reboot;
[./watch_process][1/19/6:41:36:841]: process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[./watch_process][1/19/6:41:36:841]: process=mp4record;cmd=cd /home/app;./mp4record &;
[./watch_process][1/19/6:41:36:841]: process=arp_test;cmd=cd /home/app;./arp_test &;
[./watch_process][1/19/6:41:46:641]: arp_test crashed![./dispatch][1/19/6:41:56:561]: invalid msg 0x1002
[./rmm][1/19/6:41:56:564]: got RMM_SPEAK_WAIT[./rmm][1/19/6:41:56:565]: msg snd success[./dispatch][1/19/6:42:18:669]: invalid msg 0x1002
[./rmm][1/19/6:42:18:671]: got RMM_SPEAK_WAIT[./rmm][1/19/6:42:18:672]: msg snd success[./dispatch][1/19/6:42:40:769]: invalid msg 0x1002
[./rmm][1/19/6:42:40:770]: got RMM_SPEAK_WAIT[./rmm][1/19/6:42:40:771]: msg snd success[./dispatch][1/19/6:42:42:745]: rcd may crashed
[./dispatch][1/19/6:43:2:835]: invalid msg 0x1002
[./rmm][1/19/6:43:2:837]: got RMM_SPEAK_WAIT[./rmm][1/19/6:43:2:838]: msg snd success[./dispatch][1/19/6:43:25:16]: invalid msg 0x1002
[./rmm][1/19/6:43:25:18]: got RMM_SPEAK_WAIT[./rmm][1/19/6:43:25:19]: msg snd success[./rmm][1/19/6:43:39:83]: decoded QR-Code symbol "b=itZ3kx7c3TxtFXoe&s=aG9tZTIxMA==&p=alYkIT4LAV95"
[./rmm][1/19/6:43:39:102]: trans_json info=b=itZ3kx7c3TxtFXoe&s=aG9tZTIxMA==&p=alYkIT4LAV95;key=&p=
[./rmm][1/19/6:43:39:103]: trans_json result=alYkIT4LAV95
[./rmm][1/19/6:43:39:104]: trans_json info=b=itZ3kx7c3TxtFXoe&s=aG9tZTIxMA==&p=alYkIT4LAV95;key=&s=
[./rmm][1/19/6:43:39:105]: trans_json result=aG9tZTIxMA==
[./rmm][1/19/6:43:39:106]: trans_json info=b=itZ3kx7c3TxtFXoe&s=aG9tZTIxMA==&p=alYkIT4LAV95;key=b=
[./rmm][1/19/6:43:39:107]: trans_json result=itZ3kx7c3TxtFXoe
[./rmm][1/19/6:43:39:108]: rmm got /tmp/got_wpa ssid(home210) pwd(Rongmang1) bind(itZ3kx7c3TxtFXoe)
[./dispatch][1/19/6:43:39:118]: invalid msg 0x1005
[./rmm][1/19/6:43:39:121]: got RMM_SPEAK_SCAN_OK[./rmm][1/19/6:43:39:122]: msg snd success[./dispatch][1/19/6:43:39:124]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-h19, dlproto = qiniu
[./rmm][1/19/6:43:39:125]: msg snd success[./dispatch][1/19/6:43:39:194]: msg snd success[./dispatch][1/19/6:43:39:194]: invalid msg 0x1004
[./rmm][1/19/6:43:39:198]: got RMM_SPEAK_CONNECTTING[./dispatch][1/19/6:43:39:546]: wifi disconnected, now reconnect wifi
[./dispatch][1/19/6:43:45:625]: wpa connected , now dhcp
[... cut ...]
[./dispatch][4/6/2:22:6:203]: bssid(fc:3f:7c:29:0d:94)[./dispatch][4/6/2:22:6:203]: ip(192.168.1.94)[./dispatch][4/6/2:22:6:227]: gw(192.168.1.1)[./dispatch][4/6/2:22:7:294]: gwmac(FC:3F:7C:29:D:8B)[./dispatch][4/6/2:22:7:319]: mask(255.255.255.0)[./dispatch][4/6/2:22:7:386]: mac(B0:D5:9D:7C:8A:AA)[./dispatch][4/6/2:22:7:464]: signal_quality(85)[./dispatch][4/6/2:22:7:464]: invalid msg 0x1006
[./rmm][4/6/2:22:7:467]: got RMM_SPEAK_WIFI_CONNECTTED[./dispatch][4/6/2:22:7:468]: msg snd success

I used Fing to find out what IP my camera is, but could not locate it.

batrarobin, 2017-04-06:
@add210

If you are still getting the alert  "This camera can only be used within China",  then both images haven't flashed successfully. 

Please ensure the following:
(1) you have Yi black dome camera 1080p
(2) you are using SD card of 16GB or less
(3) your SD card is formatted to FAT32
(4) the file names stored on SD root are exactly "rootfs_h20" and "home_h20"
(5) sometimes if you use Windows computer it tends to rename downloaded files to e.g. "rootfs_h20 (2)" etc. Make sure your files on SD card are named exactly "rootfs_h20" and "home_h20"
(6) Insert the SD card, power cycle the camera and ensure that you don't interrupt the process. 
(7) After booting up again, it shouldn't give you that annoying alert again. Now it's time to change the config file as shadow-1 explained in his tutorial. 

Good luck

batrarobin, 2017-04-06:
The files can be downloaded from:
https://github.com/shadow-1/yi-hack-v3/releases

shadow-1, 2017-04-06:
@batrarobin is absolutely correct with the procedure.

@add220
I have noticed you mentioned that you have a Yi Dome camera. 

At the moment, only the Yi 1080p Dome (black) camera is supported. I have also gotten it to work on the Yi 1080p Home v1 camera with help from others and will update this repository to officially support it.

To support the Yi Dome (white) camera. I will need a little help from someone with terminal access through serial port. There are small differences between the 1080p and the 720p Dome cameras. I only have the Yi 1080p Dome camera to test with.

add210, 2017-04-06:
Yes, mine is Yi Dome Camera. I read somewhere that they are pretty much the same in term of hardwares and thought this FW would work for both 720p Dome and 1080p Dome.

I am very new to this and not sure what I need to do to help you on this. If you have step by step guide, may be I can be of help.

shadow-1, 2017-04-06:
@add210 
You are right that they are very similar cameras. However they are different enough that I will need to create a separate firmware for the Yi Dome.

For me to be able to create a custom firmware, I need a full backup image of the camera. This can only be done through the terminal. With the default firmware, there is no easy way to activate telnetd so the terminal is accessible over wifi. So you have to get access to the terminal through the serial port.

If you can help. These are the initial steps that need to be followed:
1. Open up the camera.
2. Solder wires onto the serial port pads on the main circuit.
3. Using a USB to Serial converter, connect the camera to a computer.
4. Use a terminal application like PuTTY to access the cameras terminal.

Once you get there, I can help you with regards to what commands to type in to copy the files I need to create the y-hack-v3 firmware for the Yi Dome.

Keep in mind, this only needs to be done once so I can create the initial firmware.

mikeschi, 2017-04-07:
I'm a bit lost. Can anyone help? What is expected here "Note: The camera will not be fully operational through the smartphone app until a valid configuration has been entered for Proxychains-ng or Proxychains-ng is disabled."

shadow-1, 2017-04-08:
@mikeschi 
What this mean is:

If your camera is not region banned, you do not need ProxyChains-ng and it can be disabled through the System Config page on the web interface. Once disabled, the camera will work through the smartphone app.

However if your camera is region banned, you need to setup ProxyChains-ng. Once setup, the camera will work through the smartphone app.

add210, 2017-04-08:
I just ordered USB-Serial Converter online. It should arrive in a few days. I will see what I can do.

shadow-1, 2017-04-08:
@add210 
Thanks for the help!

Another alternative for others to consider is to use an Arduino as a USB to Serial converter. I had an old Arduino and used that as a USB to Serial converter. Luckily the Yi Dome cameras utilise 0-5V signal which is compatible with Arduino.

mikeschi, 2017-04-08:
Ok my YI 1080P (black) dome camera is country locked to China. Need to configure ProxyChains. May have already made too many changes to the Proxychains adding proxy addresses as well as testing: dynamic_chain or strict_chain or round_robin_chain.

How can I get back to the virgin ProxyChains file? I've already tried reflashing with your firmware: home_h20 and rootfs_h20.

Also is this the correct format for adding in proxy servers:
socks4      218.20.227.18 1080
socks4     115.29.192.194 1080
socks4	192.168.1.49	1080

Any help how to make this silly camera work in USA would be sincerely appreciated.

shadow-1, 2017-04-09:
@mikeschi 
Default configuration which comes with ProxyChains-ng is located in the source of my repository. Link below:
https://github.com/shadow-1/yi-hack-v3/blob/master/src/home/app/proxychains.conf

Yes that is the correct format for adding proxy servers. However make sure if you are adding a socks5 server to change the configuration to read socks5.

The trick is finding a working proxy server that resides in mainland china. In future releases of the firmware I will include a facility to check ProxyChains-ng configuration on the camera itself. At the moment, the easiest way to check your configuration is to change the proxyserver on your computer and browse to a website like https://whatismyipaddress.com/ to check whether the proxy server works and whether it is located in China.

I'm not sure where you got those proxy servers. However they do not appear to be working when I tested them.

wanwe, 2017-04-09:
@shadow-1 
Nice work on Yi Dome 1080p!
I tried to set proxychina with "socks4 	59.51.73.213 1080", tested this proxy works fine, but the setup failed saying "connection error".
I telnet in and found below in "/tmp/log.txt"
[./cloud][4/8/9:50:37:363]: yi_sync_time fail!
[./cloud][4/8/9:50:40:373]: cmd=/home/app/cloudAPI -c 136 -url http://api.xiaoyi.com/v2/ipc/sync_time
[./cloud][4/8/9:50:41:464]: req_info=http://api.xiaoyi.com/v2/ipc/sync_time?hmac=BIke7GBrJXU6Qe41glDqGUkYNp8%3D&seq=9

when I run "/home/app/cloudAPI -c 136 -url http://api.xiaoyi.com/v2/ipc/sync_time" manually, I got below error 
~ # /home/app/cloudAPI -c 136 -url http://api.xiaoyi.com/v2/ipc/sync_time
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
proxychains can't load process....: No such file or directory

any idea? is there anything missing for proxychain?

shadow-1, 2017-04-09:
@wanwe 
I have tested the proxy server. It appears that this particular proxy server does not support proxy dns requests. So make sure that in your configuration you are _not_ using the `dns_proxy` setting and _not_ using the `remote_dns_subnet` setting.

It also appears that it is not the most reliable proxy server, I had to retry a few times before it worked fine. For the purposes of defeating the region ban, even if the proxy server is a bit slow or a bit unreliable it will work (may take a little longer to initialise when you first start up the camera).

wanwe, 2017-04-09:
@shadow-1 
Thank you so much..I just grabbed a free proxy server from google, now searching for a more reliable one.

just curious, is it possible to integrate with hacks from this page? then no proxy is required
https://diy.2pmc.net/solved-xiaomi-xiao-yi-ant-home-camera-can-used-china/

I tried to sed check_did part to sth else, but seems not working, log says "./cloud crashes", wonder whether it has some filechecksum in latest version of firmware

shadow-1, 2017-04-09:
@wanwe 
Unfortunately that hack no longer works. It will stop the annoying ban message from being played. However the camera will not be able to be used with the smartphone app.

Xiaomi have changed something at their servers around December 2016 which caused most people who used that style of hack to no longer be able to use their region locked cameras through the smartphone app.

wanwe, 2017-04-09:
@shadow-1 
one more question, just found that the firmware seems not working with my another camera, which is on firmware version 1.9.2.0D_201612191901home_h20m, any idea?

batrarobin, 2017-04-10:
@wanwe
I am using the shadow-1's hack on version D firmware without any problems. Just ensure both files have successfully flashed. Check and ensure the 7 items that I mentioned in my comment above for a successful flashing.

shadow-1, 2017-04-10:
@wanwe 
As you are using the firmware on one camera, it is a little unusual that your second camera did not flash. A few generic tips to look out for:
1. Make sure that the filename didn't get changed. The firmware filenames for the Yi 1080p Dome camera must be `home_h20` and `rootfs_h20`.
2. Try flashing the camera with the same microSD card as with the camera you successfully flashed. The microSD card must be formatted in FAT32 for it to work. This is largely an issue with high capacity cards (>32gb) which default to being formatted in exFAT.
3. Make sure that you flash both rootfs_h20 and home_h20. Flashing one without the other will not allow the features of this firmware to work.

To help further, I will probably need to know more details about the issue. For instance, when you started the camera with the firmware files on it, did the yellow light flash for roughly 30 seconds (this is the indicator that the firmware is being flashed)?

szefkamil, 2017-04-11:
Hello Guys,

I have Yi dome 1080p black edition camera and after flash firmware with this files i'm still can't set-up wifi connection always prompt me fail to connect please reset your camera maybe i made some mistake below is my procedure:

1. Format microSDcard as fat32
2. Copy two files home_h20 and rootfs_h20
3. Plug power to camera 
4. pull sd card inside and wait for flashing procedure
5. try to connect using yi home app
6. scan QR code everything should be okay because camera spoke in chinese "Wait for wifi connection"
7. Wait 2 minutes and received information is FaileD!!!!

I'm really confused about this, please help me.

Regards
Kamil

https://github.com/shadow-1/yi-hack-v3/releases

shadow-1, 2017-04-11:
@szefkamil 
You are almost there. Just a couple of things:
I would insert the microSD card into the camera before powering up the camera. The internal check for a firmware update occurs right towards the start of the boot procedure. You will know that it is flashing as it will take roughly 30 seconds longer than usual for the camera to start moving. In addition the yellow light flashes.

Since you have not setup your cameras wifi settings yet. You will need to complete the initial setup using the smartphone app. However as you mentioned, it will fail to pair. This is because ProxyChains-ng has not been setup and is blocking communication between the camera and the Xiaomi server. However by going through the procedure, wifi on the camera have been setup. This can be confirmed as the blue light is turned on and solid if it has been setup and is connected.

Once wifi has been setup on the camera, you need to find the IP address of the camera. The easiest place to find it is on your routers setup page. However there are network scanning apps available as well. Enter the web interface of the camera by entering the IP address of the camera into your web browser. You will then see the custom firmware configuration page and be able to setup ProxyChains-ng as shown below. Once setup, your camera will pair using the smartphone app and be accessible without issues.
[image]

Refer to the "Starting the Camera" and the "Configuring the Camera" sections of the Readme.
https://github.com/shadow-1/yi-hack-v3/blob/master/README.md

junsheng89, 2017-04-11:
@xmflsct hi zhiyuan, thanks for all the help so far. 

"This can be confirmed as the blue light is turned on and solid if it has been setup and is connected." --> i got this.

i am also logged on to my camera's IP address and have pasted in the following code from @shadow-1:

# proxychains.conf  VER 4.x
#
#        HTTP, SOCKS4a, SOCKS5 tunneling proxifier with DNS.

# The option below identifies how the ProxyList is treated.
# only one option should be uncommented at time,
# otherwise the last appearing option will be accepted
#
#dynamic_chain
#
# Dynamic - Each connection will be done via chained proxies
# all proxies chained in the order as they appear in the list
# at least one proxy must be online to play in chain
# (dead proxies are skipped)
# otherwise EINTR is returned to the app
#
strict_chain
#
# Strict - Each connection will be done via chained proxies
# all proxies chained in the order as they appear in the list
# all proxies must be online to play in chain
# otherwise EINTR is returned to the app
#
#round_robin_chain
#
# Round Robin - Each connection will be done via chained proxies
# of chain_len length
# all proxies chained in the order as they appear in the list
# at least one proxy must be online to play in chain
# (dead proxies are skipped).
# the start of the current proxy chain is the proxy after the last
# proxy in the previously invoked proxy chain.
# if the end of the proxy chain is reached while looking for proxies
# start at the beginning again.
# otherwise EINTR is returned to the app
# These semantics are not guaranteed in a multithreaded environment.
#
#random_chain
#
# Random - Each connection will be done via random proxy
# (or proxy chain, see  chain_len) from the list.
# this option is good to test your IDS :)

# Make sense only if random_chain or round_robin_chain
#chain_len = 2

# Quiet mode (no output from library)
#quiet_mode

# Proxy DNS requests - no leak for DNS data
proxy_dns 

# set the class A subnet number to use for the internal remote DNS mapping
# we use the reserved 224.x.x.x range by default,
# if the proxified app does a DNS request, we will return an IP from that range.
# on further accesses to this ip we will send the saved DNS name to the proxy.
# in case some control-freak app checks the returned ip, and denies to 
# connect, you can use another subnet, e.g. 10.x.x.x or 127.x.x.x.
# of course you should make sure that the proxified app does not need
# *real* access to this subnet. 
# i.e. dont use the same subnet then in the localnet section
#remote_dns_subnet 127 
#remote_dns_subnet 10
remote_dns_subnet 224

# Some timeouts in milliseconds
tcp_read_time_out 15000
tcp_connect_time_out 8000

### Examples for localnet exclusion
## localnet ranges will *not* use a proxy to connect.
## Exclude connections to 192.168.1.0/24 with port 80
# localnet 192.168.1.0:80/255.255.255.0

## Exclude connections to 192.168.100.0/24
# localnet 192.168.100.0/255.255.255.0

## Exclude connections to ANYwhere with port 80
# localnet 0.0.0.0:80/0.0.0.0

## RFC5735 Loopback address range
## if you enable this, you have to make sure remote_dns_subnet is not 127
## you'll need to enable it if you want to use an application that 
## connects to localhost.
# localnet 127.0.0.0/255.0.0.0

## RFC1918 Private Address Ranges
# localnet 10.0.0.0/255.0.0.0
# localnet 172.16.0.0/255.240.0.0
# localnet 192.168.0.0/255.255.0.0

# ProxyList format
#       type  ip  port [user pass]
#       (values separated by 'tab' or 'blank')
#
#       only numeric ipv4 addresses are valid
#
#
#        Examples:
#
#            	socks5	192.168.67.78	1080	lamer	secret
#		http	192.168.89.3	8080	justu	hidden
#	 	socks4	192.168.1.49	1080
#	        http	192.168.39.93	8080	
#		
#
#       proxy types: http, socks4, socks5
#        ( auth types supported: "basic"-http  "user/pass"-socks )
#
[ProxyList]
# add proxy here ...
# meanwile
# defaults set to "tor"
socks4 	127.0.0.1 9050

am i suppose to reboot the camera? what additional steps am i missing? adding more proxy servers?

xmflsct, 2017-04-11:
Actually a solid blue LED **does not** mean that you can pair with it. It could also mean that the proxy server does not work, thus the camera also connect receive a **ban** command from the server.

After enter your proxy and saved the settings, press the reset button to reset the camera, and try to pair it with your phone again. If you have paired before and in your app you can still see your camera, you just need a reboot of the camera.

shadow-1, 2017-04-11:
@junsheng89 
As mentioned in the following post; https://github.com/shadow-1/yi-hack-v3/issues/4#issuecomment-293267871

You need to add proxy servers which reside in Mainland China to the end of the configuration. I deliberately didn't add any with the firmware as they do change and if one particular server gets too popular, it will probably get shutdown.

As mentioned in the other thread, the best way to ensure that everything will work is to change the proxy server on your web browser to the one you will add to ProxyChains-ng and browse to a website like; https://whatismyipaddress.com/ . If you can successfully browse to the website and the Country comes up as China on the webpage, then you know that this particular proxy server will work with the hack.

junsheng89, 2017-04-11:
@shadow-1 @xmflsct can i put multiple addresses? 

e.g. 
socks4 218.20.227.18 1080
socks4 115.29.192.194 1080
socks4	192.168.1.49	1080

in this example 3 addresses? or are you allow only 1 at each time?

following which, i should follow these steps:
After enter your proxy and saved the settings, press the reset button (physical or the reboot button(ipaddress of your cam) to reset the camera, and try to pair it with your phone again. 

If you have paired before and in your app you can still see your camera, you just need a reboot of the camera.

szefkamil, 2017-04-11:
Where can i find correct port?

maysamsh, 2017-04-11:
Using the custom images and setting above proxy address', still I cannot pair the cam with the app, but I have telnet access.

szefkamil, 2017-04-11:
Which adressses? sory but i can't see

xmflsct, 2017-04-11:
Google is your best friend, to find what you need. It may take a bit of time trying several proxies before you find one that is working and with good speed.

I can confirm this project fully working, as I have tried it several times.

maysamsh, 2017-04-11:
@szefkamil Try these, they are working:

```
socks5  27.152.181.217  80
socks5  125.67.236.195   8080
socks4  115.29.192.194  1080
```

maysamsh, 2017-04-11:
@xmflsct How can I find out what's wrong with my case?

xmflsct, 2017-04-11:
@maysamsh Please refer to my edited comment above. Do you install the right app?

wanwe, 2017-04-11:
@shadow-1 
yah, it's very weird...one of my camera works, another one does not...I tried to flash offical firmware again and still not working, 
Used exact same sd card, not working; formatted and download files again, same result.

when I insert the sd card and power on, it was solid orange led, no flash, for 10+ seconds, then it starts saying "waiting for connection". this is different from my the other camera....no idea why :-(

maysamsh, 2017-04-11:
@xmflsct If you mean the on the cam, yes. On the iPhone I'm using Mi Home app.

junsheng89, 2017-04-11:
@maysamsh @shadow-1 

i am using these addresses:

wanwe, 2017-04-11:
what's the firmware version you guys are using ?anyone is using "1.9.2.0D_201612191901home_h20m"?

xmflsct, 2017-04-11:
@wanwe "Waiting for connection" means you have to pair with it with Mi app. No?

@maysamsh No, I mean on your phone. I haven't tried with Mi Home app. But I am using Yi Smart app from **Chinese** App Store.

wanwe, 2017-04-11:
@xmflsct 
yes, if firmware works, it should take longer before we hear "waiting for connection". I tried to pair, no surprise it connects to wifi, then get region ban message...I couldn't access the http page as well...so confirm the hack is not working on my camera

szefkamil, 2017-04-11:
Should i put all three adresses?

xmflsct, 2017-04-11:
@wanwe 
So when you flash the firmware, the orange or yellow LED should blink. For flashing rootfs, it blinks only few times. For flashing home, it blinks for quite a while. This is a way to confirm flashing actually works without terminal output.

Try to boot the cam with an SD card, and see if the system dumps some log in there. If so, can you paste the log somewhere for debugging?

szefkamil, 2017-04-11:
Guys which app should i use? from google play store or from chinese market?

another question i add these adresses into proxychain_ng

[ProxyList]
# add proxy here ...
# meanwile
# defaults set to "tor"
socks5 27.152.181.217 80
socks5 125.67.236.195 8080

and isn't working.

shadow-1, 2017-04-11:
@junsheng89 Yes multiple servers can be entered into the configuration.

@szefkamil When you find a socks server to enter, there are always three pieces of information you need. They are:
* What kind of proxy server; Socks4 or Socks5
* The IP Address of the proxy server
* The port address of the proxy server

@maysamsh I tested your proxy servers and they appear to be working. @xmflsct is absolutely correct. You must use the smartphone app from the Chinese App Store. The one on Google App Store will not work on a camera locked to Mainland China. I am using the Yi Home app from the Chinese App Store.

@wanwe If you are getting the ban message, the custom firmware has definitely not been flashed onto the camera. With the custom firmware, by default, communication between the camera and the Xiaomi server is blocked, thus the ban message will not play. Not being able to access the http page is also another indicator that the firmware did not flash on your camera.

Another thing you can try is using my initial version of the custom firmware; version 0.1.0 and then upgrade to 0.1.1. I'm not exactly sure why it's not working with your second camera.

junsheng89, 2017-04-11:
yup. i have tested and it works with multiple addresses.

YI home app "works" for me but mi jia (mi home) app does not. 

Here is what puzzling. 

after the sync, the camera said in mandarin that the sync was successful BUT, on the YI Home app, under the page, RETRIEVING PAIRING STATUS > it states, Fail to acquire Please reset the camera > option below are troubleshooting & reset the camera.

maysamsh, 2017-04-11:
For some weird reasons my Chinese Apple ID is disabled an I cannot download the YI Smart app from the Chinese App Store and neither of [Chinese] YI Home nor [English] YI Smart apps work. Sad!

shadow-1, 2017-04-11:
@junsheng89 
That is quite puzzling. I have never had this kind of issue. I can only think that perhaps the proxy server is a bit unreliable? I guess you can only try again.

From my experience, even if a proxy server is slow, it will work fine but takes a bit longer to pair and a bit longer to initialise when the camera starts up.

wanwe, 2017-04-11:
@xmflsct tried to boot camera with a blank sd, no logs dumped

@shadow-1 you are amazing ! 0.1.0 can flash successfully ! I'm trying some socks config now and see if all good, before I try to update to 0.1.1..
anything logs etc you would like to collect now?

junsheng89, 2017-04-11:
CORRECTION: 

Even though the situation is like the following:
the camera said in mandarin that the sync was successful BUT, on the YI Home app, under the page, RETRIEVING PAIRING STATUS > it states, Fail to acquire Please reset the camera > option below are troubleshooting & reset the camera.

i found that it is actually still works. i just quit out of the app, reopen the app again and its finally working after 4 hours of configuration... 

Anyway, A BIG THANK YOU TO @shadow-1, @xmflsct @maysamsh, this is my 1st day on github and i am a converter. :D

maysamsh, 2017-04-11:
@junsheng89 Are you using Chinese cam with English app?

shadow-1, 2017-04-11:
@wanwe 
The camera does a very crude check to make sure that the firmware file on the SD card doesn't match what is currently on the camera. If the camera thinks the firmware file matches what is currently on the system, it will skip the upgrade.

Perhaps the initial upgrade got interrupted somehow and on subsequent boots the camera thought the firmware file matches what is currently on the system and simply skipped the upgrade?

However an easy fix for this situation is to upgrade to a different version before upgrading back to the latest.

wanwe, 2017-04-11:
@junsheng89
I think now I have same issue as you :(, YI app stuck at retrieve pairing status page...
Are you using iPhone or android? For iphone, I killed the app when it stuck, open the app, the camera is not listed in the app...

szefkamil, 2017-04-11:
I use chinese app and chinese cam proxy servers :
socks5  27.152.181.217  80
socks5  125.67.236.195   8080
socks4  115.29.192.194  1080

Camera isn't working still, I'm give up:(

maysamsh, 2017-04-11:
Log.txt at `/tmp`:
`
[./dispatch][1/1/0:0:11:631]: hw_type(2)
[./dispatch][1/1/0:0:14:466]: got sn(hNUuY4ulVv2iXCrLIC7rOnZnC9QdnfYI)[./dispatch][1/1/0:0:14:478]: got pwd(maysamnet2)[./dispatch][1/1/0:0:14:479]: got ssid(MAYSAM-PC_Network)[./dispatch][1/1/0:0:14:480]: got tnp_init_string(MOFEJKLGIEELKLHOOMEGPAEPGBDKFEHJGFFBLHJOJCBOCIABOBINBFMEOKDCABLEPELMCLOCNICCBPCHALAADOPCNC)[./dispatch][1/1/0:0:14:616]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-y20, dlproto = micn
[./dispatch][1/1/0:0:14:617]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-y20, dlproto = micn
[./dispatch][1/1/0:0:14:640]: init ok, cost time(3009 ms)[./rmm][1/1/0:0:14:693]: f22 or f02
[./rmm][1/1/0:0:14:954]:  0xf
[./rmm][1/1/0:0:14:954]:  0x22
[./rmm][1/1/0:0:15:78]: ai init ok!
[./rmm][1/1/0:0:15:79]: HI_MPI_AENC_AacInitHI_MPI_AENC_AacInit end
[./rmm][1/1/0:0:15:149]: rmm_init_aenc init ok!
[./rmm][1/1/0:0:15:220]: rmm_init_adec init ok!
[./rmm][1/1/0:0:15:222]: 
system call back libsns_f22.so_____________________
[./rmm][1/1/0:0:15:293]: sns_call_backsns_call_back
[./rmm][1/1/0:0:15:295]: HI_MPI_ISP_3A SUCCESS
[./rmm][1/1/0:0:15:911]: isp_set_ae: HI_MPI_ISP_SetExposureAttr  target_lum =60
[./rmm][1/1/0:0:15:913]: 
_____Gamma test for table 4__________
[./rmm][1/1/0:0:15:914]: modify drc (150), time:15-936282[./rmm][1/1/0:0:15:948]: 
---------Rock_COMM_VENC_BindVpss VeChn = 0, VpssGrp = 0, VpssChn = 0
[./rmm][1/1/0:0:15:950]: rc set success! venc(0)
[./rmm][1/1/0:0:15:967]: 
---------Rock_COMM_VENC_BindVpss VeChn = 1, VpssGrp = 0, VpssChn = 1
[./rmm][1/1/0:0:15:969]: rc set success! venc(1)
[./rmm][1/1/0:0:16:1]: HI_MPI_RGN_Create 3 success![./rmm][1/1/0:0:16:14]: rmm_vpp_create 3 success![./rmm][1/1/0:0:16:15]: rmm_init success[./rmm][1/1/0:0:16:17]: font:/home/base/iso8859-1-24x48.bin
[./rmm][1/1/0:0:16:33]: filesize=36864
[./rmm][1/1/0:0:16:42]: init font /home/base/iso8859-1-24x48.bin ptr size(4)[./rmm][1/1/0:0:16:55]: font:/home/base/iso8859-1-8x16.bin
[./rmm][1/1/0:0:16:57]: filesize=4096
[./rmm][1/1/0:0:16:60]: init font /home/base/iso8859-1-8x16.bin ptr size(4)[./rmm][1/1/0:0:16:62]: load_font success[./rmm][1/1/0:0:16:78]: msg_proc success[./rmm][1/1/0:0:16:82]: 
_____Gamma test for table 1__________
[./rmm][1/1/0:0:16:107]: pid[955][./rmm][1/1/0:0:16:108]: pid[956][./rmm][1/1/0:0:16:112]: pid[957][./dispatch][1/1/0:0:16:125]: invalid msg 0x1002
[./rmm][1/1/0:0:16:130]: got RMM_SPEAK_WAIT[./rmm][1/1/0:0:16:183]: msg snd success[./cloud][1/1/0:0:16:680]: open share mem ok
[./p2p_tnp][1/1/0:0:17:668]: p2p_set_tnp_init_status 1 send_msg ok!
[./dispatch][4/2/10:41:26:154]: DISPATCH_SET_DEFAULT_TIME 1491129686[./watch_process][4/2/10:41:26:155]: check_interval=10
[./watch_process][4/2/10:41:26:587]: process=dispatch;cmd=cd /home/app;./dispatch &;
[./watch_process][4/2/10:41:26:588]: process=cloud;cmd=cd /home/app;./cloud &;
[./watch_process][4/2/10:41:26:589]: process=rmm;cmd=reboot;
[./watch_process][4/2/10:41:26:589]: process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[./watch_process][4/2/10:41:26:590]: process=mp4record;cmd=cd /home/app;./mp4record &;
[./watch_process][4/2/10:41:26:590]: process=arp_test;cmd=cd /home/app;./arp_test &;
[./watch_process][4/2/10:41:26:591]: process=oss;cmd=cd /home/app;./oss &;
[./watch_process][4/2/10:41:36:156]: arp_test crashed![./dispatch][4/2/10:41:46:313]: invalid msg 0x1002
[./rmm][4/2/10:41:46:316]: got RMM_SPEAK_WAIT[./rmm][4/2/10:41:46:318]: msg snd success[./dispatch][4/2/10:42:8:482]: invalid msg 0x1002
[./rmm][4/2/10:42:8:484]: got RMM_SPEAK_WAIT[./rmm][4/2/10:42:8:485]: msg snd success[./rmm][4/2/10:42:14:576]: decoded QR-Code symbol "b=USaS6cpd1wWbqp9C&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw=="
[./rmm][4/2/10:42:14:578]: trans_json info=b=USaS6cpd1wWbqp9C&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw==;key=&p=
[./rmm][4/2/10:42:14:579]: trans_json result=VVgzNTIHAV08Zw==
[./rmm][4/2/10:42:14:580]: trans_json info=b=USaS6cpd1wWbqp9C&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw==;key=&s=
[./rmm][4/2/10:42:14:582]: trans_json result=TUFZU0FNLVBDX05ldHdvcms=
[./rmm][4/2/10:42:14:584]: trans_json info=b=USaS6cpd1wWbqp9C&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw==;key=b=
[./rmm][4/2/10:42:14:596]: trans_json result=USaS6cpd1wWbqp9C
[./rmm][4/2/10:42:14:597]: rmm got /tmp/got_wpa ssid(MAYSAM-PC_Network) pwd(maysamnet2) bind(USaS6cpd1wWbqp9C)
[./dispatch][4/2/10:42:14:598]: invalid msg 0x1005
[./rmm][4/2/10:42:14:598]: got RMM_SPEAK_SCAN_OK[./rmm][4/2/10:42:14:599]: msg snd success[./dispatch][4/2/10:42:14:601]: in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, sname = familymonitor-y20, dlproto = mius
[./rmm][4/2/10:42:14:603]: msg snd success[./dispatch][4/2/10:42:14:650]: msg snd success[./dispatch][4/2/10:42:14:650]: invalid msg 0x1004
[./rmm][4/2/10:42:14:657]: got RMM_SPEAK_CONNECTTING[./dispatch][4/2/10:42:15:110]: wifi disconnected, now reconnect wifi
[./watch_process][4/2/10:42:16:243]: cloud crashed![./watch_process][4/2/10:42:16:244]: p2p_tnp crashed![./dispatch][4/2/10:42:21:9]: wpa connected , now dhcp
[./dispatch][4/2/10:42:23:77]: wpa connected , now dhcp finish
[./dispatch][4/2/10:42:24:129]: bssid(a0:ab:1b:26:1e:94)[./dispatch][4/2/10:42:24:131]: ip(192.168.1.3)[./dispatch][4/2/10:42:24:159]: gw(192.168.1.1)[./dispatch][4/2/10:42:25:283]: gwmac(A0:AB:1B:26:1E:94)[./dispatch][4/2/10:42:25:314]: mask(255.255.255.0)[./dispatch][4/2/10:42:25:404]: mac(B0: [... cut ...]
[./cloud][4/11/15:34:46:99]: msg snd success[./dispatch][4/11/15:34:46:206]: DISPATCH_SET_TIME 1491924886 /home/base/tools/rtctool -s time 2017/4/11/15/34/46[./cloud][4/11/15:34:46:217]: mi_sync_time ok!
[./cloud][4/11/15:34:46:217]: sync_time ok!
[./cloud][4/11/15:34:46:218]: yi_proc ok[./cloud][4/11/15:34:46:219]: cmd=/home/app/cloudAPI -c 311 -url https://api.us.xiaoyi.com/v4/ipc/check_did -uid TNPCHND-099303-CYYGG -keySec 8Fo9zRDvNtQYfZj8
[./dispatch][4/11/15:34:46:225]: set DISPATCH_SET_PANORAMA_CAPTURE_STATE
[./cloud][4/11/15:34:46:226]: 
got a motion event_timeout(0) systick(2676) p2p_viewing_cnt(0)

`

xmflsct, 2017-04-11:
@junsheng89 @wanwe 
As far as I know, the camera's "connection successful" message does not mean that it is paired. It only means that it connects to WiFi successfully. Only the app will confirm pairing status.

Thus for me, I had tried several (seems to be) working proxies, and the camera has a solid blue LED, which the app times out during pairing process. Once I found one working proxy, the app will show a congratulations page that pairing is successful. This is the iOS case. Android one might be different.

szefkamil, 2017-04-11:
> [./dispatch][4/2/10:42:24:129]: bssid(a0🆎1b:26:1e:94)[./dispatch][4/2/10:42:24:131]:
> ipc/check_did -uid TNPCHND-099303-CYYGG -keySec 8Fo9zRDvNtQYfZj8
> com/v4/ipc/reset?hmac=BRJQ9YERXzNFmnFTXVB8KDS63CA%3D&seq=9&uid=
> f582e1416961d9b5cc5d3062ed557b6e&timestamp=1491924683
> xiaoyi.com/v5/alert/gen_presigned_url?hmac=VtA7gtKONwlw7jeCvJgkjdLnqy0%
> 3D&seq=9&uid=f582e1416961d9b5cc5d3062ed557b6e&type=0&sub_type=1&suffix=
> jpg,mp4&time=1491924660&timestamp=1491924686
> xiaoyi.com/v5/alert/gen_presigned_url?hmac=IS8xnB%
> 2FAuxqKYRkIo9B9jBUB4Rk%3D&seq=9&uid=f582e1416961d9b5cc5d3062ed557b
> 6e&type=0&sub_type=1&suffix=jpg,mp4&time=1491924660&timestamp=1491924707
> com/v4/ipc/reset?hmac=dj126MgrHCyAsFHG0gM1tWTIH5o%3D&seq=9&uid=
> f582e1416961d9b5cc5d3062ed557b6e&timestamp=1491924717
> xiaoyi.com/v5/alert/gen_presigned_url?hmac=%2BNQ9n5hjsbPFoYubC%
> 2BSBmi4RZp8%3D&seq=9&uid=f582e1416961d9b5cc5d3062ed557b
> 6e&type=0&sub_type=1&suffix=jpg,mp4&time=1491924660&timestamp=1491924727
> com/v4/ipc/reset?hmac=aQQSIBPHVk4z76mB4I3DezLUexA%3D&seq=9&uid=
> f582e1416961d9b5cc5d3062ed557b6e&timestamp=1491924748
> xiaoyi.com/v4/ipc/reset?hmac=BpcImyfTa9ZVOraUj7AQyo53Szw%3D&seq=9&uid=
> f582e1416961d9b5cc5d3062ed557b6e&timestamp=1491924778
> xiaoyi.com/v4/ipc/reset?hmac=YznsErYu3t2WGEBlv%2FzHQE2yMBo%3D&seq=9&uid=
> f582e1416961d9b5cc5d3062ed557b6e&timestamp=1491924811
> xiaoyi.com/v4/ipc/reset?hmac=wPiZG2OsJihmHvaUVkl%2FmF%2BM9NU%3D&seq=9&uid=
> f582e1416961d9b5cc5d3062ed557b6e&timestamp=1491924842

Maybe i made some mistake , please sent me prtscr od you proxychain_ng

szefkamil, 2017-04-11:
Sorry but i use my phone and directory change some Worda

szefkamil, 2017-04-11:
[image]
my prt screen of network settings for Yi1080p Dome
[image]
screenshot of my app yi home
[image]
My camera

Please help me!!!!

szefkamil, 2017-04-11:
[image]

When i try to connect i received this information

maysamsh, 2017-04-11:
@szefkamil 
Only I modified the end of it as follow:
```
socks5  27.152.181.217  80
socks5  125.67.236.195   8080
socks4  115.29.192.194  1080
socks4  127.0.0.1  9050
```

szefkamil, 2017-04-11:
@maysamsh

I'm trying already

szefkamil, 2017-04-11:
@maysamsh 

Not working:(

what should i do?

[52 more replies]

closed 2017-04-15 by shadow-1