shadow-1/yi-hack-v3 #2: can't pair with smarthphone

Orlandus, 2017-04-08:
Hey there.
Thanks for your work on this cameras.
Can you give me advice? I can't pair with yi home app. It errors on timeout.
But webinterface and ftp are works
My camera is yi home 1080p

shadow-1, 2017-04-08:
@Orlandus 
Is your camera region banned?

If it is not region banned, you just need to disable ProxyChains-ng. To do this, go to System Config on the web interface, click No against ProxyChains-ng and click Save.

If your camera is region banned, you will need to enter a valid configuration for ProxyChains-ng. The main trick is that you need to find a working SOCKS proxy server that resides in Mainland China.

I deliberately did not provide proxy servers built into the firmware as they tend to change often and potentially if one gets too popular, it will get shut down.

Once ProxyChains-ng is configured with a valid server within Mainland China, the camera will no longer be region banned and will be able to be paired through the smartphone app.

shadow-1, 2017-04-08:
@Orlandus Forgot to mention one more requirement, the proxy server you choose must support HTTPS. Many proxy servers do not.

batrarobin, 2017-04-08:
Pick one from here:
http://spys.ru/free-proxy-list/CN/

Orlandus, 2017-04-08:
@shadow-1  Uh, ok. I used HTTP proxy and this was not works. 
Thank you

closed 2017-04-08 by Orlandus

pasagame, 2017-04-14:
i set HTTPS on ProxyChains-ng and enable it, but i cannot pair camera with app. my camera say "wifi password error" (china language).I tried many times and was sure the password was correct. help me to to solved this issue. thank you

https://www.mx7.com/view2/zNsG8cVcovqOLrCG