shadow-1/yi-hack-v3 #3: How can I apply it to the YI Home 1080p?

maysamsh, 2017-04-10:
How can I apply it to my cam? I'm using the latest hack which provides RTSP stream but I like to use the official app and bypass the region ban.

shadow-1, 2017-04-10:
@maysamsh 
Unfortunately this custom firmware does not support RTSP at this stage.
At the moment it only supports the use of the official smartphone app and defeats the region ban.

maysamsh, 2017-04-11:
@shadow-1 Do I need custom rootfs and home image files? I cannot find them here.

shadow-1, 2017-04-11:
@maysamsh 
To use this firmware you need custom rootfs and home. They are available here:
https://github.com/shadow-1/yi-hack-v3/releases

batrarobin, 2017-04-11:
@maysamsh
Do a bit of a reading before asking questions:
https://github.com/shadow-1/yi-hack-v3/blob/master/README.md

Here's where you would find the relevant files:
 https://github.com/shadow-1/yi-hack-v3/releases

Ensure that you have the correct filenames on your SD card and use a 16GB card or less, formatted with FAT32.

maysamsh, 2017-04-11:
@batrarobin Thanks! I skipped the first few paragraphs.

closed 2017-04-13 by shadow-1