shadow-1/yi-hack-v3 #4: Regional ban hack guide

hanson06, 2017-04-11:
Can provide step by step guide to configure proxychains-ng...no exprince on that

junsheng89, 2017-04-11:
same here. @hanson06. can @shadow-1 or @xmflsct provide some guidance?

i was able to successfully flash rootfs partition & home partition, however, i am stuck at the configuration part. what changes are to be made on the ProxyChainNg configuration?

xmflsct, 2017-04-11:
Simple explanation:

You only need to modify the last line of the configuration, namely, the socks proxy. The format is:
`socks4   PROXY_IP   PROXY_PORT` or `socks5   PROXY_IP   PROXY_PORT`.
You should try these proxies on your computer to make sure it is working. 
[https://whatismyipaddress.com/](https://whatismyipaddress.com/) Do try this https website to be sure that the proxy is based in mainland China, excluding Hong Kong and Macau. This webpage also verifies that the proxy supports SSL which is essential for the trick to work.

djartemon, 2017-04-18:
What should I do if rootfs_v201 was not written to the SD card? How to put it if home_v201 was installed without it?

shadow-1, 2017-04-25:
@hanson06 @junsheng89 
I have finally gotten around to writing a more detailed guide on how to use this hack.
Let me know if you have any issues.

closed 2017-05-02 by shadow-1