shadow-1/yi-hack-v3 #5: Can use it in Yi Dome Camera?

moral3jo, 2017-04-12:
Non the 1080p dome version.
Can I help to make this firmware compatible with Yi Dome Camera?

shadow-1, 2017-04-12:
@moral3jo 
Another community member asked the same question. Unfortunately their thread got hijacked with unrelated queries. I can make this firmware compatible with Yi Dome (white/non 1080p), however as I don't have this particular camera. I will need some help from someone who has this camera and has terminal access through serial port. There are differences between the Yi Dome 1080p and the regular Yi Dome.

If you have terminal access through serial port or are willing to do a little bit of soldering to get access to the serial port of the camera. Then you can help me support this camera. I have copy and pasted the initial steps that need to be followed from the other thread.

---
For me to be able to create a custom firmware, I need a full backup image of the camera. This can only be done through the terminal. With the default firmware, there is no easy way to activate telnetd so the terminal is accessible over wifi. So you have to get access to the terminal through the serial port.

If you can help. These are the initial steps that need to be followed:

1. Open up the camera.
2. Solder wires onto the serial port pads on the main circuit.
3. Using a USB to Serial converter, connect the camera to a computer.
4. Use a terminal application like PuTTY to access the cameras terminal.

Once you get there, I can help you with regards to what commands to type in to copy the files I need to create the y-hack-v3 firmware for the Yi Dome.

Keep in mind, this only needs to be done once so I can create the initial firmware.

---
@add210 
Did you end up getting your USB to serial converter to help out with support for the Yi Dome (white)?

insantr, 2017-04-14:
Hey. I was able to connect through the serial port to the camera. 
In console i see this:

`[CPLD_PERIPH] led_blue_pwm_value = 20`
`[CPLD_PERIPH] led_blue_pwm_value = 20`
`[CPLD_PERIPH] led_blue_pwm_value = 10`
`000223.195 long time no beacons`
`000223.195 wlc_phy_resetcca_lcn40phy`
`cmd = 00000000`
`[CPLD_PERIPH] led_blue_pwm_value = 20`
`[CPLD_PERIPH] led_blue_pwm_value = 11`
`p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)`
`[CPLD_PERIPH] led_blue_pwm_value = 20`
`000228.195 long time no beacons`
`000228.195 wlc_phy_resetcca_lcn40phy`
`cmd = 00000000`
`[CPLD_PERIPH] led_blue_pwm_value = 20`

shadow-1, 2017-04-15:
@insantr 
Fantastic news! Now we need to take a backup of all the partitions on the camera. Steps are outlined below:
1. Startup the camera with a microSD card inserted.
2. Once the camera has fully started, press `Enter` a few times to see the console. You will see something like; `~ #`
3. Enter the following commands to the console:
```
dd if=/dev/mtd0 of=/tmp/sd/mtd0
dd if=/dev/mtd1 of=/tmp/sd/mtd1
dd if=/dev/mtd2 of=/tmp/sd/mtd2
dd if=/dev/mtd3 of=/tmp/sd/mtd3
dd if=/dev/mtd4 of=/tmp/sd/mtd4
dd if=/dev/mtd5 of=/tmp/sd/mtd5
dd if=/dev/mtd6 of=/tmp/sd/mtd6
dd if=/dev/mtd7 of=/tmp/sd/mtd7
```
_The really annoying thing is the Xiaomi applications running on the camera constantly write information to the console which can make typing commands very confusing. I suggest either copy and pasting the commands or temporarily close all the Xiaomi applications running on the camera so the console is not flooded with messages._

---
Once this is done, you will need to send me the files mtd0, mtd1, mtd2, mtd3, mtd4, mtd5, mtd6, mtd7 from your microSD card. From those files I can create a custom firmware compatible with Yi Dome (white).

---
In addition, can you paste the console contents during bootup with a microSD card inserted into the camera. I expect an output similar to this _(this output is from the Yi 1080p Dome)_:
```
System startup
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=1 
Interface:  MMC
  Device 0: Vendor: Man 035344 Snr 0ed375c0 Rev: 8.0 Prod: SU16G
            Type: Removable Hard Disk
            Capacity: 15193.5 MB = 14.8 GB (31116288 x 512)
Partition 1: Filesystem: FAT32 "NO NAME    "
            system volume information/

0 file(s), 1 dir(s)

u-boot_h20 not found!

env_h20 not found!

conf_h20 not found!

kernel_h20 not found!

rootfs_h20 not found!

home_h20 not found!

 do_auto_update: detect flash size: 0x1000000 
### UBOOT:test pin high! ###
Hit any key to stop autoboot:  0 
16384 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   7518-hi3518c-kernel
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1385488 Bytes = 1.3 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...
```

insantr, 2017-04-15:
@shadow-1 

> 2. Once the camera has fully started, press Enter a few times to see the console. You will see something like; ~ #

`Enter` is not worked....

In console i see:
```
000534.251 long time no beacons
000534.251 wlc_phy_resetcca_lcn40phy
cmd = 00000000
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(1)
```
every 3-5 seconds

insantr, 2017-04-15:
@shadow-1 
Console content during bootup with microsd:
```
System startup
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=1 
Interface:  MMC
  Device 0: Vendor: Man 035344 Snr e1e53cea Rev: 8.0 Prod: SL32G
            Type: Removable Hard Disk
            Capacity: 30436.5 MB = 29.7 GB (62333952 x 512)
Partition 1: Filesystem: FAT32 "NO NAME    "
            record/
    33280   log_wifi_connected.tar.gz 
    37888   log_first_login.tar.gz 
            system volume information/
            test/
            111/

2 file(s), 4 dir(s)

1: log_wifi_connected.tar.gz!
2: log_first_login.tar.gz!

u-boot_v201 not found!

env_v201 not found!

conf_v201 not found!

kernel_v201 not found!

rootfs_v201 not found!

home_v201 not found!

 do_auto_update: detect flash size: 0x1000000 
Hit any key to stop autoboot:  0 
16384 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 81000000 ...
   Image Name:   7518-hi3518c-kernel
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1384824 Bytes = 1.3 MiB
   Load Address: 80008000
   Entry Point:  80008000
   Loading Kernel Image ... OK
OK

Starting kernel ...
```

shadow-1, 2017-04-15:
@insantr 
As I mentioned in my post, the Xiaomi applications running on the camera continuously write information to the console which makes typing commands a bit difficult to see.

What happens when you type characters into the console? What happens when you press `Enter`?

insantr, 2017-04-15:
@shadow-1 
> What happens when you type characters into the console? 
> Do you see those characters within the console?

No. I don't see any my characters in console.

> What happens when you press Enter?

Nothing happens.

shadow-1, 2017-04-15:
@insantr 
If nothing happens at all when you try to send characters to the cameras, then there is most likely a connection issue between the Tx connection _(send connection)_ on your USB to Serial converter and the camera.

The Rx connection _(receive connection)_ between your USB to Serial converter and the camera is working perfectly as you are receiving everything that is being sent to the console.

Can you double check that the Tx signal is connected properly on your USB to Serial converter and that the connection is soldered properly on your cameras main circuit.

insantr, 2017-04-15:
@shadow-1 
Oh, you right..... there was a bad contact on rx line
So, i did make dumps mtd0 .... mtd7. How i can send you them?

shadow-1, 2017-04-15:
@insantr 
You can send me the files in any convenient way.

If you got a Dropbox account (or similar service), you can upload them onto Dropbox and send me a link to the files. I think you can also attach files to Github comments by dragging and dropping. I am not sure if there are any file size limitations though.

insantr, 2017-04-15:
@shadow-1   https://drive.google.com/open?id=0BwP_YUHxV98HeGUtNVdBOHZ0d1E

shadow-1, 2017-04-15:
@moral3jo @insantr @add210 
I have created the initial release which supports the Yi Dome (white).
Download is available here:
https://github.com/shadow-1/yi-hack-v3/releases

Enjoy all!

@insantr 
I was quite careful when creating this release to ensure that I don't make any mistakes.
However if something doesn't work, I might need your help with debugging.

insantr, 2017-04-15:
@shadow-1 
Not working.
I put the `home_v201` and `rootfs_v201` in root directory on sdcard.
Camera work properly, but webinterfase and telnet don't work.

Booting log:
```
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2017.04.15 20:03:13 =~=~=~=~=~=~=~=~=~=~=~=
02029.411 long time no beacons
002029.411 wlc_phy_resetcca_lcn40phy
cmd = 00000000
002034.412 long time no beacons
002034.412 wlc_phy_resetcca_lcn40phy
cmd = 00000000
p2p cur user(0) max_user_num(1) in_packet_loss(0) out_packet_loss(0)
002039.412 long time no beacons
002039.413 wlc_phy_resetcca_lcn40phy
cmd = 00000000
p2p cur user(0) max_user_num(1) in_packet_loss(0) out_packet_loss(0)
002049.414 long time no beacons
002049.414 wlc_phy_resetcca_lcn40phy
cmd = 00000000
 

System startup
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=1 
Interface:  MMC
  Device 0: Vendor: Man 035344 Snr e1e53cea Rev: 8.0 Prod: SL32G
            Type: Removable Hard Disk
            Capacity: 30436.5 MB = 29.7 GB (62333952 x 512)
Partition 1: Filesystem: FAT32 "NO NAME    "
            record/
    39424   log_first_login.tar.gz 
    33280   log_wifi_connected.tar.gz 
    45056   log_get_upgradefile_ok.tar.gz 
            system volume information/
            111/
  7613536   home_v201 
   957476   rootfs_v201 

5 file(s), 3 dir(s)

1: log_first_login.tar.gz!
2: log_wifi_connected.tar.gz!
3: log_get_upgradefile_ok.tar.gz!
4: home_v201!

 find index 5: home_v201 

5: rootfs_v201!

 find index 4: rootfs_v201 

u-boot_v201 not found!

env_v201 not found!

conf_v201 not found!

kernel_v201 not found!
reading rootfs_v201

magic 0x27051956 0x27051956 
arch 0x2 0x2 
size 0xe9be4 0x40 
type 0x7 0x2 
crc 0x8320ef04 
name 0.1.2-hi3518-rootfs 
[... 410 lines cut ...]
p2p_tnp.c(refresh_tnp_event_list-4815)[15:41:42.811]:got event(2)
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)

p2p_tnp.c(p2p_set_tnp_init_status-1846)[15:41:42.814]:p2p_set_tnp_init_status 4 send_msg ok!

000029.228 long time no beacons
000029.228 wlc_phy_resetcca_lcn40phy
cmd = 00000000
trans_json_ex result=1
trans_json_ex result=
trans_json_ex result=20000
trans_json_ex result="css_flag":0,"css_mode":1,"appParam":"","tz_offset":10800000,"timezone":"GMT+03:00","language":"en-US"
trans_json_ex result=10800000
trans_json_ex result=0
trans_json_ex result=1
trans_json_ex result=
cmd(rm /tmp/sd/log_first_login.tar.gz;rm /tmp/sd/first_login -r;mkdir /tmp/sd/first_login;ps > /tmp/sd/first_login/ps1.txt;cp /home/homever /tmp/sd/first_login;lsmod > /tmp/sd/first_login/lsmod.txt;ls -laSR /home > /tmp/sd/first_login/file.txt;cp /tmp/wpa_supplicant.conf  /tmp/sd/first_login;cp /tmp/log.txt /tmp/sd/first_login;df /home > /tmp/sd/first_login/df.txt;df / >> /tmp/sd/first_login/df.txt;tar -cf /tmp/sd/log_first_login.tar.gz /tmp/sd/first_login;rm /tmp/sd/first_login -r &)
rm: can't remove '/tmp/sd/first_login': No such file or directory
tar: removing leading '/' from member names
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
000039.230 long time no beacons
000039.230 wlc_phy_resetcca_lcn40phy
cmd = 00000000

```

shadow-1, 2017-04-15:
@insantr 
From the log file, I can see that it appears that the firmware files have flashed properly. I checked the home_v201 file and found that I accidentally made a very minor mistake in one of the files.

I have updated the release now. Can you re-download the files and try re-flashing the firmware.

insantr, 2017-04-15:
@shadow-1  
This works :thumbsup: 
[image]

> When you flash the new files, can you confirm that the yellow light blinks for roughly 30 seconds when you bootup.

Сonfirm.

shadow-1, 2017-04-16:
@insantr 
Fantastic news! I'm sure you would have noticed that I have manually upgraded the base firmware to 1.9.1.0G. You were running on the slightly older firmware; 1.9.1.0F.

closed 2017-04-16 by shadow-1

moral3jo, 2017-04-17:
how to open? i try but not sure if first head up or open in the middle?
anything hidden with sticks?

batrarobin, 2017-04-17:
@moral3jo,

Hey mate, no need to open your camera now. The custom firmware for your camera is now available to use. Just download the correct files and flash it. Make sure you read instructions before doing.

moral3jo, 2017-04-18:
ok, thanks. Can RTSP or get image camera from telnet or ftp? is possible decompile rmm or p2p_tnp?

shadow-1, 2017-04-18:
@moral3jo 
This custom firmware only supports using the official smartphone app at the moment (whilst defeating the region ban).

However RTSP support is currently under development. I am specifically interested in achieving RTSP support whilst supporting the official smartphone app at the same time.

At the moment it is not possible to get video/audio data through telnet. However FTP can be used to retrieve recordings on the microSD card that are setup through the official smartphone app.

It is not possible to decompile applications. However they can be disassembled to assembly language and certain aspects on how `rmm` and `p2p_tnp` work can be determined from assembly.

add210, 2017-04-19:
@shadow-1 it is being shipped from China. I ordered mine off of AliExpress. It could take a bit long.

shadow-1, 2017-04-19:
@add210 
Thanks to @insantr , we a tested version of this custom firmware for the Yi Dome (non 1080p/white).

Feel free to try it:
https://github.com/shadow-1/yi-hack-v3/releases

add210, 2017-04-22:
@shadow-1 I just got my USB to serial converter last night. I just saw that you have a working firmware for Yi Dome now. Sorry about the delay, man. And, thanks for the great work. I will try it on mine and see how it goes.
[image]

SimonMcN, 2018-04-10:
Hi, just a message to let you know that my keyboard didn't work in putty and I resolved it by connecting the ground wire as per this process: https://diyprojects.io/how-to-install-armbian-on-orange-pi-without-screen-or-keyboard-with-ftdi-cable/

I'm not sure, but I'm guessing that the signals are with reference to ground, so grounding it will help with negotiation ?
speed (baud): 115200
Data: 8
Stop: 1
Parity: None
Flow control XON/XOFF

Regards
Simon