shadow-1/yi-hack-v3 #6: can't pair with app because wifi password error

pasagame, 2017-04-14:
i set HTTPS on ProxyChains-ng and enable it, but i cannot pair camera with app. my camera say "wifi password error" (china language).I tried many times and was sure the password was correct. help me to to solved this issue. thank you

https://www.mx7.com/view2/zNsG8cVcovqOLrCG

shadow-1, 2017-04-14:
@pasagame 
WiFi password error has nothing to do with ProxyChains-ng.
You need ProxyChains-ng configured only to complete the final step of the pairing process.

A few things to check:
1. Are you absolutely sure that WiFi password you are entering is correct?
2. Are you trying to connect your camera to a 5GHz WiFi network? These Xiaomi cameras only support connecting to a 2.4GHz WiFi network.

pasagame, 2017-04-14:
1.wifi password is correct, if wifi password is wrong i cannot config ProxyChains-ng on browser (http://192.168.1.xxx/cgi-bin/read_proxy_config.sh)
2.my router set to 2.4GHz only

batrarobin, 2017-04-14:
@pasagame

Try changing wifi password encryption on your router to WPA2 AES or try others as well.

vaidyasr, 2017-04-14:
@pasagame 
If you were able to access the web page, then the message is not 'Wifi error' it is 'Wifi ic connected'. You need to have a working proxy to pair with the app.

pasagame, 2017-04-17:
wifi encryption is WPA2
I put proxy by image.
https://www.mx7.com/view2/zNsG8cVcovqOLrCG

shadow-1, 2017-04-19:
@pasagame 
I'm not sure whether the web link you posted is an image/screenshot or not as none of the images are loading on the page.
Can you post images through Github rather than external websites.

Can you explain what you mean by putting proxy by image? I'm not exactly sure what you mean by it.

pasagame, 2017-04-19:
[image]
From the picture, I will show that I edited the "ProxyChains-ng" (socks5 only) All other settings are default., but I can not connect to the application because the camera says wifi password error(Chinese speaking "wifi lianjian chong kong")

shadow-1, 2017-04-19:
@pasagame 
I suggest you upgrade to the latest release of the firmware available from:
https://github.com/shadow-1/yi-hack-v3/releases

This release simplifies the default configuration and provides a facility to check your ProxyChains-ng configuration from the camera itself. The check is not perfect at the moment as it can't check whether the Proxy server supports SSL connections.

I did a quick test of those proxy servers and none of them appear to be working for me.

pasagame, 2017-04-20:
I upgrade to the latest release but on about page show version 0.1.1 , i not sure my camera update to last version ready. my camera Yi 1080p (home_y20 , rootsf_y20)
[image]

batrarobin, 2017-04-20:
@pasagame 

That clearly hasn't flashed the latest firmware. Try again with correct file names and 16GB or less microSD formatted to FAT32. 

Files are here:
https://github.com/shadow-1/yi-hack-v3/releases

pasagame, 2017-04-20:
[image]

yes i use last file, but my camera not flash again.

batrarobin, 2017-04-20:
@pasagame 

That's a strange case. Suggest flashing the old version 0.1.1 files first and then do the same exercise with the newer version files 0.1.2.

batrarobin, 2017-04-20:
@pasagame 

Also, try a different SD card.

pasagame, 2017-04-20:
[image]
Thanks for your advice, I changed the sd card and flash it.

closed 2017-04-24 by pasagame