shadow-1/yi-hack-v3 #7: Stick to blue LED

maysamsh, 2017-04-14:
I've tried various proxies, checked them to make sure they are from China, and tried both YI Home and YI Smart (from Chinese App Store), but still cannot pair the cam. How can I troubleshoot the problem?

batrarobin, 2017-04-14:
@maysamsh
Try the app on android phone.

batrarobin, 2017-04-14:
@maysamsh
If you are stuck on the blue LED and the phone is not pairing, your proxy isn't reliable enough.  Try an android phone and try different proxies, it's just trial and error.

maysamsh, 2017-04-14:
I don't have an Android phone, but thanks, I'll try to sort it out.

batrarobin, 2017-04-14:
@maysamsh 

If you get hold of Android phone, make sure you don't install app from play store. Instead get the Chinese apk that I posted in the other thread.

maysamsh, 2017-04-15:
@batrarobin @shadow-1 Seems using a working proxy and Chinese app it's paired but I cannot get the video feed.

[image]

shadow-1, 2017-04-15:
@maysamsh 
You can see your real IP address because not all communications is going through the proxy server, only some of it is. The video/audio feed does not go through the proxy server (hence there is no performance penalty from using ProxyChains-ng).

Usually what I do is press the little gear icon on the top right corner, let that finish loading before going back to the video feed.

However you have some very strange messages in your log:
```
[./p2p_tnp][4/15/16:55:28:213]: check_login fail 0
...
[./p2p_tnp][4/15/16:57:14:438]: parse_msg: failed -3012 SessionHandle(1)!!
...
[./watch_process][4/15/16:57:26:732]: p2p_tnp crashed!
```

It appears that the cloud service is having trouble authenticating to the cloud service.
I would try re-pairing the camera to your smartphone.

batrarobin, 2017-04-16:
@maysamsh 

There are more chances of having success if you pair using Android phone with Chinese app.

maysamsh, 2017-04-20:
Since I don't have an Android phone, I've tried with several other proxies:

```
[./dispatch][4/20/9:58:47:552]: hw_type(2)
[./dispatch][4/20/9:58:50:387]: got sn(FxhQGQDBFs6VZFsdvTXMkTusGJvTZvye)[./dispatch][4/20/9:58:50:388]: got pwd()[./dispatch][4/20/9:58:50:389]: got ssid()[./dispatch][4/20/9:58:50:390]: got tnp_init_string()[./dispatch][4/20/9:58:50:391]: reset because no wifi config[./dispatch][4/20/9:58:50:393]: send_saveconfig_msg msg snd success[./dispatch][4/20/9:58:50:405]: get_config save conf
[./dispatch][4/20/9:58:50:407]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-y20, dlproto = micn
[./dispatch][4/20/9:58:50:470]: init ok, cost time(2923 ms)[./dispatch][4/20/9:58:50:522]: got pwd()()[./dispatch][4/20/9:58:50:523]: got ssid()()[./dispatch][4/20/9:58:50:524]: config save finish 1[./rmm][4/20/9:58:50:596]: f22 or f02
[./rmm][4/20/9:58:50:764]:  0xf
[./rmm][4/20/9:58:50:765]:  0x22
[./dispatch][4/20/9:58:50:845]: save_config[./rmm][4/20/9:58:51:186]: ai init ok!
[./rmm][4/20/9:58:51:187]: HI_MPI_AENC_AacInitHI_MPI_AENC_AacInit end
[./rmm][4/20/9:58:51:281]: rmm_init_aenc init ok!
[./rmm][4/20/9:58:51:366]: rmm_init_adec init ok!
[./rmm][4/20/9:58:51:367]: 
system call back libsns_f22.so_____________________
[./rmm][4/20/9:58:51:462]: sns_call_backsns_call_back
[./rmm][4/20/9:58:51:464]: HI_MPI_ISP_3A SUCCESS
[./rmm][4/20/9:58:52:109]: isp_set_ae: HI_MPI_ISP_SetExposureAttr  target_lum =60
[./rmm][4/20/9:58:52:123]: 
_____Gamma test for table 4__________
[./rmm][4/20/9:58:52:124]: modify drc (150), time:1492682332-127389[./rmm][4/20/9:58:52:133]: 
---------Rock_COMM_VENC_BindVpss VeChn = 0, VpssGrp = 0, VpssChn = 0
[./rmm][4/20/9:58:52:147]: rc set success! venc(0)
[./rmm][4/20/9:58:52:149]: 
---------Rock_COMM_VENC_BindVpss VeChn = 1, VpssGrp = 0, VpssChn = 1
[./rmm][4/20/9:58:52:151]: rc set success! venc(1)
[./rmm][4/20/9:58:52:168]: HI_MPI_RGN_Create 3 success![./rmm][4/20/9:58:52:169]: rmm_vpp_create 3 success![./rmm][4/20/9:58:52:170]: rmm_init success[./rmm][4/20/9:58:52:184]: font:/home/base/iso8859-1-24x48.bin
[./rmm][4/20/9:58:52:188]: filesize=36864
[./rmm][4/20/9:58:52:197]: init font /home/base/iso8859-1-24x48.bin ptr size(4)[./rmm][4/20/9:58:52:210]: font:/home/base/iso8859-1-8x16.bin
[./rmm][4/20/9:58:52:212]: filesize=4096
[./rmm][4/20/9:58:52:215]: init font /home/base/iso8859-1-8x16.bin ptr size(4)[./rmm][4/20/9:58:52:216]: load_font success[./rmm][4/20/9:58:52:218]: msg_proc success[./rmm][4/20/9:58:52:232]: 
_____Gamma test for table 1__________
[./rmm][4/20/9:58:52:237]: pid[967][./rmm][4/20/9:58:52:239]: pid[966][./rmm][4/20/9:58:52:241]: pid[965][./dispatch][4/20/9:58:52:280]: invalid msg 0x1002
[./rmm][4/20/9:58:52:284]: got RMM_SPEAK_WAIT[./rmm][4/20/9:58:52:348]: msg snd success[./cloud][4/20/9:58:52:832]: open share mem ok
[./p2p_tnp][4/20/9:58:53:536]: p2p_set_tnp_init_status 1 send_msg ok!
[./watch_process][4/20/9:58:54:149]: check_interval=10
[./watch_process][4/20/9:58:54:553]: process=dispatch;cmd=cd /home/app;./dispatch &;
[./watch_process][4/20/9:58:54:553]: process=cloud;cmd=cd /home/app;./cloud &;
[./watch_process][4/20/9:58:54:553]: process=rmm;cmd=reboot;
[./watch_process][4/20/9:58:54:554]: process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[./watch_process][4/20/9:58:54:554]: process=mp4record;cmd=cd /home/app;./mp4record &;
[./watch_process][4/20/9:58:54:555]: process=arp_test;cmd=cd /home/app;./arp_test &;
[./watch_process][4/20/9:58:54:555]: process=oss;cmd=cd /home/app;./oss &;
[./watch_process][4/20/9:59:4:178]: arp_test crashed![./rmm][4/20/9:59:6:556]: decoded QR-Code symbol "b=CNPFFKOkfJ5L2kZu&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw=="
[./rmm][4/20/9:59:6:557]: trans_json info=b=CNPFFKOkfJ5L2kZu&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw==;key=&p=
[./rmm][4/20/9:59:6:558]: trans_json result=VVgzNTIHAV08Zw==
[./rmm][4/20/9:59:6:559]: trans_json info=b=CNPFFKOkfJ5L2kZu&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw==;key=&s=
[./rmm][4/20/9:59:6:560]: trans_json result=TUFZU0FNLVBDX05ldHdvcms=
[./rmm][4/20/9:59:6:562]: trans_json info=b=CNPFFKOkfJ5L2kZu&s=TUFZU0FNLVBDX05ldHdvcms=&p=VVgzNTIHAV08Zw==;key=b=
[./rmm][4/20/9:59:6:571]: trans_json result=CNPFFKOkfJ5L2kZu
[./rmm][4/20/9:59:6:572]: rmm got /tmp/got_wpa ssid(MAYSAM-PC_Network) pwd(maysamnet2) bind(CNPFFKOkfJ5L2kZu)
[./dispatch][4/20/9:59:6:577]: invalid msg 0x1005
[./rmm][4/20/9:59:6:579]: got RMM_SPEAK_SCAN_OK[./rmm][4/20/9:59:6:580]: msg snd success[./dispatch][4/20/9:59:6:582]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-y20, dlproto = micn
[./rmm][4/20/9:59:6:584]: msg snd success[./dispatch][4/20/9:59:6:674]: msg snd success[./dispatch][4/20/9:59:6:680]: invalid msg 0x1004
[./rmm][4/20/9:59:6:681]: got RMM_SPEAK_CONNECTTING[./dispatch][4/20/9:59:7:535]: wifi disconnected, now reconnect wifi
[./dispatch][4/20/9:59:13:342]: wpa connected , now dhcp
[./watch_process][4/20/9:59:14:206]: cloud crashed![./watch_process][4/20/9:59:14:207]: p2p_tnp crashed![./dispatch][4/20/9:59:16:950]: wpa connected , now dhcp finish
[./dispatch][4/20/9:59:18:21]: bssid(a0:ab:1b:26:1e:94)[./dispatch][4/20/9:59:18:22]: ip(192.168.1.7)[./dispatch][4/20/9:59:18:55]: gw(192.168.1.1)[./dispatch][4/20/9:59:19:166]: gwmac(A0:AB:1B:26:1E:94)[./dispatch][4/20/9:59:19:197]: mask(255.255.255.0)[./dispatch][4/20/9:59:19:288]: mac(B0:D5:9D:90:BC:E3)[./dispatch][4/20/9:59:19:394]: signal_quality(100)[./dispatch][4/20/9:59: [... cut ...]
[./p2p_tnp][4/20/10:1:37:217]: Local NAT Type         :[./p2p_tnp][4/20/10:1:37:217]:  Port-Restricted Cone
[./p2p_tnp][4/20/10:1:37:218]: My Wan IP : 94.183.146.77
[./p2p_tnp][4/20/10:1:37:218]: My Lan IP : 192.168.1.7
[./p2p_tnp][4/20/10:1:37:219]: InitStr(MJFBIBLBICELLEHOOKFGPPECGLCJFKHEGKEALBJNIJBJCOAGPIIIBDMDPBDEAELAOGLCCBPBMKCNAHDPAKBM)
[./p2p_tnp][4/20/10:1:37:219]: did(TNPCHND-099303-CYYGG)
[./p2p_tnp][4/20/10:1:37:220]: key(8Fo9zRDvNtQYfZj8)
[./p2p_tnp][4/20/10:1:37:242]: p2p_set_tnp_init_status 4 send_msg ok!
```

shadow-1, 2017-04-20:
@maysamsh 
I suggest that you upgrade to the latest firmware. It includes a facility to test ProxyChains-ng config on the camera itself. Notice that there is a "Test Config" button. It is not been perfected yet. However it will catch most issues.

[image]

maysamsh, 2017-04-20:
@shadow-1 I have the latest firmware and tried the builtin test config.

shadow-1, 2017-04-21:
@maysamsh 
If the built in test fails. Then your proxy server isn't working. You will need to find one that works.

The limitation with the built in test at the moment is that it can not check whether the proxy server supports SSL connections. So some proxy servers may pass the test but still won't work. However most socks4 and socks5 proxy servers support SSL connections.

shadow-1, 2017-04-25:
@maysamsh 
As per the following thread: https://github.com/shadow-1/yi-hack-v3/issues/12#issuecomment-296974558
You have managed to resolve the connection issues which you were having by upgrading to yi-hack-v3 0.1.3.

closed 2017-04-25 by shadow-1