shadow-1/yi-hack-v3 #10: Are there options for lowering the firmware Dome Camera?

djartemon, 2017-04-18:
What should I do if rootfs_v201 was not written to the SD card? How to put it if home_v201 was installed without it?
Are there options for lowering the firmware? That would repeat everything correctly

batrarobin, 2017-04-18:
@djartemon
No need to lower the firmware, just put both files on SD card again and power cycle the camera, that should flash rootfs_v201 if not already flashed.

djartemon, 2017-04-19:
My log
[./dispatch][1/1/0:0:46:776]: hw_type(2)
[./dispatch][1/1/0:0:49:567]: got sn(7EpfpfMbZFI7NaREEFiTjmzodHUgvRqq)[./dispatch][1/1/0:0:49:568]: got pwd()[./dispatch][1/1/0:0:49:570]: got ssid()[./dispatch][1/1/0:0:49:571]: got tnp_init_string()[./dispatch][1/1/0:0:49:572]: reset because no wifi config[./dispatch][1/1/0:0:49:573]: send_saveconfig_msg msg snd success[./dispatch][1/1/0:0:49:574]: get_config save conf
[./dispatch][1/1/0:0:49:623]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-h19, dlproto = micn
[./dispatch][1/1/0:0:49:624]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-h19, dlproto = micn
[./dispatch][1/1/0:0:49:645]: init ok, cost time(2867 ms)[./dispatch][1/1/0:0:49:664]: got pwd()()[./dispatch][1/1/0:0:49:669]: got ssid()()[./dispatch][1/1/0:0:49:670]: config save finish 1[./rmm][1/1/0:0:49:705]:  0x97
[./dispatch][1/1/0:0:50:40]: save_config[./rmm][1/1/0:0:50:436]: ai init ok!
[./rmm][1/1/0:0:50:437]: HI_MPI_AENC_AacInitHI_MPI_AENC_AacInit end
[./rmm][1/1/0:0:50:520]: rmm_init_aenc init ok!
[./rmm][1/1/0:0:50:550]: rmm_init_adec init ok!
[./rmm][1/1/0:0:50:551]: 
system call back libsns_ov9732.so_____________________
[./rmm][1/1/0:0:50:648]: sns_call_backsns_call_back
[./rmm][1/1/0:0:50:650]: HI_MPI_ISP_3A SUCCESS
[./rmm][1/1/0:0:50:821]: 
_____Gamma test for table 4__________
[./rmm][1/1/0:0:50:837]: 
---------Rock_COMM_VENC_BindVpss VeChn = 0, VpssGrp = 0, VpssChn = 0
[./rmm][1/1/0:0:50:852]: rc set success! venc(0)
[./rmm][1/1/0:0:50:854]: 
---------Rock_COMM_VENC_BindVpss VeChn = 1, VpssGrp = 0, VpssChn = 1
[./rmm][1/1/0:0:50:855]: rc set success! venc(1)
[./rmm][1/1/0:0:50:869]: HI_MPI_RGN_Create 2 success![./rmm][1/1/0:0:50:872]: rmm_vpp_create 2 success![./rmm][1/1/0:0:50:873]: rmm_init success[./rmm][1/1/0:0:50:882]: font:/home/base/iso8859-1-16x32.bin
[./rmm][1/1/0:0:50:884]: filesize=16384
[./rmm][1/1/0:0:50:888]: init font /home/base/iso8859-1-16x32.bin ptr size(4)[./rmm][1/1/0:0:50:889]: font:/home/base/iso8859-1-8x16.bin
[./rmm][1/1/0:0:50:891]: filesize=4096
[./rmm][1/1/0:0:50:893]: init font /home/base/iso8859-1-8x16.bin ptr size(4)[./dispatch][1/19/6:41:36:98]: DISPATCH_SET_DEFAULT_TIME 1453185696[./rmm][1/19/6:41:36:99]: load_font success[./rmm][1/19/6:41:36:100]: msg_proc success[./rmm][1/19/6:41:36:103]: 
_____Gamma test for table 1__________
[./rmm][1/19/6:41:36:105]: pid[956][./rmm][1/19/6:41:36:112]: pid[958][./rmm][1/19/6:41:36:114]: pid[957][./cloud][1/19/6:41:36:449]: open share mem ok
[./rmm][1/19/6:41:37:32]: msg snd success[./dispatch][1/19/6:41:37:32]: invalid msg 0x1002
[./rmm][1/19/6:41:37:33]: got RMM_SPEAK_WAIT[./p2p_tnp][1/19/6:41:37:33]: p2p_set_tnp_init_status 1 send_msg ok!
[./watch_process][1/19/6:41:37:811]: check_interval=10
[./watch_process][1/19/6:41:38:24]: process=dispatch;cmd=cd /home/app;./dispatch &;
[./watch_process][1/19/6:41:38:25]: process=cloud;cmd=cd /home/app;./cloud &;
[./watch_process][1/19/6:41:38:25]: process=rmm;cmd=reboot;
[./watch_process][1/19/6:41:38:25]: process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;
[./watch_process][1/19/6:41:38:26]: process=mp4record;cmd=cd /home/app;./mp4record &;
[./watch_process][1/19/6:41:38:26]: process=arp_test;cmd=cd /home/app;./arp_test &;
[./watch_process][1/19/6:41:38:26]: process=oss;cmd=cd /home/app;./oss &;
[./watch_process][1/19/6:41:47:825]: arp_test crashed![./dispatch][1/19/6:41:58:267]: invalid msg 0x1002
[./rmm][1/19/6:41:58:269]: got RMM_SPEAK_WAIT[./rmm][1/19/6:41:58:270]: msg snd success[./rmm][1/19/6:42:7:409]: decoded QR-Code symbol "b=CNomynxFnevOTStv&s=bWstZ3Vlc3Q=&p=CQt5cmZcWDg="
[./rmm][1/19/6:42:7:410]: trans_json info=b=CNomynxFnevOTStv&s=bWstZ3Vlc3Q=&p=CQt5cmZcWDg=;key=&p=
[./rmm][1/19/6:42:7:411]: trans_json result=CQt5cmZcWDg=
[./rmm][1/19/6:42:7:412]: trans_json info=b=CNomynxFnevOTStv&s=bWstZ3Vlc3Q=&p=CQt5cmZcWDg=;key=&s=
[./rmm][1/19/6:42:7:414]: trans_json result=bWstZ3Vlc3Q=
[./rmm][1/19/6:42:7:415]: trans_json info=b=CNomynxFnevOTStv&s=bWstZ3Vlc3Q=&p=CQt5cmZcWDg=;key=b=
[./rmm][1/19/6:42:7:416]: trans_json result=CNomynxFnevOTStv
[./rmm][1/19/6:42:7:417]: rmm got /tmp/got_wpa ssid(mk-guest) pwd(12345678) bind(CNomynxFnevOTStv)
[./dispatch][1/19/6:42:7:423]: invalid msg 0x1005
[./rmm][1/19/6:42:7:426]: got RMM_SPEAK_SCAN_OK[./rmm][1/19/6:42:7:429]: msg snd success[./dispatch][1/19/6:42:7:431]: in choose_server, region_id = 1, api_server = https://api.xiaoyi.com, sname = familymonitor-h19, dlproto = micn
[./rmm][1/19/6:42:7:432]: msg snd success[./dispatch][1/19/6:42:7:495]: msg snd success[./dispatch][1/19/6:42:7:496]: invalid msg 0x1004
[./rmm][1/19/6:42:7:499]: got RMM_SPEAK_CONNECTTING[./dispatch][1/19/6:42:8:461]: wifi disconnected, now reconnect wifi
[./dispatch][1/19/6:42:14:284]: wpa connected , now dhcp
[./dispatch][1/19/6:42:14:562]: wpa connected , now dhcp finish
[./dispatch][1/19/6:42:15:597]: bssid(6c:3b:6b:7d:4d:e8)[./dispatch][1/19/6:42:15:598]: ip(192.168.55.115)[./dispatch][1/19/6:42:15:620]: gw(192.168.55.1)[./dispatch][1/19/6:42:16:719]: gwmac(6C:3B:6B:72:D8:9C)[./dispatch][1/19/6:42:16:763]: mask(255.255.255.0)[./dispatch][1/19/6:42:16:831]: mac(B0:D5:9D:8E:B5:43)[./dispatch][1/19/6:42:16:909]: signal_quality(66)[./dispatch][1/19/6:42:16:909]: invalid msg 0x1006
[./rmm][1/19/6:42:16:913]: got RMM_SPEAK_WIFI_CONNECTTED[./dispatch][1/19/6:42:16:914]: msg snd success

djartemon, 2017-04-19:
Hello, made an image of the camera sections. At me the camera (Dome_Camera_YHS_1916_CN White) it is impossible to adjust a proxy, pereprobyval absolutely different variants. Can you help me? Where to dig?)
https://drive.google.com/open?id=0B3o0lmEgcv_DTEZSMXFyRnZmTTg

shadow-1, 2017-04-19:
@djartemon 
From what I understand from your situation. You flashed the firmware and you had access to the web server. This means that both `home_v201` and `rootfs_v201` flashed to the camera properly. If either image didn't flash properly, then the web interface will not work.

Then you disabled the web server so you no longer have access to the web interface?

However for some reason you no longer access the camera through telnet either? Did you disable telnet as well?

Normally I would suggest to write a small script enabling telnet and web server through `startup.sh` on the microSD card. However another Github member notified me that there is a bug with the way I was executing `startup.sh` and this may not work for everyone. The bug has been fixed and will be pushed in the next release of the firmware.

Do you have access to the camera through FTP server? If you do, you can recover telnet access and web server through FTP.

Alternatively you can re-flash the firmware on your camera as well. You can not flash the same version of the firmware again due to checks that occur on the camera. The Yi Dome (white) only has one release of yi-hack-v3 so you can not downgrade to an older version before re-flashing to the current version.

However I have now created recovery images for those that would like to go back to stock firmware. You can flash the stock firmware back onto your camera and then flash the current release of yi-hack-v3 again. The recovery images are available in the link below in the yi-hack-v3/Recovery folder.

https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

djartemon, 2017-04-19:
Many thanks! Recovery was excellent!
Now it remains to defeat the problem with "Warning! This test can not determine whether the proxy server supports SSL connections." It is recommended to additionally test the proxy server manually.
Proxy Server Connection Failed. Ensure that ProxyChains-ng has been configured. "

djartemon, 2017-04-19:
I enter in the proxy list addresses previously checked, but anyway, this is   http://joxi.ru/E2pVe6Zf9bbVMr

shadow-1, 2017-04-19:
Good to hear that your camera is working again.

Yes, if your camera is region locked. If you disable ProxyChains-ng, the annoying ban message will play and the camera will not work.

With ProxyChains-ng, you need to enter proxy servers which are located in Mainland China. From your screenshot of the test page, it is showing that the none of the proxy servers entered are currently working on the camera.

There are many sources of free proxy servers online, however the trick is to find one that is reasonably reliable and that is actually located in Mainland China. Proxy servers located in Hong Kong, Taiwan etc will not work. Most socks proxy servers support SSL connections which is required for this hack while most http proxy servers do not.

shadow-1, 2017-04-19:
@djartemon 
Another thing I forgot to mention, it is worth retrying the test a couple of times if you are confident that the proxy servers work. Sometimes if the proxy server is a bit slow or unreliable, the test will fail due to timeout and not because the proxy server is down.

djartemon, 2017-04-19:
And you can throw off a screenshot of how the test page should look if the test succeeds

shadow-1, 2017-04-19:
@djartemon An example of how the test page looks like when it passes.

[image]

djartemon, 2017-04-20:
Can you install the firmware from the US camera to the Chinese? Since for some reason the proxy does not work, it has over-populated a huge number of addresses - and still does not

shadow-1, 2017-04-20:
@djartemon 
The same firmware is used on the US version of the camera as the Chinese version. At the moment there is no other way to defeat the region ban.

Having a huge list of proxy servers isn't going to work if none of them work. Even if some of the proxy servers work, I believe there may be timeout issues (have not tested). By the time ProxyChains-ng finds a working proxy server from a big list, the message to the Xiaomi server may have timed out already. All you need to is to find one or two proxy servers that work.

Some people have posted working proxy servers on this Github page. In addition there are numerous lists online. However like I said, the trick is to find one which works and is based in Mainland China.

shadow-1, 2017-04-25:
@djartemon 
I assume that your issues with the Yi Dome camera are now resolved as per the other thread you contributed to. Feel free to open a new thread if you find other issues.

closed 2017-04-25 by shadow-1