shadow-1/yi-hack-v3 #11: Прошивка home_h20m

PonomarevAG, 2017-04-22:
Hey. Is it possible to add a solution for the firmware home_h20m?

shadow-1, 2017-04-22:
@PonomarevAG 
What issue are you having with firmware home_h20m?
From the firmware file you described, you have a Yi 1080p Dome camera?

PonomarevAG, 2017-04-22:
Yes, quite right I have a yi 1080p camera. But it looks like home_h20m, and not as indicated in the home_h20 table. How can I flash my camera? Thank you.

shadow-1, 2017-04-22:
@PonomarevAG
The official firmware file is named home_h20m. However the firmware files for this firmware is rootfs_h20 and home_h20.

A few guidelines:
1. Download the custom firmware from https://github.com/shadow-1/yi-hack-v3/releases . The files you need for your camera are rootfs_h20 and home_h20 as indicated in the table on the main page.
2. Copy the firmware files onto a FAT32 formatted microsd card. Preferably 16gb in size or less.
3. Boot the camera. It will take roughly 30 seconds longer than usual to boot as the firmware is flashing. During this time the yellow light blinks on the camera.

After the firmware has flashed, the camera will boot as per normal. Telnet, FTP, Web server and ProxyChains-ng are enabled by default.

If your camera is region locked. You will need to configure ProxyChains-ng for the camera to work. If your camera is not region locked. ProxyChains-ng is not needed and can be disabled through the Web interface.

PonomarevAG, 2017-04-22:
Thanks for the help. I shot the camera. Now I can not connect to it via the android application. Although she stopped saying that she will not work outside the territory of China. I understand now it is necessary to choose a proxy? How can I get into the web interface of the camera? Http: // ip_of_camera / does not work at this link. The camera is connected to your computer, but the browser does not download the web

PonomarevAG, 2017-04-22:
Can I do something wrong to get into the web interface of the camera?

batrarobin, 2017-04-22:
@PonomarevAG 
You will need to know what IP address your router has assigned to the camera, for example 192.168.1.13
Then open that address in your browser, for example http://192.168.1.13.

Easy way of finding out the IP address is to install "Network Scanner" app on your android phone:

https://play.google.com/store/apps/details?id=com.myprog.netscan&hl=en

This will list all the devices connected your home router. Just find the IP address of the camera, it will be listed under "Shenzhen Zowee Co. Ltd"

PonomarevAG, 2017-04-23:
Thanks for the help. I entered the camera interface. Do you need to register a proxy in this form?

dynamic_chain

proxy_dns

remote_dns_subnet 224

tcp_read_time_out 15000
tcp_connect_time_out 8000

[ProxyList]
222.74.225.231:3128

16/5000
Libo po-drugomu?
Or is it different?

PonomarevAG, 2017-04-23:
And on the router, should I also have a proxy configured?

shadow-1, 2017-04-23:
@PonomarevAG 
No need to have any special configuration on your router.

I suggest you look for socks4 or socks5 proxy servers. They are the most likely to work correctly. The proxy server you have listed is a http proxy server. I have not found a http proxy server that works correctly yet. This is something I need to do more testing on.

Under `[ProxyList]`, the syntax for adding proxy servers is like this:
`socks4 PROXY_IP PROXY_PORT` or `socks5 PROXY_IP PROXY_PORT`

PonomarevAG, 2017-04-23:
That is, it will look like this
[ProxyList]
SOCKS5 123.56.199.152 3641?

batrarobin, 2017-04-23:
@PonomarevAG 

Your config file should look like this:

https://github.com/shadow-1/yi-hack-v3/issues/12#issuecomment-296386994

shadow-1, 2017-04-23:
@PonomarevAG 
It should be:
```
[ProxyList]
socks5 123.56.199.152 3641
```
ProxyChains-ng is case sensitive.
I have tested that particular proxy server, and it appears that it is based in Hong Kong and not Mainland China.

PonomarevAG, 2017-04-23:
Thank you all for your help. Now it remains to select a proxy. Someone, please, tell me where it is better to take a proxy?

PonomarevAG, 2017-04-23:
Thanks to all!!! It turned out, very glad that there are people who help!

shadow-1, 2017-04-23:
@PonomarevAG 
Glad that everything is working now.
It is not that easy for those that are not familiar online proxy servers and with what needs to be done.

closed 2017-04-23 by shadow-1