shadow-1/yi-hack-v3 #12: Proxy Server Connection Failed. Ensure that ProxyChains-ng has been configured.

add210, 2017-04-22:
I can't seem to get it working, trying hundreds of different proxy from GatherProxy.com, either one at a time or two or three at a time, none of those is working at all.

I have flashed the firmware to the latest version (v 0.1.2) and everything seems to be working with WiFi connected sucessfully. The ProxyChains-ng just wont do it.

add210, 2017-04-22:
Btw, the error message comes from "Test Config".

batrarobin, 2017-04-22:
@add210

Make sure you are using socks4 or socks5 proxy. Try xroxy.com to find more proxy servers. 

Also delete the following lines from config file and save:

proxy dns
remote dns subnet 224

add210, 2017-04-22:
Do i need to keep my micro SD card inside the camera? I removed mine after flashing succesfully. Does it matter?

add210, 2017-04-22:
I will try again.

add210, 2017-04-22:
I am trying different proxies from

http://free-proxy.cz/en/proxylist/country/CN/socks/speed/all

add210, 2017-04-22:
Also, trying these proxies from

http://www.xroxy.com/proxylist.php?port=&type=All_socks&ssl=ssl&country=CN&latency=&reliability=#table

[image]

[image]

Still no luck... not sure what I did wrong. I thought it would be a snap flashing and setting up.

add210, 2017-04-22:
@batrarobin I deleted that 2 lines. Still not working.

add210, 2017-04-22:
Been trying for 4 hours without success. I tried re-flashing my Yi Dome and everything. None seems to help.

Every time, the "Retrieving Pairing Status" would just give out error "Fail to load. Please reset the camera." and I have to put up new proxies, then reboot and Wi-Fi pairing over and over again.

I will try again tonight. So tired now.

batrarobin, 2017-04-22:
@add210

You don't need to flash the custom firmware again and again.  You can leave the SD card in the camera, it won't flash those 2 files again as it detects that the firmware version is the same so doesn't flash them again. 

You just need to spend your time looking for a working proxy, that's the only trick. As you mentioned that you put 2 or 3 proxies together in the list, that is good. Please don't put too many proxies in that list as it will take forever to test each of them and xiaomi server will time out if the list is long. 

Another thing that you could try is when you click "Test config" and if it results in a non-working proxies, try pressing "Test config" again because sometimes it takes a couple of tries for proxy to work.

batrarobin, 2017-04-22:
@add210 

Another thing, keep those 2 lines deleted in the config permanently.

batrarobin, 2017-04-22:
@add210

Are you using Android phone? If yes do you have Chinese apk installed?  Yi app from Play Store won't work.

add210, 2017-04-22:
@batrarobin I am using Android phone with Yi Home CN apk.

Test Config always shows

"Proxy Server Connection Failed. Ensure that ProxyChains-ng has been configured."

I will try again with some more proxies. I have a few hours to do it before going to bed.

mikeschi, 2017-04-22:
Persistence does pay off. Today I was successful in connecting my 1080P Black dome camera. Limit the number of proxys and use the build in test. I'm using a Socks5 proxy. 

Shadow-1, many thanks. You are brilliant!

add210, 2017-04-22:
I tried all of the proxies one by one, one at a time. None of them work. So weird.

Can anyone share the "ProxyChains-ng Test" page which shows the successful connection? You can blur out the IP of the proxy you are using. I am just wondering what it looks like with a working proxy. I have tried more than 400 proxies with no luck...

batrarobin, 2017-04-22:
@add210 

See this thread, it has a snapshot of a working config file.

https://github.com/shadow-1/yi-hack-v3/issues/10

add210, 2017-04-22:
@batrarobin Thanks. I read all posts on all issues. Dunno how I missed that one. At least it gives me hope to work on it a few more hours. lol

shadow-1, 2017-04-22:
@add210 
Some people on this Github thread have posted their working Proxy Servers: https://github.com/shadow-1/yi-hack-v3/issues/1#issuecomment-294007013

This seems to be a sticking point for many people. Finding a working Proxy server that is located within Mainland China. I have also noticed that whilst there are various lists online with many proxy servers, majority of the proxy servers are down.

add210, 2017-04-22:
@shadow-1 I am re-flashing the original firmware and flashing it back again to v 0.1.2 to see if it helps. I am desperated. lol

I will see if it helps. Also, I will try that proxy and see if it works.

add210, 2017-04-22:
```
dynamic_chain

tcp_read_time_out 15000
tcp_connect_time_out 8000

[ProxyList]

socks 183.232.25.100 3080
```

Still not working...

[image]

add210, 2017-04-22:
I will just keep trying. At least i know it works on others'

batrarobin, 2017-04-23:
@add210 

This proxy that you added is a working proxy, I just tested it. I had to give it a couple of tries but it did work. 

Maybe your ISP is blocking it. Do you have a spare mobile with a sim from a different ISP which could setup as a wifi hot-spot? Try that and see if that works... It might just be that your broadband ISP might be blocking certain proxies.

add210, 2017-04-23:
@batrarobin good call. I will test that out.

add210, 2017-04-23:
Not working... something is so wrong here...

[image]

add210, 2017-04-23:
I brought my camera to my brother's place and tried connecting it via different ISP. The result is the same. Proxy server connection failed. I have no idea what is wrong with my Yi Dome.

Anyone using this hack with the Yi Dome (720p White)?

shadow-1, 2017-04-23:
@add210 
We know what the issue is; the proxy server is not working where you are located.

I find that particular proxy server a bit unreliable. It sometimes works for me without issues, sometimes I need to retry a couple of times whilst sometimes it doesn't work at all.

You will have to find a proxy server that works properly where you are located. I can suggest a number of websites (all easily found with Google). However the trick is testing them to make sure they work. As you have experienced, many proxy servers listed on these websites do not work or do not provide the required functionality.

Another way to test proxy servers is to do it manually by setting the proxy server on your web browser and browsing to a webpage like; https://whatismyipaddress.com/ 

I am currently developing a more robust test page for the camera at the moment and I am attempting to create a feature where proxy servers can automatically be found online, testedand enter into the configuration.

However this will take me a bit of time to develop, test and release.

add210, 2017-04-23:
I found many working proxies, but none of them would let my Yi Dome connect successfully.

Here is one of the proxy I used that is working on my browser when tested.

[image]

shadow-1, 2017-04-23:
@add210 
That is very odd. I got a few tips for you to consider:
* Are you testing socks4 or socks5 proxy servers? _These are the only ones I have had luck with so far._
* Are you testing the proxy server by visiting https://whatismyipaddress.com/ and not http://whatismyipaddress.com/ (notice the first one starts with https). _You have to test the proxy server against a https website to ensure that SSL connections work._

Testing through your browser is more robust than the test page at the moment. If you tested a socks4/socks5 proxy server against a https website, then it should work properly on the camera. Perhaps it might be you configuration? Can you post these troublesome proxy servers that should work so I can test them and see what the issue might be.

add210, 2017-04-24:
@shadow-1 I tested it with SOCKS5.

Yes, I tested the proxies with https://whatismyipaddress.com

I am pretty sure the proxy I am testing is working, but somehow my Yi Dome wont connect thru it.

add210, 2017-04-24:
one of the proxies I used is this one.

[image]

batrarobin, 2017-04-24:
@add210 

Can you post a snapshot of your config file once again... If you have tested using https on your PC and it works, then it should work on your camera as well... It's a very strange case.

add210, 2017-04-24:
Yi Dome config file page

[image]

add210, 2017-04-24:
But it works perfectly on browser.

[image]

add210, 2017-04-24:
Is there a log on what is going on, like which proxy the camera is using (in case we put up 3-4 proxies), what error it is causing, etc?

shadow-1, 2017-04-24:
@add210 
I tested your configuration and it is working properly for me. Sometimes I need to retry once or twice on the test page on the camera. Perhaps try being a little more patient. Start the camera, attempt to pair the camera to your phone, wait until it times out. If it fails, wait a few minutes and try it a second time just to be sure.

ProxyChains-ng does not provide logging functionality (I think that is deliberate by the author as it is a anonymising tool), however this is a very good feature request as it comes in very handy when troubleshooting. It is something I need to look into.

add210, 2017-04-24:
@shadow-1 I tried waiting it out and rebooting and re-pairing over and over using one proxy at a time.

I will try again just to be sure.

add210, 2017-04-24:
I first check that the proxy i am using is working via browser (firefox). checked.

Then, I plug the Yi Dome and boot it up with that proxy configured. checked. (note that the test page always return fail result even tho the proxy is good)

Then, I pair the Yi Dome with my Yi Home (CN android app). checked.

The pairing would always fail. I reboot the Yi Dome and waited for it to boot up and re-do the pairing again.

I have no idea what is going wrong on my end. I have no way to trace it.

I suspect that the proxy somehow didn't get called into use (since the browser shows that the proxies I used are good), but I have no way to prove it.

[image]

shadow-1, 2017-04-24:
@add210 
I need to develop some troubleshooting pages on the camera itself. I have no idea why it is not working for you. Just about everyone I helped has managed to get it working once they find a working proxy server.

I will have to guide you on how to test ProxyChains-ng manually through telnet as I don't have anything automated yet.

Run the following command through telnet a few times and post what is being shown:
`proxychains4 wget -O- -q http://ipinfo.io`

Notice that I am suggesting to use http://ipinfo.io instead of http://www.freegeoip.net/csv which is what the test page is using as I find it more reliable.

This will show the error message that is coming up. I suggest trying the command a few times to see if it passes sometimes and not other times or if it is always failing.

djartemon, 2017-04-24:
http://joxi.ru/eAOlebOsxQXber

shadow-1, 2017-04-24:
@djartemon 
That is a very concerning error message! I have to double check whether I didn't accidentally miss libproxychains4.so from the firmware.

Can you run the following commands and post the result:
```
cd /home/app/locallib
ls
```

djartemon, 2017-04-24:
http://joxi.ru/n2YndlJho4z9bA

shadow-1, 2017-04-24:
@djartemon 
It's very very strange that you are getting this error message. I don't have this camera, so I am relying on you and @add210 to help me out.

From your screenshot I can see that libproxychains4.so is located in the correct location and has the correct permissions.
Can you run the following command on the camera through telnet and post the result:
`echo $LD_LIBRARY_PATH`

djartemon, 2017-04-24:
```
cd /home/app/locallib
ls -l
echo $LD_LIBRARY_PATH
```

after 
`proxychains4 wget -O- -q http://ipinfo.io`

I got the following
http://joxi.ru/v294OvYs39yvPm 

That is, this way it will be checked

add210, 2017-04-24:
I am on it. Will post update.

add210, 2017-04-24:
`
(none) login: root
Password:
Welcome to HiLinux.
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
couldnt locate libproxychains4.so
~ # ls
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
couldnt locate libproxychains4.so
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
couldnt locate libproxychains4.so
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
couldnt locate libproxychains4.so
`

add210, 2017-04-24:
~ # cd /home/app/locallib
/home/app/locallib # ls -l
total 125
-rwxr-xr-x    1 root     root         82714 Apr 15  2017 libproxychains4.so
-rwxr-xr-x    1 root     root         29113 Apr 15  2017 libsns_ov9732.so
-rwxr-xr-x    1 root     root         15040 Apr 15  2017 luaext.so
/home/app/locallib # echo $LD_LIBRARY_PATH
/usr/local/lib:/usr/lib:/home/lib:/home/hisiko/hisilib:/home/app/locallib
/home/app/locallib #

add210, 2017-04-24:
after

cd /home/app/locallib

I can now run the command

/home/app/locallib # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading ./libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Dynamic chain  ...  183.232.25.100:3080  ...  52.221.12.168:80  ...  OK
wget: error getting response: Operation now in progress
/home/app/locallib #

shadow-1, 2017-04-24:
@djartemon @add210 
Very strange! No wonder you two were having trouble with the firmware.
Try the following workaround by entering the following commands:
```
cd /lib
ln -s /home/app/locallib/libproxychains4.so libproxychains4.so
```

Once you have done this. Run the test command again a few times:
`proxychains4 wget -O- -q http://ipinfo.io`

add210, 2017-04-24:
/home/app/locallib # echo $LD_LIBRARY_PATH
/usr/local/lib:/usr/lib:/home/lib:/home/hisiko/hisilib:/home/app/locallib
/home/app/locallib #

add210, 2017-04-24:
/home/app/locallib # cd /lib
/lib # ln -s /home/app/locallib/libproxychains4.so libproxychains4.so
/lib # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading ./libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Dynamic chain  ...  183.232.25.100:3080  ...  54.255.155.142:80  ...  OK
{
  "ip": "207.226.142.68",
  "hostname": "207-226-142-68.static.pccwglobal.net",
  "city": "Shanghai",
  "region": "Shanghai",
  "country": "CN",
  "loc": "31.0456,121.3997",
  "org": "AS3491 Beyond The Network America, Inc."
}/lib # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading ./libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Dynamic chain  ...  183.232.25.100:3080  ...  54.255.155.142:80  ...  OK
{
  "ip": "207.226.142.68",
  "hostname": "207-226-142-68.static.pccwglobal.net",
  "city": "Shanghai",
  "region": "Shanghai",
  "country": "CN",
  "loc": "31.0456,121.3997",
  "org": "AS3491 Beyond The Network America, Inc."
}/lib #

add210, 2017-04-24:
@shadow-1 It seems like we are getting somewhere now.

shadow-1, 2017-04-24:
@add210 
Congratulations! It is now working!! When you try to pair your camera to your phone it will now work.
There is an issue with the firmware file home_v201 itself. I accidentally compiled proxychains4 incorrectly for this particular firmware.

add210, 2017-04-24:
The telnet timed out. I telnet again and re-run it a few more times.

~ # cd /lib
/lib # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading ./libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Dynamic chain  ...  183.232.25.100:3080  ...  52.221.12.168:80  ...  OK
{
  "ip": "207.226.142.68",
  "hostname": "No Hostname",
  "city": "Shanghai",
  "region": "Shanghai",
  "country": "CN",
  "loc": "31.0456,121.3997",
  "org": "AS3491 Beyond The Network America, Inc."
}/lib # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading ./libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Dynamic chain  ...  183.232.25.100:3080  ...  52.221.12.168:80  ...  OK
{
  "ip": "207.226.142.68",
  "hostname": "207-226-142-68.static.pccwglobal.net",
  "city": "Shanghai",
  "region": "Shanghai",
  "country": "CN",
  "loc": "31.0456,121.3997",
  "org": "AS3491 Beyond The Network America, Inc."
}/lib #

add210, 2017-04-24:
My app just returned fail pairing... so weird. I will reboot and re-pairing it again.

add210, 2017-04-24:
@shadow-1 Hey, I got it finally!

Yay, you the man! Thanks a lot for your step-by-step guidance.

[image]

shadow-1, 2017-04-24:
@add210 @djartemon 
Can you please confirm that the camera is now working after this workaround.
If it is working, I will correct home_v201 for version 0.1.2 of yi-hack-v3 and I will release a minor update tomorrow 0.1.3 to avoid confusion with those that may still have the old firmware installed.

djartemon, 2017-04-24:
@add210  What did you do to make it work?

add210, 2017-04-24:
@djartemon go to the camera config page and reboot. Then after the reboot, just pair it with your phone as you normally would. This time, it will pair within a few seconds.

shadow-1, 2017-04-24:
@djartemon 
Enter the following commands as a workaround:
```
cd /lib
ln -s /home/app/locallib/libproxychains4.so libproxychains4.so
```

From your last screenshot, it appears that you have some issues with your proxy servers as you are getting a Connection Refused error.

From your screenshot, I suggest removing lines; `proxy_dns` and `remote_dns_subnet 224` from your ProxyChains-ng configuration. In addition, try using only one or two proxy servers for now. I will release a fix in version 0.1.3 which will allow a large lists to work correctly.

add210, 2017-04-24:
@shadow-1 It is now working as it should. I have a few questions and concerns tho.

Is it safe to use proxy to view & record video at home? Can anyone intercept it if it is proxied?

and what if the proxy I am using just stop working? Do I need to re-do all the telnet things or I can just edit the proxy under config page?

shadow-1, 2017-04-24:
@add210 
Not all communications is going through the proxy. Only some communication. All video/audio data does not go through the proxy (that is why there is no performance issues when using this hack). In addition, all video/audio data is encrypted.

If the proxy suddenly stops working, your camera will continue to be operation like normal until the next time the camera performs the region check. That is, upon bootup of the camera, it also appears that the region check is also performed roughly after 3-4 days of inactivity as well (hard to test exactly how long).

Luckily even after the region check, if the proxy server is down, the annoying ban message is not played. However the camera won't work with the smartphone app anymore.

If you need to add working proxy servers onto the list, you will only need to edit the configuration on the web interface. This telnet workaround is only required to be done once and hopefully never again once I fix the firmware.

[11 more replies]

closed 2017-04-25 by shadow-1