shadow-1/yi-hack-v3 #13: Access Video Files Over Telnet

jinx10000, 2017-04-23:
How can you access video files over FTP?  
SD Card does not appear to be mounted in the FTP server.  
However, I can see the files when pulling the SD Card and looking in a card reader.

shadow-1, 2017-04-23:
@jinx10000 
Yes it is possible to access video files that have been setup through the official app.
The microSD card is mounted under `/tmp/sd` on these cameras.

jinx10000, 2017-04-23:
Do I need to edit something to make this visible in ftp?  

Or is it only possible through Telnet?  I can see it in telnet since you pointed it out. 

I want to set up a script to copy mp4's and push to my own cloud server.  Not sure how to do this through Telnet.

shadow-1, 2017-04-23:
@jinx10000
To do what you want, ftp is the right tool for the job. Telnet won't help you.

Are you sure you can't see any files within `/tmp/sd` through ftp? I have always been able to browse and download/upload files directly to the microsd card through ftp. I do this all the time. If you can see the files through telnet, I am a little confused why you can't see them in ftp.

Can you post a few screenshots of what you see through telnet and ftp?

jinx10000, 2017-04-23:
@shadow-1 
[image]
[image]
Posted Screenshots

shadow-1, 2017-04-23:
@jinx10000
I don't normally use Explorer to browse ftp. It's very strange that your ftp appears to be limited to `/home/app`.

Try a dedicated ftp program file FileZilla. You will be able to browse to `/tmp/sd`.

jinx10000, 2017-04-24:
Thank you.  That worked.  I wanted to use Explorer because I don't work with FTP much.  Figured I could map it and use xcopy and/or robocopy to make scripting it easy.  I will just figure out how to do it in FileZilla.  

Thank you very much for all your efforts with this.  I returned the camera at one point because of its lack of ftp support.  This makes it worth having.

shadow-1, 2017-04-24:
@jinx10000 
For an automated script, I don't think xcopy or robocopy would work too well as they lack ftp support. The built in ftp command line program that comes with windows doesn't support recursive downloads  (can only download one file at a time rather than a whole folder).

So I think you will need to look into a third party ftp tool. For scripting purposes, a command line tool which doesn't require user intervention is probably best. FileZilla is a very nice tool. However it is a program with a graphical interface which doesn't suit automated scripts.

jinx10000, 2017-04-24:
I'm not too bad at researching options.  I will find a way...  Can't guarantee that it will be the best way but I'll get it done.

Thank you again for all your help.

jinx10000, 2017-04-24:
@shadow-1
Have one more question for you.  Found a couple ways to sync but I am noticing that all of my files have a timestamp that is one month old in both WinSCP and FileZilla.  Any thoughts on what may be causing this?  Yi Home is showing the correct date.

shadow-1, 2017-04-24:
@jinx10000 
To be honest, I never used the recorded files so I am unaware of the differences between timestamps.
I know the Xiaomi software synchronises the date/time with the Xiaomi server. However it only uses this synchronised time for their software and don't synchronise the date/time on the Operating System itself.

To check this. Run the command `date` through telnet and see if that aligns with the timestamp given in FileZilla / WinSCP.

jinx10000, 2017-04-25:
The date using that command is correct.  The time stamp on the videos is one month ago.  Very strange.  I'll figure something else out.  Right now, I delete everything older than 7 days in the place I was going to put it.  It was deleting them as fast as I was downloading them from the camera.  Took me a bit to realize what was going on.  I was watching it download and then finding an empty folder... 

I'll just move it and make that folder delete after one month and 7 days...   Or something similar. 

Thanks for your help with this.  Really appreciate it.

shadow-1, 2017-04-25:
@jinx10000 
A little bit of testing might be required.

Like I said, I don't normally use this feature. However I enabled it just to check and I can confirm that on my camera the timestamp in FileZilla appears to be spot on.

jinx10000, 2017-04-25:
Very strange.  I'll have to play with it more.  The folders had the right date name, the date function has the right date, yours has the right date.  Weird stuff.  I'm not home now but i will play with it more Wednesday when i get home.   

The only thing I can think of is I may have plugged it in very shortly before each video I caught.  I don't have it on when I'm home.  The motion freaks out my son for some reason.

jinx10000, 2017-04-27:
This issue is definitely on my side and specific to the PC.

I pulled another computer out and all the dates look good in winscp ...  I removed Filezilla and winscp but the problem still remains.  

I'm going to clear anything related in registry and by searching file structure.  Not sure what could cause this...   I'm also using an older version of winscp on the other computer.  May try that but this seems to be something shared between Filezilla and winscp.  Maybe some crazy setting I changed when playing with sync...

jinx10000, 2017-04-27:
Looks like WinSCP up to 5.5.6 preserves timestamp correctly (gmt) .  It is one month ahead up to 5.7.7 and one month behind on 5.8.x up to current.  Not sure if that's just me.

shadow-1, 2017-04-29:
@jinx10000 
This sounds very strange! From what I can gather, the timestamp on the camera is correct. However depending on the program that is currently running on your PC, the timestamp is shown differently.

Perhaps there is a setting in WinSCP to change the way the date/time is displayed? You also said that the same issue occurs in FileZilla?

I tested on WinSCP 5.9.2 which I already had on my computer and the timestamp is correct.
Are you sure your regional settings within Windows is setup correctly on that PC?

jinx10000, 2017-04-30:
I tried filezilla on a different PC and the dates are still off.  I checked the timezone information and everything is correct on both.  
I got it syncing like I want it to using the older version of winscp.  It's just curious.  Sounds like it's something on my side though.

jinx10000, 2017-05-04:
@shadow-1
I honestly don't think this matters but I have the white international 1080p version.  Is it possible that this may be the cause?

Either way, last question and I'm good to close the issue.  It's working with an older version of WinScp and doing everything I need.  I'm really just curious at this point.  

I do have an unrelated question that is not an issue.  I don't have a lot of experience with bash or Linux.  However, as i said I'm pretty good at finding ways to make things work.  Would it be possible to write a script that searches for mp4 files and offloads to another location, from the camera side?  Easy, moderate, difficult, extremely difficult for someone with experience? 

To be clear, I'm not asking you to write it.  Just curious if it's possible before I start doing the research.

jinx10000, 2017-05-04:
@shadow-1
I would like to dump the files directly to my router ftp usb drive and bypass the PC.

shadow-1, 2017-05-05:
@jinx10000 
Most people don't realise this, however the same firmware is run across all versions of the camera.

So you would like the camera to send the files to your router? Or would you prefer your router to fetch the files from the camera?

Either way the script you will write will almost be identical. To do this on the camera itself would probably be a little more difficult as I only included a ftp server and no ftp client with the firmware. So with the current firmware, you would have to cross-compile a ftp client. This would probably be moderately difficult. I also haven't tried to setup cron jobs on the camera (scheduled tasks). It may not be enabled in the firmware at the moment. I can probably help you with this as it would be quite easy for me to enable these features by recompiling busybox and packaging it in a custom firmware.

To run the script on the router may be easier than on the camera depending upon on your router. Your router may already come with a ftp client and if you have a router that is capable of running a custom firmware such as Tomato, DD-WRT, Gargoyle etc. There are probably already packages or addons you can install without compiling anything, so you would only need to write a script and schedule it to run regularly.

jinx10000, 2017-05-05:
@shadow-1
Interesting.  I didn't know they all ran the same firmware.   So that's definitely not the issue....  I'm good there though.  Long as it works, I'm fine.  Sometimes curiosity will make me dig deeper than i need to but I'm done for now. 

I have an RT-AC68U running ASUSWRT-Merlin. I believe it's loosely based on tomato.  

I never even thought about trying it from the router to the camera...  

I know it has a ftp server but i don't think it has a client.  

But you are absolutely right.   Someone has probably already compiled an ftp client for it.  I'll look into it from that side to start.  Thank you!

shadow-1, 2017-05-05:
@jinx10000 
The Asus rt-ac68u is a very good router. With a quick Google search I found the following:
https://github.com/RMerl/asuswrt-merlin/tree/master/release/src/router/ftpclient

So it looks like a ftp client has already been compiled for the rt-ac68u.

I would say it would be easier to write a script on your router to fetch the files from the camera on a periodic basis rather than the other way around.

jinx10000, 2017-05-05:
I didn't see that one was already compiled directly in the firmware but I did see there are several available via optware after you mentioned doing it from the router side.  I honestly just didn't think about doing it from the router side.  

I saw one of the ftp clients on optware (ftpcopy) may make it easier.   Description: A simple FTP client written to copy files or directories (recursively) from a FTP server. 

There is also an ftp sync option available via the gui that I had never noticed before, but from what I read it doesn't support anonymous logins.  I haven't had time to try anything yet though.

I have written and/or edited a few basic scripts for the router so I have a general idea of what will be needed.  It's kind of weird that I didn't think about this honestly. 

Thanks again for all of your help and good luck with RTSP.  That will be a nice addition if it works out.

shadow-1, 2017-05-06:
@jinx10000 
Good news that you have found some good utilities for your router.
Just for your information, the latest version of yi-hack-v3 (0.1.3), I have changed the default behaviour of the ftp server. The ftp server uses the same credentials as telnet which by default is:
* Username: root
* Password: **_(blank)_**

jinx10000, 2017-05-13:
@shadow-1
I found a program lftp that seems to be working great.  For simple script was needed.  Had a lot on my plate so just got to it. 

Was trying to set rclone up to push it straight to Google Drive for added security but the memory on the rt-ac68u doesn't seem to be enough so that step still happens on the PC.  I'll probably just end up hiding a second router with a drive plugged in to store the video in a more secure local location in case a thief is smart and grabs the cameras, router, drives and PCs....

DJay79, 2017-05-13:
Hi there. Without any trouble and thanks to the good step-by-step instruction I installed the yi-hack-v3. But now I have problem to even get with ftp to the cam.
I'm on macOS using the App ForkLift and with my cam ip and and Port 21 user root and no password, there is an error Server response: '530 Login failed'.
So I tried the build in Finder ftp which prompts me the login but rejects also. Maybe they have a problem with a blank password field? Is there a way to change the password? Or maybe you have another hint for me.
Thanks in advance.

jinx10000, 2017-05-13:
@DJay79 
I changed the password before trying on the new version so don't know if it's your problem.  Telnet into it.  Login. Root/blank.  Type passwd.  Every new password twice.

shadow-1, 2017-05-14:
@DJay79 
I don't have a Mac to test ForkLift to see what the issue might be. I have tested in on Linux and on Windows and it works perfectly. Perhaps you can try another FTP client? As @jinx10000 stated, you can change the systems password by running the command `passwd` through Telnet.

@jinx10000 
As an alternative to a second router, I suggest looking into a embedded system such as the Raspberry Pi, Beaglebone, Odroid etc. These SoC boards will have more than enough power to do what you want. That is; sync the cameras recordings with an external drive and an online cloud service such as Google Drive, Dropbox etc.

Raspberry Pi has by far the best community support in my opinion which makes it a good choice.

jinx10000, 2017-05-14:
@shadow-1 

True.  That's a good idea.  I actually have a raspberry pie resetting my modem, router, and security cameras as needed.  Plan on having it reboot or reset power on the computer as well when it goes down.  Love it but still learning my way around it. 

I have another one I plan on using to control some other stuff and I'm pretty sure I can hide it better than the first one...  It probably will make a great secondary device to push backups to Google Drive and maybe even save a second copy on a hard drive in a better hidden place...   Didn't think of that. 

I may need a second router at some point either way though because of the number of devices I have.... Have 25+ connected all the time.  Figured it might be good to split the load.

DJay79, 2017-05-14:
Thank you guys. With now knowing what to do with Telnet I was able to set a password with command `passwd`. And now with a password also every FTP client here on macOS work. 👍🏻
@shadow-1 Once again thanks for your work. Very looking forward and excited to your ongoing development. Very appreciate.

shadow-1, 2017-05-30:
@jinx10000 @DJay79 
Good to hear that everything is now working.

closed 2017-05-30 by shadow-1

AndMu, 2017-11-01:
In anybody interested, this is utility to automatically download (if necessary compress)  files from camera.
https://github.com/AndMu/YiScanner/releases/tag/1.0.25

vip991988, 2018-01-27:
hi @shadow-1 , i'm using YI DOME 1080p , May you make it can backup to Mi Router Gen3 Hard Disk .
i can't backup to mi wifi wwith Yi Hack V3. Thank you so much !!!

burii89, 2018-04-10:
Hi, first of all, thanks to the author for the excellent work done on Yi_hack_V3! But, please help me by answering a couple of questions! Is it possible to change the FTP port, in the settings of the 1080p Yi Home camera? In what way could I take files from cameras via ftp, if the cameras on another network from the PC on which they need to be stored? Unfortunately I had almost no experience with FTP before, but I think it is quite possible to understand.

mrgoodfellow, 2019-02-03:
My preferred configuration would to have the camera reach out via sftp and push new files from any sub-directory in /tmp/sd/record  this way the camera can push files to a remote server regardless of the network it is currently connected to.

I'm working on setting up dropbear for the secure connection, any advice on how to get this configured or links to resources would be awesome!
Thank you for your help and contributions, I am very much enjoying this project.