shadow-1/yi-hack-v3 #14: Is here any script or API for getting the current frame of the camera?

cnberg, 2017-04-26:
I want to use telnet to login the camera, then take a snapshot and use ftp to pull back to my home assistant.

Is there any way to do this?

Thanks.

shadow-1, 2017-04-26:
At the moment there is no facility to get trigger video or image capture through telnet. The official Xiaomi applications running on the camera are proprietary and don't provide an API.

At the moment I am focusing on making it easier to setup ProxyChains-ng as it is not that easy for those that are not familiar with online proxy servers and RTSP streaming support.

---

If you are a developer or have an interest in programming, you can help develop this feature. On my Box account, I have hosted the SDK for the chipset used on this camera. It has a lot of documentation and the tools required to compile applications for the camera. However the documentation and examples provided are quite low level.

https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

cnberg, 2017-04-29:
Thanks for your reply.

Since the firmware of my camera is the latest one, I can only use your hack
right now. I am trying to downgrade now (to get full access via
yi-hack-1080)  :)

--------------
Yours sincerely
berg
http://cnberg.com
@cnberg

closed 2017-04-29 by cnberg

shadow-1, 2017-04-29:
@cnberg 
Since creating version 0.1.3 of the firmware, I have significantly updated the documentation on the main page and included a section on how to go back to stock firmware. If you follow these instructions, you will downgrade your firmware to 2.0.0.1A_201612051401. Link to the relevant section in the readme is below:
https://github.com/shadow-1/yi-hack-v3#going-back-to-stock-firmware