shadow-1/yi-hack-v3 #15: RTSP Server

jodupuis, 2017-05-01:
Great job guys with these projects, very impressive.
I just bought a new Yi 1080 Dome thinking the V1 hack could get me into the RTSP server to leverage with my BlueIris software, then when I was installing the new new Hack (v3), I realized it's not available.

Before I return this camera, just want to check with you if there's any plan soon to get RTSP?

Thanks again

John.

shadow-1, 2017-05-02:
@jodupuis 
I am definitely planning to develop RTSP support for this camera.
However at this stage it is a little hard to say exactly when support will be added to the firmware.

jodupuis, 2017-05-02:
thank you shadow-1 for the quick response.

DJay79, 2017-05-06:
Hi and first of all thanks @shadow-1 for investing your time in this and your passionated work. 
I'm an absolutely newbie and more or less just a passionated user who is doing handicraft work with some fantastic step-by-step guidance like yours is.
So I did with the **Homebridge Project** [https://github.com/nfarina/homebridge](url) which works great with my Apple HomeKit environment. I figured out that RTSP is a must have to get a camera work with i.e. a plugin for homebridge like [https://github.com/KhaosT/homebridge-camera-ffmpeg](url)
So I'm very excited and looking forward to your ongoing development with RTSP support.
Tough I have some questions...
- Recently there was an Firmware update on Yi Dome 1080p with V1.9.2.0E_201705031503.
Any problems to expect when I go back to "older" hack version? 
- Is there at the moment a way to switch the Yi Dome 1080p on/off by a http command? It's just because with this I already could implement the camera in homebridge with a switch plugin. [https://github.com/rudders/homebridge-http](url)

Once again thanks for great work and sharing.
DJay

shadow-1, 2017-05-06:
@DJay79 
To answer your questions:
* There are no problems going back to the "older" hack version based on 1.9.2.0D_201612191901. For the next release of the hack, I will most likely upgrade the base version to 1.9.2.0E_201705031503 for the Yi 1080p Dome.
* At the moment, the only way to view and control the camera is through the official smartphone app. Once RTSP support is developed, we can look into implementing extra features such as the ones you are suggesting.

DJay79, 2017-05-06:
Thanks for keeping my suggestion in mind. 👍🏻

xX-Nexus-Xx, 2017-06-05:
Great work .. I'm as well looking forward to the RTSP implementation.

Is there a chance of having a work-around by putting the last event-recording (e.g. motion triggered) to be copied into a dedicated folder with a dedicated name ?

This would enable us to establish a permanent link in a browser or app to load that recording via FTP or http request!

thanks for considering

shadow-1, 2017-06-05:
@xX-Nexus-Xx 
These Xiaomi cameras feature last-event recording to the microSD card (configured through the Xiaomi app).

These recording are saved in a folder named `record`. A sub-folder is created for each hour of recording. the folder name has the year, month, day and hour. For example `2017Y06M05D10H`. Within this folder, 1 minute mp4 video files are copied and the filename indicates which minute was recorded. For example `01M00S.mp4`.

Another Github member @jinx10000 has developed a script which periodically runs on a router / raspberry pi and syncs these recording with a NAS and potentially Google Drive. Have a look at this thread:
https://github.com/shadow-1/yi-hack-v3/issues/13

Is this something like what you are after?

xX-Nexus-Xx, 2017-06-05:
Thanks for the quick reply !

I was aware that the app is recording to MicroSD but could not find the location and thought it was encrypted!

This is almost what I was looking for. I will try to create a FTP-request via HTTP to display the latest recording in a browser or app (like kodi)

jodupuis, 2017-06-26:
Hello Shadow-1, been following all these threads, what a great job! I'm sorry to bug you again on this but do you have a better idea as to when you might have a RTSP version?

shadow-1, 2017-06-28:
@jodupuis 
I have just released the new firmware which should greatly simply the configuration of ProxyChains-ng for most people. The next big thing on my radar is RTSP support. I have some ideas on how to develop the feature. Unfortunately I am not sure when I will be able to finish developing it as I am quite limited in time. I am developing this firmware as a hobby in my spare time.

jodupuis, 2017-06-28:
Hello shadow-1, no worry, we all do understand and very appreciative of all the great work you do here, no kidding!! I wish there will be a donation option somewhere. Keep up the excellent work!

jodupuis, 2017-08-07:
Hello guys, just realized this thread has been closed and looking through it and other threads, I can't find out if this RTSP piece was successful. A lot of great work went into this project, very impressive. 

I upgraded the cam to Yack-3, had to perform a reset to get it back to work. Could not find this RTSP server in it. 

Any guidance?

Shahow-1, thank you so much for everything!!
Note: I'm using the BlueIris SW to monitor multiple cameras, the generic version (if not in the list of supported camera) is to leverage RTSP.

xmflsct, 2017-08-08:
@jodupuis I assume full RTSP support is not ready yet. I am also waiting for @shadow-1's great work on it!

sweup, 2017-08-14:
Thanks for your hard work. Also hoping to see rtsp support

galadril, 2017-08-16:
Would love to see RTSP support work and integrate my cams with my domoticz system 👍

wrongbongo, 2017-08-21:
Shadow, please setup a bounty and/or donation box. I and assume others would gladly throw money at you for RTSP. On another note, let me know if you need anything as far as hardware, dumps from serial connectivity, etc from a 27US ANTS.

lastb0isct, 2017-09-03:
Any progress on this?

shadow-1, 2017-09-07:
@lastb0isct 
I have been quite busy lately. However it will be coming.

ericboehs, 2017-09-07:
Wahoo. I just bought two Yi Home (US27) and I'm looking forward to the RTSP server (only reason I want to hack it really).

Thanks for your work on this @shadow-1. I'd gladly donate $10 if you set up a donation box (e.g. patreon.com).

uavana, 2017-09-21:
RTSP server is the only reason I want to hack it really +1

xvolte, 2017-09-23:
+1, i got 3 cameras on yi hack v1. Sadly the last one, was on the new firmware 1.8.7, so i finally managed to install yi hack v3 on this new camera, however, there is no rtsp yet so can't use it for my needs.
In addition, because we don't know what is still going through Xiaomi/Yi, it would be cool if you could include a Iptables firewall as well.
Also, i managed in yi hack v1 to change the init script, so that i don't get all the files written on the SD card, i remove the connection to cloud services, etc ... in your hack, it would be SOOOOOO great if we could just disable everything, except the rtsp. i only need the wifi and the rtsp field. I don't need the rest, and would like it to be disabled.
I noticed the configuration file, so i disable FTP and proxy, but i would like to restrict network to local network only (192.168.X.X), and also to remove automated recording on the sd card + enable RTSP + Iptable.
I can help as i know how to code. the only problem is that i don't know how to work with a non standard linux (busybox thing ...)

Anyway, iptable + RTSP would definitely be great!

nikooo777, 2017-10-17:
hey guys!
I'm just going to jump on the hype train of RTSP :) 

With huge respect for all the people involved in the project, and with all the gratitude for their time, I'll be very interested in seeing this feature available.

I well know what working on a project on your free time means, so I'm not asking for it nor pretending anything, just adding my own to the list of people that would be more than happy to see this happening.

Best regards and happy coding!

Niko

wtstreetglow, 2017-11-18:
+1 Bring on the RTSP server and the donation box. Would gladly donate as others have also agreed to support the cause for the RTSP.

gitmoplex, 2017-12-03:
+1 TAKE MY MONEY,,,, please 

will work for RTSP

wrongbongo, 2017-12-05:
Alright all you "+1" people. Time to put up or shut up. https://www.bountysource.com/issues/44753535-rtsp-server

c0mm0n, 2017-12-05:
Paypal won't accept payment. Maintenance or whatever :D

gitmoplex, 2017-12-06:
PAID !!

gitmoplex, 2017-12-07:
ok... donation is open @ https://www.bountysource.com/issues/44753535-rtsp-server

Results so far ,, are a little bit underwhelming.

@shadow-1 
we all love your work, and is there anything we can do to help with the projectz  ?

galadril, 2017-12-07:
Paid!.... (gitmoplex, only 5$ ?? ;-) )

galadril, 2017-12-07:
@ all =1 posters >> Do you now the phrase:

**_Origin of the Phrase_**

According to Christine Ammer, The Facts on File Dictionary of Clichés, second edition (2006), the phrase has been around in the United States since approximately 1930:

## **

> put one's money where one's mouth

** is Back up your stated position with action. This term, according to Eric Partridge's informants, was current in the United States from at least 1930 and caught on in Great Britain and other English-speaking countries shortly after Wold War II. In 1975 the British government used it as an advertising slogan to persuade people to invest their savings in the National Savings Bank Accounts Department.

kruschman, 2017-12-07:
In for $15

gitmoplex, 2017-12-08:
@galadril 
Corrected the payment , now im on par again.

And for all who do support... YOU AREW AWESOME !!

galadril, 2017-12-08:
@gitmoplex i saw it ;) you rock!
Sorry for the pressure 👍

gitmoplex, 2017-12-08:
@galadril no pressure at all
I want this to succeed !!

;)

c0mm0n, 2017-12-10:
Paypal working now, 10$ paid.

ericboehs, 2017-12-17:
In for $15. Would love to see this working with WPA2!!

DJay79, 2017-12-18:
Hmm... Has @shadow-1 even replied to the honorably "funding"?? Of course we all would love to see RTSP working with Yi and a donation is of course no problem. But some information about how, for wich cameras and at least a time line would be more serious. 🤔

shadow-1, 2017-12-20:
@DJay79
I have had trouble getting a solid amount of time locked down to develop this firmware. Specifically RTSP support. This is part of the reason why I have not been actively asking for donations to promote development.

However RTSP support with the vision that I have will work with all supported cameras. They all essentially work exactly the same way. So one solution will fit all. The exception would be the original Yi Home as that camera is a bit different to all the others that I support (well will support).

A solution similar to the Fritz firmware is also achievable. However that would likely result in a separate version needing to be compiled for each type of camera. However a single application could be developed with only a compiler option being different depending for which camera you want to compile the application for.

In addition, this repository is completely open. I have included full source code. There are no proprietary elements like many other firmware projects. Anyone is welcome to pick up the source code and develop further features and/or improve existing features. I am willing to help anyone with development for these cameras.

seanmeads, 2018-01-04:
rtsp working on CN hardware:
http://xiaoyi.querex.be/instructions/
uses server (not lwsws)  and rtspsvr (requires  library libstdc++.so.6)

fabceolin, 2018-01-04:
Where can I get the libstdc++.so.6?

fabceolin, 2018-01-04:
Under 1.8.6.1B_rtspfix we have the files rtspsvrI, rtspsvrK and rtspsvrM. Is there a chance that some of them work on Yi Dome Camera? (I am downloading the SDK to generate the libstdc++.so.6 and try)

shadow-1, 2018-01-04:
@seanmeads
This implementation only works with the original Yi Home. It does not work with the newer cameras based on HiSilicon Hi3518eV200 chipset.

@fabceolin 
libstdc++.so.6 is available with the SDK which I host. From memory, it is roughly 5mb in size, so it will need to be stored on the microSD card.

kb5won, 2018-01-25:
I chipped in a little for the bounty.  I just got zoneminder up and running, and already have a Yi dome 1080, would love to be able to use it with zoneminder.   Thanks!

5k1ttl3, 2018-01-31:
@shadow-1 is it possible for me to get a copy of libstdc++.so.6? Also where can I send you money or other support? I've got 4 of these US27 cameras and they're all going back to Amazon if I can't get RTSP going :(

I for one am 100% fine dropping app support to get RTSP - I'd much prefer it. But I understand that's not the case for everyone

ontdamian, 2018-04-13:
Hi @shadow-1 ,
any progress on this issue? Any "valid" plans to develop RTSP for Yi Home 720p (HiSilicon Hi3518eV200 chipset)?
I think many of us are waiting for it, and don't know if should invest in other models to have RTSP functionality.

thanhnhiel, 2018-04-30:
@shadow-1 I am interested. If you don't have more time, I can do this. Could you can share sdk and document for developing or some idea. I will implement them. Thank you.
Now I have camera Yi Home (720) global (42US).

Sigri44, 2018-05-13:
:(

Sigri44, 2018-05-13:
='(

nico81, 2018-06-01:
+1 for RTSP and Homebridge support!

lance36, 2018-06-12:
RTSP! take my money!

bit111, 2018-06-30:
Hello,
I don't understand if RTSP for Yi Dome Camera 1080 was implemented or not. If it was implemented, how can I have  it?

Sigri44, 2018-07-01:
Can't, it's a SCAM :)

bit111, 2018-07-01:
What do you mean with SCAM?
Was the rtsp feature development abandoned?

Sigri44, 2018-07-01:
No, he have launch a kickstarter last year, and he is "lost" now. :)

batrarobin, 2018-07-01:
@Sigri44

What rubbish!  I know shadow-1 personally and I know he has been busy lately with his work and personal life. At this time he is not being able to return to this project due to other priorities. I see that some other people have recently started to carry on with this project so hopefully they would have better chance of accomplishing the "Mission RSTP".

wrongbongo, 2018-07-02:
To be clear, a Kickstarter != a bounty. Please keep this thread clear of conjecture. Anyone capable is more than welcome to the money providing a solution.

Sigri44, 2018-07-02:
So Why he have choice to have many money quickly since last year ? I can ask for it when your projet is finish, but not before if you don't have the time..

dmnbi4, 2018-12-25:
+ Follow

TheCrypt0, 2019-03-12:
@shadow-1 

I've finally finished my RTSP implementation, it works nicely with Home Assistant, Shinobi and other DVRs.
It is available on the Discord for the supporters of the new project.

I'd like to reward all the people that have put their support on the [Bounty](https://www.bountysource.com/issues/44753535-rtsp-server) by giving them full access to the RTSP on Discord.

Is it possible to close the issue now? Without that, Bountysource cannot mark it as solved.

Thanks,
Crypto

----
**Here's the BountySource message:**

> 
> Hi, I've been working on a functional RTSP implementation for the past 3 months. 2 weeks ago I published on Discord the first working closed-beta, available to supporters only. Since then I fixed some of issues with Xiaomi's H264 encoder that happened to hang ffmpeg of Shinobi and Home Assistant. Now everything works as it should and the app functionalities are intact (but they can be disabled if you wish).
> 
> If you have supported this bounty join the Discord server (https://discord.gg/upPsFWZ) and send me a private message with your username, I will give you access to the private #supporters channel where you will find the download link and all the instructions on how to install and use the server.
> 
> Thank you,
> Crypto

_Tagging all the people that joined the interest of a working RTSP_

@jodupuis @DJay-X @xX-Nexus-Xx @xmflsct @sweup @galadril @wrongbongo @c0mm0n @sebastienb @lastb0isct @ericboehs @uavana @xvolte @Robot72 @nikooo777 @markusd1984 @wawryk @cinoque5 @proautodinc @wtstreetglow @jelvs @mblokland @gitmoplex @SeydX @kruschman @seanmeads @fabceolin @kb5won @5k1ttl3 @ontdamian @Sigri44 @nico81 @lance36 @bit111 @batrarobin @dmnbi4

shadow-1, 2019-03-13:
@TheCrypt0 
I have had a look on the Discord server and it appears that RTSP solution is working. Even if it is beta/alpha stage, I believe this issue can be closed.

PS. If at all possible, I would be interested in trying this RTSP implementation at some stage. 😄

[8 more replies]

closed 2019-03-13 by shadow-1