shadow-1/yi-hack-v3 #16: Binding success but not camera added in main menu

MatChung, 2017-05-02:
Thx for the hack. But I have a problem that I can not add the camera.

ProxyChains-ng is configured and I sure that the binding has completed successfully, but after returning to main screen of App, there is not camera shown.

Is there any help on this?

Thx.

[image]

shadow-1, 2017-05-03:
@MatChung 
This is a bit of an unusual problem. Most people have issues finding a working proxy server. Once a proxy server is found, everything tends to work without issues.

The problem you are having is an issue with the smartphone app itself.

A few things you can try:
* Try resetting and pair/bind the camera to your phone again.
* Make sure you have updated your smartphone app to the latest version.

shadow-1, 2017-06-29:
@MatChung 
I have found out that this issue may be caused by having an old version of the smartphone app installed.
Also try the latest version of the hack as it has made configuring proxies much easier.
Feel free to open a new issue if you are still having issues.

closed 2017-06-29 by shadow-1