shadow-1/yi-hack-v3 #17: How and where to put IP address

Ahmadbe, 2017-05-05:
Dear Sir,
Thank you for good job is rooting my yi camera Chines V.
I had flash the toot and Home file on the camera. but I did not understand where and how to put the Ip address. and how to find IP that work.

Thank you

batrarobin, 2017-05-05:
@Ahmadbe
Did you even read this:

https://github.com/shadow-1/yi-hack-v3/blob/master/README.md#getting-started---step-by-step-guide

Ahmadbe, 2017-05-05:
Yes Sir, I've read it... but to be honest I did not understand Where to put the IP... Is it under [proxy list]
and how the phrase should be.. can you give me example plz.
and where can I find those proxies.

Thanks

https://github.com/shadow-1/yi-hack-v3/blob/master/README.md#getting-started---step-by-step-guide

shadow-1, 2017-05-05:
@Ahmadbe 
Step 15 of the step-by-step guide that @batrarobin linked to states that the proxy servers go underneath `[proxy list]` and also gives examples on how it should look like:
```
socks5 183.232.25.100 3080
socks5 27.152.181.217 80
socks5 125.67.236.195 8080
socks4 115.29.192.194 1080
```
_Please be aware that the proxy servers above are only examples and may not actually be working._

There are many websites online that have proxies listed. A few things to keep in mind:
* You must test that the proxy server works. Many proxy servers listed online may have been shutdown and not been removed yet.
* The proxy server must be located in Mainland China. If it is not, you will get the annoying ban message "This camera can only be used within China."
* Look for socks4 and socks5 proxy servers as they are the most likely to work properly. Majority of them support SSL connections which is required.

Ahmadbe, 2017-05-05:
I tried so many proxies, but I can't connect the cam to my android phone.

I think I'm missing something.

I've tried to make all the steps and download the yi chines app from your link, change the proxy on the camera and try to connect it to wifi..... the cam connecting successful but can't connect to my my android phone.

shadow-1, 2017-05-06:
Duplicate thread.

closed 2017-05-06 by shadow-1