shadow-1/yi-hack-v3 #18: connecting to android app

Ahmadbe, 2017-05-06:
Dear Sir,
The camera connected to the Wifi successfully but I can't connect it to the  android app.( I installed the android app from the link u gave). It gave me (Fail To load, Please reset the camera).
I tried to connect it with other phone but it didn't work.
any solution please!

batrarobin, 2017-05-06:
@Ahmadbe

The proxy that you are using might not be working. Did you click "Test Config" and see if that passes?

shadow-1, 2017-05-06:
@Ahmadbe 
For future reference, please do not open multiple threads for the same issue that you are having.
I have now closed your other issue as it is a duplicate.

---

As @batrarobin stated. The most likely issue you are having is that the proxy servers which you have entered are not working. Please post a screen shot of the "Test Config" page from the web interface.

Ahmadbe, 2017-05-07:
I've tried more than proxies but no one all gave me "Proxy Server Connection Failed. Ensure that ProxyChains-ng has been configured.

batrarobin, 2017-05-07:
@Ahmadbe

Try this one:

socks5 115.236.100.104 80

Ahmadbe, 2017-05-07:
This is the last Proxy that I tried, It gave me
Proxy Server Connection Failed. Ensure that ProxyChains-ng has been configured.
The camera connected successfully but can not connect to the app.

shadow-1, 2017-05-07:
@Ahmadbe 
I have tested the proxy server that @batrarobin provided and can confirm that it is working properly.
The "Test Config" page is not perfect, in future updates I will improve it. Please try/confirm the following:
1. Delete all configured proxy servers under `[ProxyList]`. Having a big list of proxy servers is not going to help if many of them do not work.
2. Add the proxy server that @batrarobin provided.
3. Run the "Test Config" page. If it fails, retry a few times. Sometimes the proxy server is a little slow and the test may fail occasionally.
4. If it is still not working, I suspect that this particular proxy server may be blocked in your country. You will need to find a proxy server that works in your area.

Ahmadbe, 2017-05-08:
Thank you so much for your help, I've tried everything you notice but non of them works.
I will wait for the hack-v4... maybe it will work on my case.
Thanks again for your time.

ken007892004, 2017-05-25:
Hi shadow-1,

I have the same issue with Ahmadbe, WiFi is connected and the camera is successfully paired but at the app, it says "Fail To load, Please reset the camera".

> 
> 
> System startup
> hello, open console
> 
> do_auto_sd_update: detect flash size: 0x1000000, gsdexist=0
>
> do_auto_update: detect flash size: 0x1000000
>### UBOOT:test pin high! ###
>Hit any key to stop autoboot:  0
>16384 KiB hi_fmc at 0:0 is now current device
>
>## Booting kernel from Legacy Image at 82000000 ...
>   Image Name:   7518-hi3518c-kernel
>   Image Type:   ARM Linux Kernel Image (uncompressed)
>   Data Size:    1384232 Bytes = 1.3 MiB
>   Load Address: 80008000
>   Entry Point:  80008000
>   Loading Kernel Image ... OK
>OK
>
>Starting kernel ...
>
>Uncompressing Linux... done, booting the kernel.
>Booting Linux on physical CPU 0
>Linux version 3.4.35 (chenshibo@ANTS-SH-SV02) (gcc version 4.8.3 20131202 (prerelease)(Hisilicon_v300) ) #22 Mon Nov 28 15:39:11 CST 2016
>CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
>CPU: VIVT data cache, VIVT instruction cache
>Machine: hi3518ev200
>Memory policy: ECC disabled, Data cache writeback
>Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8636
>Kernel command line: mem=34M console=ttyAMA0,115200 root=/dev/mtdblock4 rootfstype=jffs2 mtdparts=hi_sfc:256k(boot)ro,64k(env),64k(conf),1600k(os),1280k(rootfs),12992k(home),64k(vd1),64k(ver)!!!
>PID hash table entries: 256 (order: -2, 1024 bytes)
>Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
>Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
>Memory: 34MB = 34MB total
>Memory: 30648k/30648k available, 4168k reserved, 0K highmem
>Virtual kernel memory layout:
>    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
>    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
>    vmalloc : 0xc2800000 - 0xff000000   ( 968 MB)
>    lowmem  : 0xc0000000 - 0xc2200000   (  34 MB)
>    modules : 0xbf000000 - 0xc0000000   (  16 MB)
>      .text : 0xc0008000 - 0xc0340000   (3296 kB)
>      .init : 0xc0340000 - 0xc035cc64   ( 116 kB)
>      .data : 0xc035e000 - 0xc037f260   ( 133 kB)
>       .bss : 0xc037f284 - 0xc03b1850   ( 202 kB)
>SLUB: Genslabs=13, HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
>NR_IRQS:32
>VIC @fe0d0000: id 0x00641190, vendor 0x41
>sched_clock: 32 bits at 49MHz, resolution 20ns, wraps every 86767ms
>Console: colour dummy device 80x30
>Calibrating delay loop... 269.10 BogoMIPS (lpj=1345536)
>pid_max: default: 32768 minimum: 301
>Mount-cache hash table entries: 512
>Initializing cgroup subsys freezer
>CPU: Testing write buffer coherency: ok
>Setting up static identity map for 0x8026ab08 - 0x8026ab60
>dummy:
>NET: Registered protocol family 16
>Serial: AMBA PL011 UART driver
>uart:0: ttyAMA0 at MMIO 0x20080000 (irq = 5) is a PL011 rev2
>console [ttyAMA0] enabled
>uart:1: ttyAMA1 at MMIO 0x20090000 (irq = 30) is a PL011 rev2
>uart:2: ttyAMA2 at MMIO 0x200a0000 (irq = 25) is a PL011 rev2
>bio: create slab <bio-0> at 0
[... 349 lines cut ...]
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>motion:1, rect:0-0-320-192
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>g_dispatch_info.mmap_info->ptz_x_runtime =0 2
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>motion:1, rect:0-0-320-192
>
>Login timed out after 60 seconds
>Auto login as root ...
>(none) login: p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>
>p2p_tnp.c(tnp_proc-5289)[01:46:32.597]:key(ZLu3WQoq6mrJlkXQ)
>
>p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
>g_dispatch_info.mmap_info->ptz_x_runtime =0 2
>
>Login timed out after 60 seconds
>Auto login as root ...
>(none) login: p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)

shadow-1, 2017-05-25:
@ken007892004 
Hmm, I have never had this issue.
Thanks for sharing the log as it does give some clues what the issue may be.

From your log it looks like the following proxy servers are not working:
* 27.152.181.217:80
* 125.67.236.195:8080

You can try removing these proxy servers from ProxyChains-ng configuration or commenting it out by starting the line with the '#' character.

The current default configuration chooses a proxy server at random from the list within the configuration file for each connection. It appears that some of your connections are timing out (most likely due to the proxy server not working). When pairing, perhaps too many connections are timing out.

Another thing I suggest is to update to the latest version of Yi Home on your smartphone.

ken007892004, 2017-05-25:
@shadow-1 

I have the latest app. I tested with 5 cameras, only 2 work. It's rather strange :(

Thank you for the crystal clear explanation, I hope you can find the problem soon if you have the time, if you need anything else in assisting in your investigation, you can leave a message here.

Thank you so much for the efforts you made here :)

shadow-1, 2017-05-25:
@ken007892004 
To help me troubleshoot. Can you leave only one proxy server that is working and try again.

Can you also post your start-up log when you only have one working proxy server in the list.

At least that will get rid of the timeout errors and maybe we will be able to notice something else that looks strange.

ken007892004, 2017-05-26:
@shadow-1 

Log for first boot after reset and setting one working proxy server only :)
`

System startup
hello, open console

 do_auto_sd_update: detect flash size: 0x1000000, gsdexist=1
Interface:  MMC
  Device 0: Vendor: Man 035344 Snr 22ab38e7 Rev: 8.0 Prod: SS08G
            Type: Removable Hard Disk
            Capacity: 7580.0 MB = 7.4 GB (15523840 x 512)
Partition 1: Filesystem: FAT32 "NO NAME    "
            record/
    48128   log_first_login.tar.gz
    50688   log_bind_success.tar.gz
            system volume information/
   956808   rootfs_y20
    59904   log_start_reset.tar.gz
    37376   log_wifi_connected.tar.gz
  7352056   home_y20

6 file(s), 2 dir(s)

1: log_first_login.tar.gz!
2: log_bind_success.tar.gz!
3: rootfs_y20!
 find index 4: rootfs_y20

4: log_start_reset.tar.gz!
5: log_wifi_connected.tar.gz!
6: home_y20!
 find index 5: home_y20

u-boot_y20 not found!

env_y20 not found!

conf_y20 not found!

kernel_y20 not found!
reading rootfs_y20

magic 0x27051956 0x27051956
arch 0x2 0x2
size 0xe9948 0x40
type 0x7 0x2
crc 0x781fd023
name 0.1.3-hi3518-rootfs
idx4: old : 58fe3daf, new : 58fe3daf
Image rootfs_y20 is same time
reading home_y20

magic 0x27051956 0x27051956
arch 0x2 0x2
size 0x702eb8 0x40
type 0x7 0x2
crc 0x301cb58c
name 0.1.3-hi3518-home
idx5: old : 58fe3ff6, new : 58fe3ff6
Image home_y20 is same time

 do_auto_update: detect flash size: 0x1000000
### UBOOT:test pin high! ###
Hit any key to stop autoboot:  0
16384 KiB hi_fmc at 0:0 is now current device

## Booting kernel from Legacy Image at 82000000 ...
   Image Name:   7518-hi3518c-kernel
[... 462 lines cut ...]
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  118.193.132.113:1080  ...  139.129.76.123:443  ...  OK
trans_json_ex result=20000
motion:1, rect:0-0-320-192
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
motion:1, rect:0-0-320-192
motion:1, rect:0-0-320-192
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)

Login timed out after 60 seconds
Auto login as root ...
(none) login:
p2p_tnp.c(tnp_proc-5289)[02:45:31.575]:key(oNFrgLhH8nzNkJpS)

p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
motion:1, rect:0-0-320-192
p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
motion:1, rect:0-0-320-192
g_dispatch_info.mmap_info->ptz_x_runtime =0 2

p2p_tnp.c(tnp_proc-5289)[02:45:51.683]:key(oNFrgLhH8nzNkJpS)

p2p cur user(0) max_user_num(0) in_packet_loss(0) out_packet_loss(0)
motion:1, rect:0-0-320-192
`

shadow-1, 2017-05-26:
@ken007892004 
From your log, it looks like the proxy server that has been configured is still having trouble connecting half of the time.
I have tested the same proxy server on my camera and I am having the same issue, i get `socket error or timeout!` about 2-3 times out of 4. This may be causing your issues when pairing. Although it probably won't matter once it is paired.

Can you try another proxy server. Try this one as it is working properly for me:
`socks4 115.29.161.167 1080`

Can you post your log again with the above proxy server configured.

shadow-1, 2017-06-29:
@ken007892004 
I'm not sure whether your issue has been resolved or not.
However if you are having trouble with proxy servers, please try the latest version of the hack which has made it much easier to configure proxy servers.

Ahmadbe, 2017-06-30:
Thanks for the proxy new update, I've tried it and it worked. but still, can't connect it with Android phone :(

shadow-1, 2017-07-01:
@Ahmadbe 
It's good to hear that you found proxy configuration much easier.
If you can not connect to Android phone, there are two things to check:
1. _Are you sure you are using the correct Yi Home app for Android?_

You must use the Chinese version of the app if you have the Chinese version of the camera.

2. _Are you using the latest version of Yi Home app for Android?_

Some people have had issues with older versions of the app. The older versions no longer works and what is worse, the update feature on the older version no longer works for some reason. On my Box account, I am now hosting the APK file for the latest version of Yi Home (Chinese version). It is available here under the directory `Yi Home - Android --> 2.10.2.19_20170512`:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

Ahmadbe, 2017-07-01:
I've used the correct app, from your box.
Is it possible that I have something wronge with my account?

shadow-1, 2017-07-01:
@Ahmadbe 
Can you post a screenshot of your "Test Proxy List" page when it has finished completing the test.

Ahmadbe, 2017-07-01:
My Friend,
Thank you so much for your help, the camera connected successfully, you are the best :D

shadow-1, 2017-07-02:
@Ahmadbe 
Glad that everything is working now!

closed 2017-07-02 by shadow-1