shadow-1/yi-hack-v3 #19: FTP server and password

comicchang, 2017-05-17:
Hello, 

Is there any way to add password authentication to the ftp server?

thank you

batrarobin, 2017-05-17:
@comicchang

This has already been answered here:

https://github.com/shadow-1/yi-hack-v3/issues/13#issuecomment-301286176

comicchang, 2017-05-17:
@batrarobin I read that issue already, but ... after I set password for root via telnet, FTP is still accessible by anonymous account and empty password

shadow-1, 2017-05-17:
@comicchang 
This feature was implemented in the latest version of yi-hack-v3.
When you update yi-hack-v3 to version 0.1.3, anonymous ftp access is disabled. Ftp is accessible using the same username and password as the system (telnet).

The latest release of yi-hack-v3 is available from:
https://github.com/shadow-1/yi-hack-v3/releases

comicchang, 2017-05-17:
@shadow-1 

 I just found out I'm using v0.1.2, will upgrade later

thank you

closed 2017-05-17 by comicchang