shadow-1/yi-hack-v3 #20: Files not being read

kohorok, 2017-05-21:
Dear Shadow,

Thanks for all your work, I am truly impressed.

However, I tried to follow your steps but, my Yi Dome didn't seem to read the files.

On your step 6. "The yellow light will come ON and flash for roughly 30 seconds, which means the firmware is being flashed successfully. The camera will boot up."

[image]

When I powered it on, it didn't have the symptom as you mentioned (30seconds). It get to the "Waiting for connect" state pretty quick. I have tried multiple times with countless of format to FAT32.

My device serial number is 13CNxxxxx
Mine is Yi Dome (1.9.1.0G_201701161401), I downloaded the firmware from http://www.xiaoyi.com/home/firmware/download/?version=3.

Hope to hear from you soon.

Cheers,
Sam

kohorok, 2017-05-21:
Hi Shadow,

I have also tried to download the official firmware that you shared on Appbox >Official Firmware > Yi Dome Camera > 1.9.1.0F_201701041701home_v200m (then I modified the name to only "home_v200"). It did re-flashed with the firmware in the SDcard. Since it only went into the "Waiting to connect" state after 3-5mins later, I know the flashed was completed.

However, again I tried to format my sdcard and put in these two files **rootfs_v201 & home_v201**, but still the device didn't seem to boot the files.

Need your advice.

Best Regards,
Sam

shadow-1, 2017-05-21:
@kohorok 
It is a little hard to determine what the problem may be.

From your screenshot it appears that you are using a SDXC card which means that you are using a high capacity card (>32gb). There are known issues with using high capacity cards and the same workarounds need to be performed like on other embedded boards like Raspberry Pi. How are you formatting your microSD card with FAT32? I know Windows does not allow it with the built-in utilities.

The simplest workaround is probably to borrow a low capacity card to flash the firmware (less than or equal to 16gb).

If you want to use a high capacity card to flash the firmware. I suggest formatting the card first with the official SD card formatting tool. You will most likely need to enable "FORMAT SIZE ADJUSTMENT" by setting it to "ON". The official SD card formatting tool is available from the following link:
https://www.sdcard.org/downloads/formatter_4/

Then you will need to format the SD card to FAT32 format. You can use the free utility fat32format. The Windows GUI version is available from the following link:
http://www.ridgecrop.demon.co.uk/index.htm?guiformat.htm

---

You can try flashing the appropriate recovery image for the Yi Dome before flashing yi-hack-v3 again. It is unlikely to help in your situation. However it might on the off chance that flashing the firmware got interrupted the first time and the same image is no longer being accepted by the camera as it thinks it has already been flashed. Recovery images are available in the following link:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/1/23494084746

---

Just for reference, the official firmware images work on an entirely different principle to the firmware I created. To flash the official firmware, you will need to rename the firmware image to home_v200m (rather than a pair of images named rootfs_v201 and home_v201). The official firmware should flash with exFAT formatted microSD cards.

kohorok, 2017-05-22:
@shadow-1 

Thanks for your reply.

I did the format to FAT32 using third party software (Easuse).

Anyway, I managed to resolved this by putting in your 0.1.2 and I noticed the flashing was working, then I tried again with 0.1.3 and it is correctly flashed now.

I can go into the web UI for proxychain config.
I spent more than an hour going through countless website which gives free china socks5 proxy, but still couldn't find one.

I feel that I only left with 1 step, but it's one of the hardest.

May I know, if the region check only happen during the first pairing? Or does it check every once a while?
What if the IP I put in the proxychain no longer working? Does it mean I have to keep looking for proxy which is working?

Best Regards,
Sam

kohorok, 2017-05-22:
@shadow-1 
Just to add in, I am using xiaomi phone and the yi home app on my phone was download from the appbox you shared
Under Yi Home - Android > com.ants360.yicamera.apk      Uploaded Apr 24, 2017  34.3 MB

Cheers,
Sam

shadow-1, 2017-05-22:
@kohorok 
Good to hear that you got the firmware working. Try this proxy server:
`socks4 115.29.161.167 1080`
This one is working for me.

The region check occurs on the camera upon boot. Once it has booted up and completed the region check, the camera does a region check only time to time. I am not sure exactly how often it does the check, however it appears to only occur after 3-4 days of inactivity.

If your Proxy IP no longer works , your camera will continue to operate as per usual until it next does the region check. For most people, this would be when the camera reboots. Luckily if the proxy server is not working, the camera does not play the annoying ban message. However it will not work either. The test config page can be used to check whether the configuration is still working.

I am currently working on a system to make it much easier to find and populate working proxy servers. I will release it with yi-hack-v3 v0.1.4. However it will take me a little bit of time to finish developing.

kohorok, 2017-05-22:
@shadow-1 
The proxy server works!
however as soon as I saw the page "Congratulations" and I click continue, it went back to the first page and forced me to update the software.

[image]

Actually I am seeing this after installing the software but the way I bypass this message is to quickly click add camera (the + sign on top).

After click exit and login app again, I was unable to find the camera that was previously successfully added.

Should I go ahead and update the app?

shadow-1, 2017-05-22:
@kohorok 
It is great to hear that the proxy server is working for you.
Yes it is safe to update the Yi Home app to the latest version.
The version I have hosted on my Box account is quite old now.

kohorok, 2017-05-22:
@shadow-1 
Amazing response! 
God bless you..

Let me try and I will update you with the result.

One last question, how did you manage to find a working proxy? :)

Best Regards,
Sam

shadow-1, 2017-05-22:
@kohorok 
I simply go to a few websites that host free proxy servers. The main thing is that you have to test whether they work and whether they are indeed based in Mainland China.

I know many proxy servers listed online are down. I can only assume that they were working at some stage, however the websites were not updated to show that they are down. I have also come across proxy servers described as 'China'. However when you test them, they are actually based in Hong Kong or Thailand. The other common issue I have come across is that the proxy server needs to support SSL connections for this hack to work. Many listed online do not. That is why i suggest to look for socks4 or socks5 right away as most of them support SSL connection.

I have had reasonable success with xroxy as a source.
http://www.xroxy.com/proxylist.php?port=&type=&ssl=ssl&country=CN&latency=&reliability=#table

kohorok, 2017-05-28:
Thank you so much @shadow-1 .
You have helped me resolved the issue.

Cheers,
Sam

shadow-1, 2017-05-30:
@kohorok 
Glad to hear that your issues have been resolved!

closed 2017-05-30 by shadow-1