shadow-1/yi-hack-v3 #21: V3 Software dont seems to be able to work..

Dp4610yz, 2017-05-26:


Dp4610yz, 2017-05-26:
using 1080P Black Dome..

shadow-1, 2017-05-26:
@Dp4610yz 
Can you please be more specific what your problem is.
Please follow the step-by-step guide and explain which step you are having difficulty with.
https://github.com/shadow-1/yi-hack-v3#getting-started---step-by-step-guide

Dp4610yz, 2017-05-26:
due to my error i updated the firmware in the ip camera... in order get it to work again, i loaded the two programs into the sdcard which was formatted for fat32... 16gb card... but i put it into the camera but it does not load the software..only when i tried the previous version of the software; the firmware got loaded... 

https://github.com/shadow-1/yi-hack-v3#getting-started---step-by-step-guide

shadow-1, 2017-05-26:
@Dp4610yz 
If you are having trouble loading the latest firmware (maybe because the update got interrupted somehow), you can downgrade to an older version first before upgrading again to the latest to get it working.

The camera does a check to see if the firmware image on the microSD card matches what is already installed on the camera. If the camera thinks that the upgrade files match what is already on the camera, it will skip the firmware update.

The check that the camera performs is not very smart and occasionally if something goes wrong, the camera may think it has already updated the firmware but it has not.

closed 2017-06-29 by shadow-1