shadow-1/yi-hack-v3 #24: Not able to open the IP address in the browser

umgangl, 2017-05-28:
Hi there,

I found the IP address that's assigned to the camera, but when I entered it in the browsers (IE, Firefox, Google Chrome) I received the message "403 forbidden File is protected".

Not sure if the router setting needs to be changed. I did some searching and tried to change the settings in router and the network, but it's still not working. Just wonder if you know how to resolve it.

Thank you

[image]

shadow-1, 2017-05-29:
@umgangl 
The most common cause for this issue is that yi-hack-v3 has not flashed properly.
In particular, you need to make sure that both rootfs and home firmware files have been flashed to the camera. You must flash both files, otherwise yi-hack-v3 will not work.

Assuming that there has been an issue whilst flashing the firmware, you can flash an older version of yi-hack-v3 (for example 0.1.2) before flashing the latest version again (at this time 0.1.3). This will ensure that the firmware gets flashed again.

umgangl, 2017-05-30:
@shadow-1 

Thank you soooo much for the help. it's working now! You're awesome!!!

I followed your guides to use the older version and it worked. I had problems with the proxies, but it all got worked after checking other thread.

I bought few Yi Dome and Yi Home last November and tried few times. Didn't work at all. At least I have the Yi Dome to use now.

Greatly appreciated!

MarioHA, 2017-08-31:
I am also having this issue with the Yi Dome and I tried installing each firmware from version 0.1.2 on up to 0.1.4 beta2.
Anything you can do to help would be greatly appreciated.
Thank you in advance.

shadow-1, 2017-08-31:
@MarioHA 
Next time, please try to open a new topic rather than comment on a closed thread.
What issue are you having with the firmware?

@batrarobin has suggested to try a different power adapter in this thread https://github.com/shadow-1/yi-hack-v3/issues/45 .
It appears that this has helped some people with similar issues.

MarioHA, 2017-08-31:
@shadow-1 Apologies for not creating a new topic.
I will try a different power adapter, though I think that 5V 3A should suffice with the one I used.
Everything goes exactly as you said in the instructions except for being able to log in to the IP address. 
Also, the camera shows up as "Yi-hack-v3" in my router after firmware flash, so it seems like it worked but the local connect to the link you provided at http://yi-hack-v3 simply returns a bad dns error and never goes anywhere. I also tried by typing in directly the IP of the camera itself and it times out.

shadow-1, 2017-10-11:
@MarioHA 
If you continue to have issues. Feel free to open a new issue.

closed 2017-10-11 by shadow-1