shadow-1/yi-hack-v3 #25: Will this "hack" be available for YI Home camera (aka Ant)?

tbolsh, 2017-06-05:
Hello!

Is it possible to use this hack for YI Home camera? The "/fritz-smh/yi-hack" project works with the very old firmware and, in fact, I had problems to make it work at all. If you can describe how to compile new fw for YI Home camera - as you did in another issue - I will try to follow and share the experience.

Thank you for providing the "hack"!

shadow-1, 2017-06-06:
@tbolsh 
The Yi Home camera is very similar to the Yi Dome / Yi 1080p Dome / Yi 1080p Home cameras which I currently support. However it is different enough that I would need to develop a separate set of firmware images for it.

As far as I know, the Yi Home camera uses a slightly different chipset to the Yi Dome / Yi 1080p Dome / Yi 1080p Home cameras. I believe the Yi Home uses the Hi3518_V100 chipset whilst the cameras I currently support utilise the Hi3518E_V200 chipset.

That will mean that a different compiler will need to be used to compile programs for this camera. However I already have the SDK for the Hi3518_V100 chipset.

I don't have this particular camera, so I will need your help to develop custom firmware images.

Do you have Telnet access on your camera? Do you happen to have serial port access to your camera? It would be helpful to see the bootup messages that the camera displays through the serial port.

---

On a separate note, I am wondering why you want to use my firmware on the Yi Home? What features are you looking for?
I thought there were several custom firmware images available for the Yi Home.

tbolsh, 2017-06-06:
Hello, shadow-1!
Thank you for the fast answer.
I will start from the last questions:
     I currently have the US version of the camera, but I do not have neither telnet access nor serial port access to the camera to see bootup message. I will try to look into that today.
     My goal is to develop a certain prototype of (potentially multi-party) video communication system that will utilize several different cameras and custom (or rather not standard) HW. One peer (Android app) can potentially have several video/audio streams, other peers (WebRTC applications) have one std. camera and ability to choose between streams from the presenter. Recently I switched my attention to YI cameras - they are cheap and seem to be highly customizable. Currently, I am playing with YI Dome 1080P, YI Home, and YI Action 4k. So far I looked at them only for a couple of days but see a great potential. So, if you can share the SDK and some instructions or place where I can download the SDK and get the instructions - I can try to compile FW myself and share the results with you.

Why your FW? I would highly prefer to have one platform and several cameras to choose from. I may choose not include YI Home into the set if it would not support that "one platform". 

Why your FW is the one platform?  It seem to support the latest standard FW from YI, seem to be live and updated at least weekly. Discussions in issues also have sense.

Thank you once again, good luck with everything,

tbolsh

P.S. If it will seem to be too hard - I will stop playing with YI Home.

tbolsh, 2017-06-06:
P.S. The features: 
-- ability to develop/compile for the platform,
-- ssh or telnet support during development, 
-- RTSP or WebRTC support (I can bridge myself),
-- web-server (mostly for debugging purposes).
And yours FW simply works right out of the box.

shadow-1, 2017-06-07:
@tbolsh 
It sounds like you have a very ambitious project.

The SDK for both the HiSilicon Hi3518E_V200 and HiSilicon Hi3518_V100 is available on my box account.
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

The latest version of the SDK I currently have for the Hi3518E_V200 is; Hi3518E_V200R001C01SPC040
The latest version of the SDK I currently have for the Hi3518_V100; Hi3518E_V100R001C02SPC0A0
This one is probably a little outdated as I don't have the Yi Home camera.

That should give you all the tools you need to start developing on both camera platforms.

I may be able to help you out with Telnet and web server access on the Yi Home. However it would be easiest for me if you can obtain Telnet access using Fritz firmware (even if it uses a very old firmware) and obtain a backup of all the partitions on your camera.

On the Yi Home, you can use the RTSP server included with the Fritz firmware (however I believe you will loose support for the Xiaomi smartphone app). It is a shame that the source code for the RTSP streaming program is not available in all of the custom firmwares I have seen for the Yi cameras. It would be much quicker for me to adapt RTSP from the Yi Home for use on the Yi Dome cameras rather than writing a RTSP streaming program from scratch.

cnberg, 2017-07-01:
Up.

I have a camera cannot be hack by Fritz hack (https://github.com/fritz-smh/yi-hack/issues/173), so I think your hack will help. Since Fritz maybe stop maintenance his project( I submit this issue two month ago but no one reply).

Thanks.

shadow-1, 2017-07-01:
@cnberg 
If I had to guess, I think the camera that you can't flash the Fritz firmware is the Yi Home version 17CN.
Yi Home 17CN has a very different firmware to the previous versions of Yi Home. However it is very similar to Yi Dome and Yi 1080p Home.

---

I think the only way to get alternative firmware running on the camera would be to connect to the camera through the serial port and to make some changes to the filesystem. This may require soldering. There may or may not be a connector inside the camera for serial port. If there is no connector, the serial port will need to be soldered.

I can probably develop yi-hack-v3 for this camera. However as I don't have this camera, I will need help from someone with serial port access to the camera. A backup of all partitions on the camera should be sufficient for me to develop an initial version of yi-hack-v3 for this camera.

If you are willing to experiment on your camera, then I may be able to help.

cnberg, 2017-07-03:
Yes!  I'm using 17CN version camera...

What I can do now? Like this? https://www.youtube.com/watch?v=3e8jXuxiRzY

shadow-1, 2017-07-03:
@cnberg 
From the YouTube video, it appears that the author of the video is using an older version of the Yi Home (not 17CN). However the procedure is exactly the same.

I have read online that the connector did not work for some people and they had to solder onto the pads shown on the video. However I have never seen a picture of the setup from those people. Maybe they mixed up the pins or their connection to the pins on the connector was not very good. On the Yi Dome cameras, there is no connector and the only choice is to solder onto the pads.

Remember that you must use a USB-Serial converter. The serial port on these cameras operate at a voltage of 0 to 5V. If you have an old PC with native serial port, these normally operate at a voltage of -12 to 12V which may fry the camera. USB-Serial converters may operate at 0 to 3.3V or 0 to 5V (this is more common and is what is required in this case). If you have a spare Arduino, you may be able to use it as a USB-Serial converter if you don't have a dedicated device (it is what I use as I find it easier to connect to than my other USB-Serial converter). Arduino microcontrollers operate on 0 to 5V signals.

---

If you do manage to get access to the serial port of the camera. Start the camera with an microSD card inserted. Run the following commands:
```
dd if=/dev/mtd0 of=/tmp/sd/mtd0
dd if=/dev/mtd1 of=/tmp/sd/mtd1
dd if=/dev/mtd2 of=/tmp/sd/mtd2
dd if=/dev/mtd3 of=/tmp/sd/mtd3
dd if=/dev/mtd4 of=/tmp/sd/mtd4
dd if=/dev/mtd5 of=/tmp/sd/mtd5
dd if=/dev/mtd6 of=/tmp/sd/mtd6
dd if=/dev/mtd7 of=/tmp/sd/mtd7
```

You can then post the files mtd0, mtd1, mtd2, mtd3, mtd4, mtd5, mtd6, mtd7 from the microSD card to me and I should be able to create an initial test firmware release for the camera 

---

It would also be helpful if you could post the output of the following command:
```
cat /proc/cpuinfo
```
This will confirm which chipset the camera uses and thus which compiler I need to use to.

cnberg, 2017-07-07:
Sorry for my late reply.

I got these files, and the cpuinfo below. Since I have physical access to the camera, is there any way to hack the camera with full rtsp function?

Thanks for your help!

[Archive.zip](https://github.com/shadow-1/yi-hack-v3/files/1131395/Archive.zip)

Processor       : ARM926EJ-S rev 5 (v5l)
BogoMIPS        : 298.59
Features        : swp half fastmult edsp java
CPU implementer : 0x41
CPU architecture: 5TEJ
CPU variant     : 0x0
CPU part        : 0x926
CPU revision    : 5

Hardware        : hi3518ev200
Revision        : 0000
Serial          : 0000000000000000

shadow-1, 2017-07-08:
@cnberg 
Fantastic! I should be able to create a test version of yi-hack-v3 for Yi Home 17CN soon. It is very interesting that the Yi Home 17CN uses exactly the same processor as the Yi Dome cameras and the Yi 1080p Home camera. It does not use the same processor as the original Yi Home.

---
Can you please post the output of the following command for me as well.
```
ls /tmp
```
---

In terms of unlocking RTSP function. All the current implementations do not allow for the official Xiaomi app to work at the same time as RTSP. So on your camera you will have to kill most of the official Xiaomi programs.
```
killall watch_process
killall p2p_tnp
killall cloud
killall mp4record
killall rmm
killall oss
```
Note: It is very important that you kill watch_process first. This program automatically restarts the Xiaomi programs whenever they close. Presumably the Xiaomi programs crash from time to time and this ensures that they get restarted automatically.

In terms of RTSP functionality, you can try running the Fritz firmware RTSP implementation. Unfortunately I do not think it will work as the processor between the original Yi Home and Yi Home 17CN is different.
https://github.com/fritz-smh/yi-hack/raw/master/sd/test/rtspsvrM

I am helping out with RTSP implementation with the people working on this project: https://github.com/xmflsct/yi-hack-1080p . This project has got an initial test version of RTSP working. It is very limited at the moment and not completely stable. However it may work well enough for you for the time being. Link to the RTSP program: https://github.com/xmflsct/yi-hack-1080p/raw/master/sd/test/app/rtsp2301

---

It will probably take quite a while to get RTSP working reliably and properly on these cameras because we are starting from scratch. We do not have the source code for RTSP used on Fritz firmware for example. If we did, we could modify it to work on these other cameras and also extend the functionality (extra features).

I am also working to make RTSP work at the same time as the official Xiaomi programs.

shadow-1, 2017-07-10:
@cnberg 
I have released a new beta version of yi-hack-v3 (0.1.4-beta2). I have now added support for the Yi Home 17CN. Feel free to try it out. Let me know if you have any issues. As I don't have this camera, I am relying on community feedback.

cnberg, 2017-07-10:
Thanks for your great work!!

I'm really looking forward to have a hack can work on xiaoyi v2 & v3 camera which supports RTSP.

ls /tmp, I got: 

> audio                log.txt              mmap.info            view
init                 logsock              sd                   wpa_supplicant.conf

I will try your new version right now.

shadow-1, 2017-07-10:
@cnberg
I guessed that would be the contents of /tmp.
Yi Home 17CN appears to be identical to the Yi Dome (except PTZ functionality).

At least with this release you will get web server, telnet server, ftp server working. RTSP is a work in progress.

cnberg, 2017-07-10:
This ROM works very well! I got a webpage titled "Yi Home 17CN". telnet & ftp working perfectly.

But there is no voice prompt when the camera started or connecting wifi.

shadow-1, 2017-07-10:
@cnberg 
I was under the impression that the voice prompts only occur when the camera is not configured, is reset or when the camera is being used outside of Mainland China ("This camera can only be used within China" voice prompt).
Did you reset the camera to test the voice prompts?

cnberg, 2017-07-10:
Yes. First time I booted this camera without any prompt, I reset it, and boot again, also no voice prompts (Such as waiting for connect/connecting wifi/wifi is connected).

Actually, I was in mainland China, so I never heard "This camera can only be used within China" :P

shadow-1, 2017-07-10:
Hmm...that is strange. The audio files that the camera uses are linked in the following location `/tmp/audio`.
What is output when you enter the following commands:
```
cd /tmp
ls
cd audio
ls
```

shadow-1, 2017-07-18:
@cnberg 
I have created a new test firmware that is compatible with both Yi Home 17CN and Yi Home 27US (I had no idea the 27US camera even existed until recently).

Can you try the following pair of images to make sure it still works properly for the 17CN. In addition, it may fix the audio issue you were having before. Firmware files below:

home: https://www.dropbox.com/s/lxwxvm1haydkqmb/home_y18?dl=0
rootfs: https://www.dropbox.com/s/w37nkc7bz0isv2o/rootfs_y18?dl=0

TheWayAway, 2017-08-01:
Has anyone had experience hacking the Yi Home 27US camera?
How do I confirm which version of the Yi Camera I have (so I can make sure)

I'm assuming it's this one on Amazon https://www.amazon.com/YI-Camera-Wireless-Security-Surveillance/dp/B019MMRV1M

I'll be getting one today and hopefully I'll try to see if it works if I have time tonight.

shadow-1, 2017-08-05:
@TheWayAway 
Great to hear that everything worked as expected! Thanks for sharing your experiences.

nosmex, 2017-08-09:
@shadow-1 the 27US version seem to work fine. I was able to load it I can hit the cam by IP.  I have a black US edition cam. Would you please post an update if the FW with RTSP is available. Thanks!

shadow-1, 2017-08-09:
@nosmex 
RTSP has not been developed yet. I will release it when it gets developed.

amggupta, 2017-08-13:
I tried this hack on 27US version camera, and it worked beautifully.

I plan to use this camera as a car reversing camera, and see the feed on a mobile phone that would be running a wifi hotspot too. How do i ensure that everything works and remain on local network. How can i ensure that it does not require internet access for it to work ? (I'm hoping that rtsp work would make it possible to directly see the feed on browser or any other app)

[BTW, it also seems that camera requires active internet connection for establishing the link between camera and app. Once the link is on, even if i turn the data off on the phone streaming works fine.]

shadow-1, 2017-08-14:
@amggupta 
At the moment, this firmware only supports the official Xiaomi app. For the app to work, the camera needs to authenticate with the Xiaomi server prior to use (this is where the region banning becomes an issue for those using the Chinese version of the camera outside of Mainland China).

RTSP should resolve this issue as it will be relying on the Xiaomi app to work

From your experimentation, it appears once the camera has authenticated with the Xiaomi server (only occurs at boot) and has connected to the camera, you no longer require an Internet connection.

amggupta, 2017-08-15:
@nosmex 

Yeah, if RSTP can work on this setup that would be awesome. 

along with the boot, I think it requires the working internet link from phone to camera at the time of launch of feed. Not very optimal as reverse cam.

Also i dont want it to 3g/4g data on the mobile to keep sending some part of feed data to servers. Dont care about security, but worry about data usage.

nosmex, 2017-08-15:
Looks like there is a new FW update: 1.8.7.0D_201708091510

shadow-1, 2017-08-16:
On firmware versions 0.1.4-beta2 onwards, it is safe to perform Xiaomi updates. It will not affect yi-hack-v3.

Pinnhate, 2017-08-17:
After flashing 27US with
home: https://www.dropbox.com/s/lxwxvm1haydkqmb/home_y18?dl=0
rootfs: https://www.dropbox.com/s/w37nkc7bz0isv2o/rootfs_y18?dl=0
what version of YI Home i need? I try both int and cn but without success/

galadril, 2017-08-17:
@Pinnhate i used the playstore yi home app like stated for internation versions on the readme

shadow-1, 2017-08-18:
@pinnhate
As the read me states. You will need to disable ProxyChains-ng as it is not required for the International versions of the camera.

Yi Home from the Google App Store is required for the International versions of the camera.

ericboehs, 2017-09-02:
Got it working on my 27US as well. Just dropped the two files on the SD card and rebooted. I look forward to RTSP support. :)

S178, 2017-10-06:
@shadow-1 
Do you manage to hack the firmware for yihome 1080 version 2? I can provide my yihome for your experience...

shadow-1, 2017-10-06:
@S178 
Unfortunately the Yi Home 2 is not supported with this firmware.
The Yi Home 2 firmware uses a completely different chipset to the original Yi Home.

The Yi Home 2 makes use of the Ambarella S2LM chip.
This firmware only supports cameras based on the HiSilicon Hi3518eV200 chipset.

The firmware linked below supports the Yi Home 2. I am not sure if it supports the 1080p version of the camera though.
https://github.com/niclet/yi-hack-v2

S178, 2017-10-07:
@shadow-1
Thank you for your prompt reply. I've tried but still fail. I guess it's because I know nuts about IT. Do you happen to provide service to hack the firmware?

deanet, 2017-10-15:
can't wait for rtsp support release..

untotren, 2017-10-19:
RTSP tho

Ordep, 2017-10-22:
Hi 
I own a Yi home camera 22US and I  wonder if it would be possible to make it work like the 27US.?

itreers, 2017-10-23:
Hi. Successfully updated my 27US with latest version of hack (dropbox links). Thank you. Will wait for RTSP.

shadow-1, 2017-10-24:
@Ordep 
I think the 22US version of the camera is the original Yi Home camera. Support is still a work in progress. It is not very far off from a pre-release.

@itreers 
Unfortunately I have not had the time to develop RTSP support yet.
You can definitely help if you are a keen developer.

itreers, 2017-10-24:
I have an issue with firmware for 27US. After install and set Proxy to disable all was fine. After I change password for root and restart a camera it goes to constant reboot (it must be tested) I restored a camera with restorational firmware provided by (you?) for 27US cameras and install your latest firmware again. Some hours, couple of reboots and power off (without changing root pass) - working ok.

deanet, 2017-10-25:
@shadow-1 
i'm trying to run rtsp server from https://github.com/xmflsct/yi-hack-1080p/tree/master/sd/test/app  and run successfully but only black screen on streaming.

any advice ?

shadow-1, 2017-10-25:
@deanet 
What camera are you using that RTSP implementation with?

It has been tested on the Yi Home 1080p and is working. However it will not work with the 720p versions of the camera (Yi Home, Yi Home 17CN / 27US, Yi Dome) without modifying the code and recompiling. I am unsure whether it will work with the Yi Dome 1080p without modification.

Also be aware that there are some limitations with this implementation at this stage.

deanet, 2017-10-26:
@shadow-1 

my camera version is Yi Home 17CN. How to recompile the code ? it's small linux box with different processor (amd or intel).

shadow-1, 2017-10-26:
@deanet 
More is required than just recompiling. The code has to be developed/modified and compiled against the image sensor library that is used by the 720p camera.

That particular implementation that you linked to  is compiled against the image sensor library applicable to the 1080p version of the camera.

---

I have hosted the SDK for the chipset used on the Yi Home 17CN (HiSilicon Hi3518eV200) on my Box account: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr The latest version of the SDK I currently have is; Hi3518E V200R001C01SPC040.

The source code for this particular RTSP implementation is available here: https://github.com/xmflsct/yi-hack-1080p/files/908602/rtsp2301_20170409.zip

markusd1984, 2017-10-30:
@shadow-1

Many thanks! Another happy user on Yi Home 17CN!
In fact, my camera was locked down after the missus accidentally updated the FW and I already bought a TTL USB adapter as a last resort to downgrade, so happy to have found your firmware hack that made it work again!

Now patiently waiting to try RTSP. A hack to adjust the time zone would be a big + :)

Ordep, 2017-11-09:
> @Ordep I think the 22US version of the camera is the original Yi Home camera. Support is still a work in progress. It is not very far off from a pre-release.
--

@shadow-1 Let me know if you need someone to test it, when the time comes.

shadow-1, 2017-11-19:
@Ordep @tbolsh 
I have created a version of the firmware compatible with Yi Home. It is completely untested as I do not have this camera.

Feel free to test it. If there are any issues, can you let me know so they can be fixed. I believe the Yi Home will only require the rootfs partition to be flashed, unlike the other Xiaomi cameras which I support.

rootfs: https://www.dropbox.com/s/ukqd6uptf8nvlfw/rootfs?dl=0

In case there are issues and you would like to go back to stock firmware. I have created a recovery image compatible with the Yi Home camera:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

Ordep, 2017-11-19:
@shadow-1 That's great.

Here's the steps I took.

1) Formatted micro sd card to FAT32
2) threw the file there
3) inserted into the camera
4) started the camera
5) it turned blue
6) but cant access the webpage

What's the best way to send you the log files?

Thank you

shadow-1, 2017-11-20:
@Ordep 
Did the cameras yellow light come on and blink when you started the camera with the firmware file on the microSD card? This is the indicator that the firmware is being flashed. This only occurs once when you initially flash the firmware, if the camera is restarted, it will detect that the firmware file matches what is currently on the camera and skip the firmware update.

Keep in mind that the firmware file for the Yi Home is smaller than the firmware on the other supported cameras, so I expect the yellow light to flash for 10-20 seconds before the camera boots rather than the 30 seconds like the other supported cameras.

Assuming that the firmware has been flashed, does Telnet of FTP work? Does the camera still work with the Yi Home app?

I have a suspicion that I have a bit more work to complete with BusyBox. However it is best to confirm.

Ordep, 2017-11-20:
@shadow-1 Couldn't see the yellow light blinking, although it seems I have ftp access, using the root user and it is still working fine using the app.

When I try telnet it I get a login incorrect.
The webpage doesnt seem to be working, either.

PS: Tried to reset the camera this time, in case it helped.

Cheers

shadow-1, 2017-11-22:
@Ordep 
Hmm...because the yellow light didn't blink when starting up. The firmware did not flash. Perhaps you can try flashing the Recovery image before going back to the custom firmware.
Recovery: https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr
Custom: https://www.dropbox.com/s/ukqd6uptf8nvlfw/rootfs?dl=0

If the yellow light still doesn't blink when booting up with either the Recovery firmware or the Custom firmware. Can you try another microSD card?

Ordep, 2017-11-22:
@shadow-1 some progress :)
after all it was the program I was using to format the sdcard that wasn't working properly.

I manage to install the custom software.
I got access to the web portal
port 21 is open but root is not working. I'm getting Login Failed
port 23 is closed

PS: I did notice the camera is once in a while blinking blue

Thank you

shadow-1, 2017-11-28:
@Ordep 
Looks like I have a little bit more work to do.
Very strange that FTP and Telnet are not working. Perhaps you can try using the password `1234qwer`.

kylerw, 2017-11-29:
@shadow-1 that password works for me!

shlmn, 2017-11-29:
@shadow-1 web access is available, telnet/ftp unavailable (connection refused).
Blue light is blinking every once in a while (wi-fi lost and then coming back).

shadow-1, 2017-11-30:
@shlmn @kylerw 
Thanks for the feedback! I should have a new test build to try within a few days.

ajayjohn, 2017-12-25:
Have been actively following this thread am confused if the 27US version will work on the 22US too :( Any luck with getting 22US working?
I can test any builds if needed.

bola04053, 2018-01-25:
@shadow-1 

> I have created a version of the firmware compatible with Yi Home. It is completely untested as I do not have this camera.
>
> Feel free to test it. If there are any issues, can you let me know so they [...]

this rootfs works with my 12cn version

Version
 Camera Yi Home
 Base Firmware -
 yi-hack-v3 Firmware 0.1.4-beta3

the only thing is ( ERROR: Could not connect to freevpn.ninja, ensure that you are connectected to the Internet. ) and sometime the blue led blinking because it lost connection with wifi ? after blinking few second , camera back online

bola04053, 2018-01-25:
> I have created a version of the firmware compatible with Yi Home. It is completely untested as I do not have this camera.
>
> Feel free to test it. If there are any issues, can you let me know so they [...]

12cn camera flash this rootfs first , after that config the proxy by manual type in 

random_chain
chain_len = 1
tcp_read_time_out 15000
tcp_connect_time_out 8000
[ProxyList]
socks5 223.241.214.44 1217
socks5 211.95.255.42 1080
socks5 118.89.231.84 30687
socks5 119.29.163.54 1080
socks5 61.145.213.14 1080
socks5 218.58.210.233 1080
socks5 221.209.62.170 1080
socks5 117.135.132.107 3000

change the proxy by going to http://spys.one/proxys/CN/ , after that go to test proxy and save it 

go to ( http://www.xiaoyi.com/home/firmware/download/?version=1 ) download the latest firmware ( 1.8.7.0C_201705091058home.zip ) from official webside and throw home and home_y18m inside sd card , put the sd card into camera , after that pair it with app , and it success connected to app , you can also go to the ProxyChains-ng by camera ip

stephen-yeung, 2018-03-17:
@bola04053 or anyone, can you explain a bit more detail.
rootfs: https://www.dropbox.com/s/ukqd6uptf8nvlfw/rootfs?dl=0
I have a 12CN, I throw the above rootfs into sdcard and it doesnt do anything. which home should I use it with?

Thanks

[7 more replies]