shadow-1/yi-hack-v3 #26: How to send  control commands to YI Dome 1080p and YI Dome cameras?

tbolsh, 2017-06-05:
Hello!

Sorry for raining issues on your head ...
Is it possible to send control commands for the Dome camera:
 - rotate left / right / up / down
 - zoom (digital, right?)
 - pan (if there is a digital zoom there should be pan available, right?).

Thank you for doing this great work!

shadow-1, 2017-06-06:
@tbolsh 
The Yi Dome cameras can be controlled directly through the official Xiaomi app. At the moment it is not possible to directly control the cameras as they run proprietary Xiaomi programs which are undocumented.

I believe controlling the PTZ functions on the camera will be very useful after I develop RTSP support. It is definitely something I would like to look into in the future.

You are correct that zoom would have to be implemented digitally. All the Xiaomi cameras I have seen do not have zoom lenses. However I believe it would be better for the end device (network video server, PC software, mobile phone app etc.) to perform digital zoom along with pan. I don't think these cameras have enough processing power to do it on the camera itself. However I could be wrong.

tbolsh, 2017-06-06:
Thank you, shadow-1!
I will be looking forward to any new information about PTZ control or RTSP support. 
I might be wrong, but I think that processor should have enough power to support digital zoom to some extent - pretty often the sensor is bigger than video resolution and digital zoom / pan is just a selection of the sensor area being used for the video. So, to some degree, "normal" video uses some additional power for averaging pixels. Digital zoom might be available by "predefined" steps - along with the pan. 
Any information about that will be highly appreciated - I will also share whatever I will be able to find out.

Thank you, again, for doing the great job,
tbolsh