shadow-1/yi-hack-v3 #27: Mijia 1080 support ?

xX-Nexus-Xx, 2017-06-06:
will the new XIAOMI  Mijia 1080 cam be supported in future or is it already ?

http://www.geekbuying.com/item/Xiaomi-Mijia-Smart-IP-Camera-1080P---White-376769.html

shadow-1, 2017-06-06:
@xX-Nexus-Xx 
The Xiaomi MiJia cameras are very different to the cameras I currently support (Yi Dome, Yi 1080p Dome, Yi 1080p Home). They use a completely different chipset. The cameras I currently support utilise the HiSilicon Hi3518E_V200 chipset whilst the Xiaomi MiJia cameras I know about utilise the Ambarella S2Lm chipset.

The Xiaomi MiJia 1080p is a very new model, so I am not exactly sure what chip it uses.

Because the MiJia cameras use a completely different processor, a completely different hack would need to be developed for this camera. As I do not have this camera, I don't think I will be able to develop a hack for this camera.

By the way, is there a reason why you want to use a custom firmware for this camera? I thought this camera was very advanced and includes many extra features built in.

xX-Nexus-Xx, 2017-06-07:
Hi Shadow,

thank you for the quick reply.  I was not certain about the chipset in these cameras.

the camera's do not support the RTSP protocol and i want them integrated into my openhab home automation environment and do need the IP access to these cameras. I know RTSP is not part of your hack yet but it was worth waiting for ;)

I have two of these camera's now and they have superb picture quality day and night (better than Yi Dome 1080) as well the other option for distortion correction is working fine. Still, I would like to have the streaming capability

shadow-1, 2017-06-07:
@xX-Nexus-Xx 
That makes sense now.
It is a bit unfortunate that I don't think I will be able to develop for this platform.

It appears that MiJia and Yi are two different sub-brands of Xiaomi. I think this is the reason why these cameras are so different. They even use a different smartphone app.

bronzerain, 2017-06-15:
xX_Maguro_Xx 
The MiJia 1080p will send pictures and videos to NAS, if the NAS is plugged into your WIFI router and the router supports it.

closed 2017-08-07 by shadow-1

maelcum, 2018-06-03:
@shadow-1: If somebody, like a benevolent benefactor, would donate one of these cameras to you, would you be interested in developing for these types as well? No strings attached? Worst case, you get a cam for free... :)
I am an owner of these too and their build quality, picture quality and reliability is superb. The Cam saves to the SD-Card and any SMB share (like bronzerain mentioned) but needs the GUI to work - of which I am no fan of. No http interface, no ssh, no nothing else. I'd be happy to invest in one or two more cams, sending them your way, if that could motivate you to put your expertise to work.