shadow-1/yi-hack-v3 #28: Proxy cant connect

jecash, 2017-06-10:
Cant conncect camera
[image]

Settings:
[image]

/ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  123.56.199.203:5271  ...  34.202.90.94:80  ...  OK
{
  "ip": "47.88.79.0",
  "hostname": "No Hostname",
  "city": "Hong Kong",
  "region": "HONG KONG",
  "country": "HK",
  "loc": "22.2833,114.1500",
  "org": "AS45102 Alibaba (China) Technology Co., Ltd."
}/ #

but 
[image]

help?

shadow-1, 2017-06-10:
@jecash 
I suggest you clean up your proxy list.
I have tested the proxy servers in your list and where I am located I found the following:

---
```
socks5 119.28.15.252 2018
socks4 115.29.161.167 1080
```
These two proxy servers above appear to work properly for me.

---
```
socks5 123.56.200.75 5261
socks5 183.207.176.252 1080
socks5 123.56.199.203 5271
```
The above three proxy servers do not work for me where I am located.

---
Occasionally `socks5 183.207.176.252 1080` works for me. However for some reason the location service I am using ([ipinfo.io](http://ipinfo.io)) shows that it is located in California, USA. From your screenshot, it appears that `socks5 123.56.199.203 5271` works for you (it does not for me). However the command line output that you posted shows that the location is Hong Kong. The proxy server must be located within Mainland China. Hong Kong and Taiwan will not work as they are considered as being outside of Mainland China.

---
My suggestion is to leave only the following proxy servers in the your list:
```
socks5 119.28.15.252 2018
socks4 115.29.161.167 1080
```

Remember that you might have to retry the test a few times for it to pass.

shadow-1, 2017-06-10:
@jecash 
Can you post the output from the following command in Telnet:
```
proxychains4 wget -O- -q https://ipinfo.io
```

Run the command a few times in a row (3-4 times) and post the output for each time it runs.

jecash, 2017-06-10:
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  119.28.15.252:2018  ...  34.202.90.94:80  ...  OK
{
  "ip": "119.28.15.252",
  "hostname": "No Hostname",
  "city": "Choucheng",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "29.3151,120.0768",
  "org": "AS133478 Tencent Cloud Computing (Beijing) Co., Ltd"
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  115.29.161.167:1080  ...  52.86.112.58:80  ...  OK
{
  "ip": "115.29.161.167",
  "hostname": "No Hostname",
  "city": "Hangzhou",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "30.2936,120.1614",
  "org": "AS37963 Hangzhou Alibaba Advertising Co.,Ltd."
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  115.29.161.167:1080  ...  34.196.176.140:80  ...  OK
{
  "ip": "115.29.161.167",
  "hostname": "No Hostname",
  "city": "Hangzhou",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "30.2936,120.1614",
  "org": "AS37963 Hangzhou Alibaba Advertising Co.,Ltd."
}~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  119.28.15.252:2018  ...  34.202.90.94:80  ...  OK
{
  "ip": "119.28.15.252",
  "hostname": "No Hostname",
  "city": "Choucheng",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "29.3151,120.0768",
  "org": "AS133478 Tencent Cloud Computing (Beijing) Co., Ltd"
~ # proxychains4 wget -O- -q http://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  119.28.15.252:2018  ...  52.86.112.58:80  ...  OK
{
  "ip": "119.28.15.252",
  "hostname": "No Hostname",
  "city": "Choucheng",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "29.3151,120.0768",
  "org": "AS133478 Tencent Cloud Computing (Beijing) Co., Ltd"
}~ #

shadow-1, 2017-06-10:
@jecash 
From the output of the command in Telnet, everything is now working perfectly.
You can now try pairing your camera with your smartphone.

Unfortunately the test config webpage is not perfect. It will be made more robust in the next release of the firmware.
Sometimes you need to retry the test on the webpage a few times for it to pass.

shadow-1, 2017-06-10:
@jecash 
Another thing you can post is the output from the following command in Telnet:
```
proxychains4 wget -O- -q https://ipinfo.io
```
Notice that the command has `https` and not `http`. In y-hack-v3 0.1.3, I enabled wget ssl support in the firmware, however I did not have time to tweak the webpage to use it.

jecash, 2017-06-10:
~ # proxychains4 wget -O- -q https://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  119.28.15.252:2018  ...  52.86.112.58:443  ...  OK
{
  "ip": "119.28.15.252",
  "hostname": "No Hostname",
  "city": "Choucheng",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "29.3151,120.0768",
  "org": "AS133478 Tencent Cloud Computing (Beijing) Co., Ltd"
}~ #

shadow-1, 2017-06-10:
@jecash 
Yes, everything is still working properly.
You should be able to pair your camera to your smartphone without issues.

jecash, 2017-06-10:
Thanks all.
reinstal yi-hack3
Camera switched co china language, and connected to phone.

badtime123, 2017-06-18:
@shadow-1 i have the same problem ...unable to pair camera to smartphone...
version used is same with jecash..

below are the output from telnet
~ # proxychains4 wget -O- -q https://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  119.28.15.252:2018  ...  54.71.17.15:443  ...  OK
{
  "ip": "119.28.15.252",
  "hostname": "No Hostname",
  "city": "Choucheng",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "29.3151,120.0768",
  "org": "AS133478 Tencent Cloud Computing (Beijing) Co., Ltd"
~ # proxychains4 wget -O- -q https://ipinfo.io
[proxychains] config file found: /home/app/proxychains.conf
[proxychains] preloading /home/app/locallib/libproxychains4.so
[proxychains] DLL init: proxychains-ng 4.12
[proxychains] Random chain  ...  119.28.15.252:2018  ...  35.164.175.43:443  ...  OK
{
  "ip": "119.28.15.252",
  "hostname": "No Hostname",
  "city": "Choucheng",
  "region": "Zhejiang",
  "country": "CN",
  "loc": "29.3151,120.0768",
  "org": "AS133478 Tencent Cloud Computing (Beijing) Co., Ltd"
~ #

shadow-1, 2017-06-18:
@badtime123 
Based on the logs you posted, it seems like the proxy server is working properly.

---

Have you updated Yi Home app to the latest version?

There is no harm in updating the smartphone app. The Android app hosted on my Box account is a bit old now. Old versions of the app may no longer work properly.

---

Are you sure you are using the correct smartphone app?
You must install the Chinese version of the app if you have the Chinese version of the camera. At the same time you must install the International version of the app if you have the USA, EU, SG etc version of the camera.

I don't have an iPhone, however my understanding is that you must switch your region to China and download the app through the Chinese App Store. After the app is installed you can switch your region back to what it should be.

badtime123, 2017-06-20:
@shadow-1  Thanks connected to phone due to phone application .

jecash, 2017-06-20:
and second problem
in the same network camand phone working fine
but if i connect phone to other wifi or cel network, program on phone say can't connect

shadow-1, 2017-06-20:
@badtime123 
Good to hear that everything is working!

@jecash 
This is a very strange issue as I never encountered it. I never had issues accessing the camera through mobile network or another network with Internet access. This sounds like an issue that is caused by your home network setup somehow.

Do you have a firewall (something like pfSense or IPCop), maybe even running your own DNS server/adblocker (something like Pi-Hole) which could be blocking the P2P cloud communications between the camera and the Internet?

Just as a test you can try turning on WiFi hotspot feature on your smartphone and pair the camera with the mobile hotspot. Then you try accessing the camera using another smartphone or tablet that has mobile Internet.

This test can be used to completely rule out your home network as an issue. From there you can keep the camera running on your mobile hotspot and try accessing the camera through your home WiFi network through the Xiaomi PC software or the app on another smartphone or tablet.

shadow-1, 2017-07-10:
@jecash 
This topic has been open for a while now. If you are still having issues, please feel free to open a new issue.

closed 2017-07-10 by shadow-1