shadow-1/yi-hack-v3 #29: The Camera Does Not Appear On My Apps

huazzz, 2017-06-22:
Hi,
THanks for the useful guide.
I have downloaded the firmware and the relevant apps for china version of YI CAM 1080P ( not the new one, is the one similar to the 720p design but 1080P ).
And successfully paired it with the apps, HOWEVER, the camera does not show in the apps.
What seems to be wrong here ? Is it something i misdone ? cos pairing is successful.

Location : Malaysia

shadow-1, 2017-06-22:
@huazzz 
You are the second person that has asked about this issue. I find it a bit hard to troubleshoot as I never encountered this issue.

One thing I definitely recommend is to update your smartphone app to the latest version. The version hosted on my Box account is quite old now. I am not sure whether the old app works properly anymore.

Another things to try is resetting and repairing the camera to your phone.

Keep me posted if you solve this issue. The last person that had this issue just went quiet when I gave a few tips to try. So I assume the problem went away.

huazzz, 2017-06-22:
I did ran the CHECK UPDATE but nothing came in. And i believe the version you provide is the latest already. What about changing the proxy servers ? Will it help ? i wonder..... as i am not a programming guy...

shadow-1, 2017-06-22:
The firmware that is running on the camera is the latest version. However the version of the app is very old now. I have updated the app on my phone many times. A popup comes up asking to "Update" or select "Later".

If you pair successfully, it is unlikely to be the proxy server. However if it fails to pair, it is likely that the proxy servers do not work.

huazzz, 2017-06-22:
My apps version : 2.10.2.11.20161124

Whats yours version now ?

shadow-1, 2017-06-22:
My app is: 2.20.2.10_20170510

huazzz, 2017-06-22:
maybe thats the problem... any idea why i hit the check update button, nothing came out?

shadow-1, 2017-06-22:
I'm not sure why you are not able to update manually. You may have to find the APK file for a newer version of Yi Home.

huazzz, 2017-06-22:
Thanks A Lot ! The issue is caused by older version of yihome apps. just download it manually for the latest version then its good to go !

shadow-1, 2017-06-28:
@huazzz 
Thanks for letting me know what was the cause of the issue!
I am now hosting the APK file for the newest version of Yi Home Android app on my Box account.

closed 2017-06-29 by shadow-1