shadow-1/yi-hack-v3 #30: 1.9.1.0H_20170622 for Dome Camera

Luis2u4u, 2017-06-26:
Hi. There is a new firmware for the Yi Dome Camera, 1.9.1.0H_20170622. Could you hack this new firmware so I can use it on my chinese camera? I already use the v3 hack and it works very nice, but I hope that this new firmware could be better and maybe add new features. Thanks.

DJay79, 2017-06-26:
Here in Germany I see a new Firmware for the Yi Dome 1080p with Version 1.9.2.0F_201706211024

Luis2u4u, 2017-06-26:
Yes, for the 720p version the newest firmware is 1.9.1.0H_20170622 and for 1080p version, the newest firmware is the one you said.

shadow-1, 2017-06-26:
@Luis2u4u @DJay79 
I am very close to releasing the next firmware release, most likely in the next day or two. There have been many changes in this next release, however I believe it has reached the maturity that warrants a beta release.

The next firmware will be based on 1.9.1.0H_201706221046 for the Yi Dome and 1.9.2.0F_201706211024 for the Yi 1080p Dome.

Luis2u4u, 2017-06-26:
That's good to know, thank you very much.

shadow-1, 2017-06-28:
@Luis2u4u @DJay79 
I have now posted a new version of yi-hack-v3 (0.1.4-beta1).
In this release there have been many changes to the web server and the web pages with a lot of extra functionality around ProxyChains-ng. At the same time I updated the base firmware for the Yi Dome and Yi 1080p Dome to the latest revision.

Please keep in mind that it is a beta release. Let me know if you have any issues.

DJay79, 2017-06-28:
@shadow-1 Thanks. Update is complete. I had to reset the cam a couple of times but then everything works. Great! Another step in the direction to [RTSP](https://github.com/shadow-1/yi-hack-v3/issues/15#issuecomment-299618285) ;) 
And do you think there will be a way to send the cam an http request in the future to turn it on/off? Just asking because of my homebridge HomeKit Project.

If there is anything I can do to help, test or whatever, please of course let me know. I thing all other guys here are also appreciate you great work. 👍🏻

shadow-1, 2017-06-29:
For other that may be upgrading to the newest version of the hack. Due to the complete change to the webpages, I believe some people may need to clear the browsers cache.

I would have to look into it. However we might be able to send http requests in the future.

genz10, 2017-06-30:
Hi, is there any way to block client for upgrade their firmware manually ? Like version locking

shadow-1, 2017-07-01:
@genz10 
You might be able to block clients from updating their firmware manually by changing the base firmware version number. You can try editing `/home/app/.appver` manually to a higher number through telnet or ftp and see if that works. This may or may not work (I have not tried it).

In future update to the firmware, I will make it possible to update the official firmware without affecting yi-hack-v3 at all. This should be relatively easy to achieve.

genz10, 2017-07-01:
@shadow-1 Ah, that seems more practical, glad to hear it. Good luck on your development!

shadow-1, 2017-07-10:
@genz10 
I have released a new beta version of yi-hack-v3 (0.1.4-beta2). The firmware is now no longer affected by Xiaomi firmware updates.

Luis2u4u, 2017-07-10:
So now we can update the firmware of the camera using the app and all the settings and the hack will remain the same? Even with chinese version of the cameras that will not work outside China without the hack? Thanks.

shadow-1, 2017-07-10:
@Luis2u4u 
That is absolutely correct. I have tested it by downgrading the firmware using one of the official Xiaomi firmware releases before upgrading back to the latest. yi-hack-v3 was unaffected by the downgrade nor the upgrade.

closed 2017-08-07 by shadow-1