shadow-1/yi-hack-v3 #31: Does the proxy need to be alive for the entire time or only when initial conection ?

genz10, 2017-06-27:
Hi,

Does the proxy need to be alive for the entire time or only when initial conection ?

What happen if the proxy is dead when we use the cam ?

Thanks

shadow-1, 2017-06-28:
@genz10 
The proxy server does not need to be alive for the entire time. The proxy server only needs to be alive when the camera does a region check. This occurs when the camera boots up. After that it appears that the camera does a region check after roughly 3-4 days of inactivity (the camera is not accessed through the smartphone app for roughly 3-4 days). This is a bit hard to determine.

If the proxy server goes dead but the region check has already occurred. The camera will still be fully functional until the next time the region check occurs. For most people this will be when the camera is rebooted.

genz10, 2017-06-28:
"The camera will still be fully functional until the next time the region check occurs. "

So if there 2 consecutive region check failure the camera can't be use?

"After that it appears that the camera does a region check after roughly 3-4 days of inactivity (the camera is not accessed through the smartphone app for roughly 3-4 days). This is a bit hard to determine"

So if the camera active for the entire time (no inactivity) there won't be another region checking?

Thanks for your answer :)

shadow-1, 2017-06-29:
@genz10 
Not quite, if the region check fails and detects that the camera is not located within Mainland China. The camera will not work and the annoying ban message will play. However if the proxy servers are down and the region check occurs. The camera will fail to connect to the Xiaomi servers. The camera will try to connect through the proxy server, however it will timeout. The ban message will not play. However the camera will not work either.

As far as I can tell. If the camera is active the whole time there will not be another region check.

genz10, 2017-06-29:
All right, thanks for your answer :+1:

markusd1984, 2017-10-30:
@shadow-1

Would it be possible to schedule an automatic regular updating of the proxy server list to help avoid this, to make sure there are at least 2-3 active proxies saved?

Not sure I understand when the region check could fail while also detecting the camera's location to be outside of Mainland China? How is this different from proxy being down?

In either case, will it continue to try the region check or will we need to manually reboot / update the proxy list to get the camera working again?

shadow-1, 2017-10-31:
@markusd1984 
Automatic regular updating of the proxy servers will likely come in the future.

I am not sure exactly when the camera checks the region it is operating in other than startup. It seems to be linked to the amount of time that the camera is inactive (accessed through the app). I have found that after roughly 3 days of inactivity (not access the camera through the app for 3 days), the camera performs a region check. If I access the camera through the app say every day, the camera never appears to perform a region check other than initial startup.

The difference between not having a proxy (or one that is not based in Mainland China) is that when the camera (well the Xiaomi server to be more accurate) will detect that the camera is being operated outside of Mainland China, it will instruct the camera to lock down and play the annoying ban message.

However if the proxy server is down, the camera is unable to connect to the Xiaomi servers. As a connection could never get established with the server, the camera never receives a response and keeps on retrying to connect to the Xiaomi servers. Unfortunately this means that the camera will stop working through the app. However the camera will not be locked down and the camera will not play the ban message. As soon as a working proxy server is saved in the ProxyChains-ng config file (or the current proxy server becomes available again), connection with the Xiaomi server will be able to be reestablished and the camera will continue working again seamlessly.