shadow-1/yi-hack-v3 #32: Save to SD card

johnnyxvii, 2017-06-27:
Hello,

First off, thank you for taking the time to create the hack and to keep updating it. It's really appreciated.

I successfully hacked two 360 Dome Cameras (Black) and they work fine when viewing real-time. However, I noticed that the cameras are not saving anything to the SD card. Normally, with older model cameras as my reference, the cameras will keep a rolling recording on the SD card, so you can track back as needed.

Is there something I need to configure in regards to the hack to get this working?

Thanks.

shadow-1, 2017-06-28:
@johnnyxvii 
This functionality is part of the proprietary Xiaomi programs running on the camera. It is configured through the smartphone app.

There is an option "Activity Detection Recording" in the app which needs to be enabled. I have had an issue once where the camera would not record to the microSD card. I fixed the problem by formatting the microSD card through the smartphone app.

shadow-1, 2017-08-07:
@johnnyxvii 
I assume this issue has been resolved. Feel free to open a new issue if this is still a problem.

closed 2017-08-07 by shadow-1

Shuunen, 2018-02-12:
FYI : I changed to another sd card & did like @shadow-1 said , I formated the microSD card through the Yi smartphone app and it worked 👍 

Thanks 😄