shadow-1/yi-hack-v3 #33: Time zone setting

lezik, 2017-07-01:
Xiaomi Yi Dome 720 p

How to set your time zone?
For your country

lezik, 2017-07-01:
Need adjusting the time display

shadow-1, 2017-07-01:
@lezik 
Through my experimentation when developing the region hack. I can confirm that the cameras timezone information comes from the Xiaomi servers. It is not based on the system timezone on the camera.

What is very annoying is that the International version of the smartphone app has the ability to change the timezone on the camera. However the Chinese version of the smartphone app is has the timezone locked to Shanghai, China timezone (GMT+08:00).

It will require some work, however it may be possible to intercept the the timezone that gets returned from the Xiaomi server and modify it before it gets used by the programs running on the camera. However I believe this will cause lots of problems. I believe the smartphone app will be using Shanghai time whilst the camera will be using your modified timezone. Therefore the app will have recorded events that occurred in the 'past' or in the 'future' (depending on the timezone you setup).

markusd1984, 2017-10-30:
@shadow-1
Might you have already reviewed the [ntpclient](https://github.com/fritz-smh/yi-hack) solution from fritz-smh or others who are using the [equip_test.sh](http://robinhenniges.com/en/fix-xiaomi-xiaoyi-small-ant) to modify the time zone? Wondering if this could help.

I understand there is a difference between the timezone used to log the videos to work with alerts and built-in app timeline vs the timestamp put into the video itself. 

The visual timestamp in the video would be really great to have adjusted as it remains with the video vs the timestamp in the file properties that might be lost when copying files.

Besides, I take that it's not possible or too complex to get the international app working with the Chinese camera.

shadow-1, 2017-10-31:
@markusd1984 
I did a quick experiment in the past and concluded that changing the timezone on the camera using this method does in fact change the visual timestamp on the video feed (it was some time ago now).

However I ended up having all sorts of issues with the Xiaomi app. Because the timezone on the camera did not match the timezone of the app (which is locked to GMT+8 Hours - Shanghai Time for the Chinese version). The timeline in the app and alerts were all over the place. I had alerts from the future or from the past in the app and could not use the timeline as all my recorded events were shown from the future or the past.

The time in the app was offset by the difference between GMT+8 and the timezone I setup on the camera. That is why I was wary of providing such an option in the firmware before something like RTSP gets developed.

To be fair, I have not done a lot of experimenting with timezones on the camera. I am not aware of a way to change the time that shown in the video feed without affecting the app.

It would take a fair bit of hacking of the Xiaomi apps running on the camera to determine whether the Chinese version of the camera can be made to work with the International version of the app. I have not had the time to experiment with this.