shadow-1/yi-hack-v3 #34: Does this support the Yi Home 720p camera?

untotren, 2017-07-04:
https://www.amazon.com/gp/product/B016F3M7OM/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

shadow-1, 2017-07-04:
@untotren 
This is a repeat of the following issue/request: https://github.com/shadow-1/yi-hack-v3/issues/25

At the moment, the original Yi Home is not supported. However I will probably make a build for it.
For the time being, only the Yi 1080p Home is supported. I am not aware of any other hack that is able to get around the region locking on the Yi Dome series and Yi 1080p Home cameras. Getting around the region locking on the original Yi Home cameras is much more straight forward.

There are other hacks available with RTSP support for the original Yi Home camera. However you will need to downgrade to an older version of the firmware. In addition, I am not sure whether you will be able to upgrade to a newer firmware version without causing issues.

untotren, 2017-07-06:
@shadow-1 I have 6 of these and the privacy concerns honestly scare me greatly as we have many unwanted visitors.  If you could do this it would be incredibly appreciated.  (Abusive ex to the children who is an IT security guru :/)  When you have time, if you are able to, this would mean the world to me.

shadow-1, 2017-07-08:
@untotren 
I will try to create an initial version of this firmware. For RTSP, I will have to borrow an implementation from one of the other projects.

untotren, 2017-07-11:
HTTP is fine too!  Anything to use it with something like iSpy would be amazing!

weedy171, 2017-07-19:
I bought a 720p YI home and accidentally updated it and been unable to downgrade it. Anyone have ideas how to downgrade to enable the RTSC feeds. Thanks for the help in advance!

shadow-1, 2017-07-19:
I have downloaded a number of different firmware for the original Yi Home and have hosted them:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

They are under the directory `Official Firmware --> Yi Home`

weedy171, 2017-07-22:
Is their a firmware version that you recommend I try to downgrade from the lastest. Thanks !

shadow-1, 2017-07-23:
@weedy171 
I'm not sure which version of the firmware stopped providing the ability to run a script from microSD card. However as a first go, I would try 1.8.6.1B.

untotren, 2017-10-19:
@shadow-1  Any update on this?  I wait patiently :)

shadow-1, 2017-10-20:
@untotren 
I have recompiled almost everything for the Yi Home camera. I still have a bit of work with recompiling BusyBox. Some updates will be required to the scripts and the web server as the partitions are laid out differently with the original Yi Home.

untotren, 2017-10-22:
:D

frekel, 2017-11-28:
@shadow-1:
I just received 2 cameras. Unfortunality with the same above problems..... Any progress?
(or a paypal link to give you some beers ;)

Greets,
Frank

shadow-1, 2017-11-28:
@frekel 
I have made some progress and you can see where I got to in this thread. https://github.com/shadow-1/yi-hack-v3/issues/25 It still needs a little work. 

By the way, if you just received Yi Home cameras (and they are new), there is a good chance that they are the 17CN or 27US versions rather than the original Yi Home cameras which are different to the original Yi Home.

frekel, 2017-11-28:
@shadow-1:

1. How can I see if my camera is the 17CN or 27US version? The camera is labeled: yhs-113-ir
2. I will follow #25 
3. Still missing the paypal link for some beers ;)

Thanks again,
Frank

shadow-1, 2017-11-28:
@frekel 
Just have a look at the serial number at the bottom of the camera. Refer to the picture below:
[image]

frekel, 2017-11-28:
Yippie! I have a 27US!
I am goind to read the readme file ;)

shadow-1, 2017-11-30:
@frekel 
Just make sure you use the firmware file from this post: https://github.com/shadow-1/yi-hack-v3/issues/40#issuecomment-315593035

I have not officially released the 27US compatible firmware yet. Probably should release what I have done so far to stop confusion.

frekel, 2017-11-30:
@shadow-1:  I had tried to use the releases but those failed. After reading #40 I found the correct firmware files and have a working 27US cam!