shadow-1/yi-hack-v3 #35: Auto turn on and off alerts of motion detection

S0lmyr, 2017-07-04:
Hello,
Is there any option to turn on and off alerts of motion detection when my mobile phone (android) is on particular wifi.
The time customization of alerts built in app isnt the best for me..

I have an idea, when my phone connects to my wifi, it will send command via telnet (or...?) to activate motion detection.. It is even possible? I tried connect to telnet:

Built-in commands:
------------------
        . : [ [[ alias bg break cd chdir command continue echo eval exec
        exit export false fg getopts hash help history jobs kill let
        local printf pwd read readonly return set shift source test times
        trap true type ulimit umask unalias unset wait

Not really wise of those :)

Thank You

shadow-1, 2017-07-08:
@S0lmyr 
Motion detection is handled by the proprietary Xiaomi programs running on the camera. Is there a reason you do not want to disable "Activity Detection Recording" in the app manually?

To do what you want, you will have to develop some smarts to get your phone to automatically execute some commands when connected to your home WiFi and when you leave your home WiFi (most likely through Telnet), although there are many ways you can tackle this problem if you are a keen developer.

I have not tried this. However as an experiment, you can try running these commands on your camera through Telnet and see if it stops motion detection.
```
killall watch_process
killall mp4_record
```
This should stop the camera from recording motion detection activities on the camera.

To re-enable, run the following commands through Telnet:
```
cd /home/app
./mp4_record &
./watch_process &
```

S0lmyr, 2017-07-19:
Thank You for reply.

The reason for this is due to notification on android. I do not want hundreds of totifications when I am at home (my mobile phone connected to my wifi)

I tried it via telnet. but with no success. 
after: `killall watch_process` there was no reply from command line
and after `killall mp4_record` the reply from command line: killall: mp4_record: no process killed
any suggestion?

I checked the android app, but there was no change in "Smart detection" menu..
Are there any documentations with possible commands on telnet for Yi Dome?

Thank You a lot for Your time.

shadow-1, 2017-07-22:
Most Linux commands will work through Telnet. The main issue is that the Xiaomi programs running on the camera are undocumented. So it will take quite a bit of experimentation to work something like this out.

To do what you want through the official app. You need to turn off Motion Detection. This will disable notifications but the camera will still record to microSD card.
[image]