shadow-1/yi-hack-v3 #37: Trigger "only can be used in china""

genz10, 2017-07-11:
hi, my yi dom sometimes triggered only can be used within china, but when i restart the camera it works perfectly again. I have check the proxy it is alive

shadow-1, 2017-07-11:
@genz10 
Do you have the Yi Dome or the Yi 1080p Dome? I noticed I made a mistake with the Yi 1080p Dome camera firmware and the region hack did not work properly. However I have now corrected it (15 minutes ago or so).

---

If it is working sometimes and not other times. It is likely that at least one of your configured proxy servers is not actually located within Mainland China. With the default configuration, each time the camera connects to the Xiaomi server, a proxy server is chosen at random from your list. So if one of your proxy servers is not located within Mainland China, it makes sense that the check will fail occasionally but work most of the time.

Keep in mind that the built-in test is a good indicator. However it is not perfect.

---

I have also noticed that there are issues with some HTTP proxy servers (even if they pass the built-in test). In the latest version of the firmware, I have provided an option to download and test only SOCKS proxy servers. This should be more reliable.

Let me know if this corrects the issue.

genz10, 2017-07-11:
I have Yi Dome (720) not 1080 one. I can make sure that every proxy is located on mainland (I use 3rd party to check the location got 5 proxy as backup)

Let me try using the latest release firmware, will post update here.

genz10, 2017-07-12:
latest release version cannot be use in yi dome (home_v201) i dont know why

shadow-1, 2017-07-12:
@genz10 
What do you mean you cannot use the latest release?
Does the firmware not work at all? Or is it just the region hack that is not working?

This is the one camera I can test and it appears to be working properly for me.
Maybe try downgrading to an old version of the firmware before upgrading again to the latest version.
This usually fixes any potential problems during the flashing process.

genz10, 2017-07-13:
I already give another try to flash it from begining, still the firmware doesnt work at all

shadow-1, 2017-07-13:
@genz10 
That is very strange. I have the same camera and used exactly the same files as hosted on GitHub and it works properly for me.

One thing I have had to do a few times and something you can try is format the microSD card before copying the latest firmware to flash. I always use the official SD Card Formatter tool. It is available from here:
https://www.sdcard.org/downloads/formatter_4/

When you flash the newest firmware, does it go into a reboot loop?

genz10, 2017-07-15:
Yes i've tried to format sd card, re download rootfs and home_v201 and include.startup.sh in those sd card, but it seems doesnt work at all i dont know why.

shadow-1, 2017-07-15:
@genz10 
I am very puzzled now. I am running out of ideas on what the issue could be.

Do you happen to have serial port access on your camera? Through serial port, we will be able to see exactly what is happening on the camera and the error message that will be written through the serial port will give me an idea on what the issue might be.

genz10, 2017-07-16:
@shadow-1 

Here's my startup.sh , It will replace proxychain.conf and put some socks there

```
#!/bin/sh
echo "random_chain" > /home/app/proxychains.conf

echo "chain_len = 1" >> /home/app/proxychains.conf

echo "tcp_read_time_out 15000" >> /home/app/proxychains.conf

echo "tcp_connect_time_out 8000" >> /home/app/proxychains.conf

echo "[ProxyList]" >> /home/app/proxychains.conf

echo "socks5 XXX.XXX.XXX.XXX PORT" >> /home/app/proxychains.conf
```

Is it fine ?

shadow-1, 2017-07-16:
@genz10 
That looks good for firmware 0.1.4-beta1 and older. I have changed the location of proxychains.conf in 0.1.4-beta2 to ensure that the configuration is not affected by firmware updates. The new location is `/home/yi-hack-v3/etc/proxychains.conf`.

You can simplify your script a little bit like this:
```
#!/bin/sh

cat > /home/app/proxychains.conf << EOL
random_chain
chain_len = 1
tcp_read_time_out 15000
tcp_connect_time_out 8000
[ProxyList]
socks5 XXX.XXX.XXX.XXX PORT
EOL
```

genz10, 2017-07-16:
@shadow-1 

I've try using the new location, but it seems startup.sh wasn't executed, I using FTP protocol to see proxychains.conf and it doens't change as intended (still default configuration).

shadow-1, 2017-07-16:
@genz10 
Try executing the script manually through telnet to check that it works properly. Enter the following command in Telnet:
```
/tmp/sd/startup.sh
```

If it doesn't work when executed manually, then there is a problem with your script. However if it works properly when executed manually, I have to look into why it is not executed at startup as it should.

shadow-1, 2017-07-16:
@genz10 
Yes it will be corrected in the next release.
In the mean time, you can do the mod described above and it will work as it should.

genz10, 2017-07-16:
@shadow-1 

After I follow your instruction, yi dome usually rotate twice (to the right and then back to left) but now it won't rotate back to the left and stop working, when I try to flash using beta1 firmware, it works again. Does your instruction triggered another error or create another bug that makes firmware crash ?

shadow-1, 2017-07-16:
@genz10 
Are you sure you didn't inadvertently change something else within `/home/base/init.sh` or `/home/app/init.sh`?

The change from `/home/base/init.sh &` to `/home/base/init.sh` does not change anything that gets executed upon startup.

---

On a side note, the first time the camera starts up after flashing 0.1.4-beta2. The camera will move to the right for longer than usual, you may just have to be patient whilst the camera initialises for the first time.

Perhaps I should stop the camera from moving early when the camera initialises for the first time.

genz10, 2017-07-16:
@shadow-1 

Yes I can make sure the only thing I removed is only the fork command,

I've been waiting for 10 minutes and the yi dome still not moving to the left.

shadow-1, 2017-07-16:
@genz10 
I am really really puzzled now.
The new 0.1.4-beta2 does not work properly on your camera, yet it works perfectly for me with the same camera. However 0.1.4-beta1 works properly.

There is no change in the base firmware between the two versions as well.
I assume you do not have serial port access.

I think I will recreate the firmware again from scratch to see if that works. I am a bit blind as I can not recreate the issue.

genz10, 2017-07-16:
If i dont remove the fork syntax, it works but i must manually run the startup script. By removing fork syntax it makes the camera wont move to the left after move to right

shadow-1, 2017-07-16:
@genz10 
I find that really really odd.
I have no idea why removing the fork affects the start-up of the camera.

Again my camera starts up perfectly with the fork removed.

Another way of getting around the issue (although this is more of a hack than a solution) is to introduce a delay into `/etc/init.d/S20yi-hack-v3`.

I have tested the following at it works for me. Modify `/etc/init.d/S20yi-hack-v3` as follows:
```
#!/bin/sh

sleep 2
/home/yi-hack-v3/script/system.sh
```

Feel free to increase the sleep time if necessary.

The idea is to pause the execution of the script until the camera has progressed far enough through the boot sequence that the microSD card is mounted.

genz10, 2017-07-17:
For now I'm still use beta-1 and i'll try the next release firmware.

Thanks!

shadow-1, 2017-07-18:
Looks like I need to come up with a cleverer way to ensure that the startup script executes after the camera has mounted the microSD card.

genz10, 2017-07-21:
@shadow-1 

Do you need help?

genz10, 2017-07-23:
@shadow-1 

Is there any updates here ? or anything that I can help you ?