shadow-1/yi-hack-v3 #39: How to identify Yi-Home 17CN camera?

vaidyasr, 2017-07-14:
I would like to use the latest v3 hack to my existing 2 720p cameras. How do I make sure those are Yi-Home 17CN cameras?. The cpuinfo is below

```
Processor       : ARM926EJ-S rev 5 (v5l)                                                                                                
BogoMIPS        : 218.72                                                                                                                
Features        : swp half thumb fastmult edsp java                                                                                     
CPU implementer : 0x41                                                                                                                  
CPU architecture: 5TEJ                                                                                                                  
CPU variant     : 0x0                                                                                                                   
CPU part        : 0x926                                                                                                                 
CPU revision    : 5                                                                                                                     
                                                                                                                                        
Hardware        : hi3518                                                                                                                
Revision        : 0000                                                                                                                  
Serial          : 0000000000000000
```

cnberg, 2017-07-14:
Take a look at the QR code behind your camera, on top of it there is a
serial number, start with 12 or 17. If it is 17, you got a 17CN camera.

--------------
Yours sincerely
berg
http://cnberg.com
@cnberg

shadow-1, 2017-07-14:
@vaidyasr 
@cnberg is absolutely correct. If you look at the bottom of your camera, there is a serial number. If you have a 17CN camera, the serial number will start with 17CN.
[image]

The Hardware type for the original Yi Home is: `hi3518` while the hardware type for the Yi Home 17CN is: `hi3518ev200`.

To help me develop a firmware compatible with the original Yi Home, you can provide me with a backup of all the partitions on your camera. To do this, execute the following commands on your camera with a microSD card inserted:
```
dd if=/dev/mtd0 of=/home/hd1/mtd0
dd if=/dev/mtd1 of=/home/hd1/mtd1
dd if=/dev/mtd2 of=/home/hd1/mtd2
dd if=/dev/mtd3 of=/home/hd1/mtd3
dd if=/dev/mtd4 of=/home/hd1/mtd4
dd if=/dev/mtd5 of=/home/hd1/mtd5
dd if=/dev/mtd6 of=/home/hd1/mtd6
dd if=/dev/mtd7 of=/home/hd1/mtd7
```

It will take me a bit longer to create a firmware compatible with the original Yi Home. The layout of the filesystem is different to all the other cameras which I support. In addition, because it uses a different chipset, I will need to use a different compiler to compile programs for the original Yi Home.

closed 2017-07-22 by shadow-1

vaidyasr, 2017-07-25:
Sorry for the delay. I have taken the backup of the file system you have mentioned, except mtd7 which does not exist.
[yi_mtd0-6.zip](https://github.com/shadow-1/yi-hack-v3/files/1172186/yi_mtd0-6.zip)

zhouyuwei128, 2017-12-17:
Hello, i got a Yi Home camera with back SN of WNLG2PSYX9 . Is it compatible with v3 hack?

shadow-1, 2017-12-17:
@zhouyuwei128 
The serial number does not look familiar to me. Can you post a photo of the front of your camera along with the label where the serial number is.

From that I may be able to tell.

lance36, 2018-05-31:
> Take a look at the QR code behind your camera, on top of it there is a
> serial number, start with 12 or 17. If it is 17, you got a 17CN camera.
> …

this should be in the README, i was looking at the model number and was cursing about buying the wrong model :)

freeman974, 2019-01-27:
Bonjour, 

all this sounds great except that I have the original hi3518 version.
I made a backup of the partitions.
[yi_home_hi3518.tar.gz](https://github.com/shadow-1/yi-hack-v3/files/2799847/yi_home_hi3518.tar.gz)

You're still working on it.

Redwid, 2020-02-13:
So yi-hack-v3 wont work on Hardware: hi3518?
Non of compiled binary as well?