shadow-1/yi-hack-v3 #40: 27US almost works, problems with WPA crashing

yaddatrance, 2017-07-15:
The latest beta https://github.com/shadow-1/yi-hack-v3/releases/tag/0.1.4-beta2 got me the furthest on my 27US model camera.

It scans the QR code properly, setting the right info into WPA_supplicant.conf but wpa crashes with a log file named log_no_wifi_dev.tar.gz which to my untrained eye sounds like it can't find the wifi device?

Its so close to working! Anybody have any ideas or tips?

shadow-1, 2017-07-15:
@yaddatrance 
What camera are you using with yi-hack-v3? I am not familiar with the serial numbers for all the cameras.
Have you tried to install an older version of yi-hack-v3?
Does is work properly when going back to stock firmware (using recovery image)?

When pairing the camera to your smartphone, the camera provides voice prompts and after connecting to WiFi, the camera says "WiFi is connected". What does your camera say during pairing?

If it works properly on an older release and just stopped working in this latest beta version, we should be able to isolate what the issue is.

yaddatrance, 2017-07-15:
The US27 is the latest US/international version which uses the v200 chip similar to the 17CN. It works properly when restored using the hizel restore image https://github.com/hizel/yi_dump. It doesn't get to "WiFi is connected" it says "Connecting to Wifi" then turns amber and after a while goes back to "Waiting to Connect"

Log file on sd card shows

[./rmm][1/19/6:48:54:551]: msg snd success[./rmm][1/19/6:48:54:575]: set ai volume via himm 0x201200d8 0x1e1ec001[./rmm][1/19/6:48:54:631]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/19/6:48:54:685]: set ao volume via himm 0x201200d4 0x03122424[./rmm][1/19/6:48:55:901]: got RMM_SPEAK_WAIT[./rmm][1/19/6:49:0:405]: decoded QR-Code symbol "b=USb5WaySAqL6qpS6&s=Y2FyYm9u&p=VFAsIz8DCV0kPAQN"
[./rmm][1/19/6:49:0:406]: trans_json info=b=USb5WaySAqL6qpS6&s=Y2FyYm9u&p=VFAsIz8DCV0kPAQN;key=&p=
[./rmm][1/19/6:49:0:407]: trans_json result=VFAsIz8DCV0kPAQN
[./rmm][1/19/6:49:0:408]: trans_json info=b=USb5WaySAqL6qpS6&s=Y2FyYm9u&p=VFAsIz8DCV0kPAQN;key=&s=
[./rmm][1/19/6:49:0:409]: trans_json result=Y2FyYm9u
[./rmm][1/19/6:49:0:409]: trans_json info=b=USb5WaySAqL6qpS6&s=Y2FyYm9u&p=VFAsIz8DCV0kPAQN;key=b=
[./rmm][1/19/6:49:0:411]: trans_json result=USb5WaySAqL6qpS6
[./rmm][1/19/6:49:0:412]: rmm got /tmp/got_wpa ssid(carbon) pwd(lifelifelife) bind(USb5WaySAqL6qpS6)
[./dispatch][1/19/6:49:0:418]: invalid msg 0x1005
[./rmm][1/19/6:49:0:419]: msg snd success[./dispatch][1/19/6:49:0:421]: in choose_server, region_id = 17, api_server = https://api.us.xiaoyi.com, sname = familymonitor-y18, dlproto = mius
[./rmm][1/19/6:49:0:421]: msg snd success[./rmm][1/19/6:49:0:496]: set ai volume via himm 0x201200d8 0x1e1ec001[./dispatch][1/19/6:49:0:519]: msg snd success[./dispatch][1/19/6:49:0:520]: invalid msg 0x1004
[./rmm][1/19/6:49:0:527]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/19/6:49:0:556]: set ao volume via himm 0x201200d4 0x03122424[./dispatch][1/19/6:49:1:496]: wifi disconnected, now reconnect wifi
[./watch_process][1/19/6:49:2:6]: server crashed![./rmm][1/19/6:49:2:612]: HI_MPI_ADEC_SendStream ret 0xa0188040[./rmm][1/19/6:49:2:612]: got RMM_SPEAK_SCAN_OK[./rmm][1/19/6:49:2:634]: set ai volume via himm 0x201200d8 0x1e1ec001[./rmm][1/19/6:49:2:660]: set ai volume via himm 0x201200c8 0x23c2e[./rmm][1/19/6:49:2:683]: set ao volume via himm 0x201200d4 0x03122424[./rmm][1/19/6:49:4:572]: HI_MPI_ADEC_SendStream ret 0xa0188040[./rmm][1/19/6:49:4:573]: got RMM_SPEAK_CONNECTTING[./dispatch][1/19/6:49:9:876]: wpa may crashed, now reset
[./dispatch][1/19/6:49:9:914]: no usb dev

lsmod.txt shows

Module                  Size  Used by    Tainted: P  
pid_list                1383  1 
watchdog                4777  0 
hi_cipher              74214  0 
hi_mipi                22525  0 
hi3518e_adec            6546  1 
hi3518e_aenc           47788  1 
hi3518e_ao            204318  1 
hi3518e_ai            208635  2 hi3518e_aenc
hi3518e_aio            28507  0 
acodec                  8218  0 
pwm                     2471  0 
sensor_i2c              1878  0 
hi3518e_jpege          51857  0 
hi3518e_h264e         175257  0 
hi3518e_chnl           33635  0 
hi3518e_venc          221695  6 hi3518e_jpege,hi3518e_h264e
hi3518e_rc             72053  0 
hi3518e_vpss          252336  4 
hi3518e_viu           257547  3 hi3518e_vpss
hi3518e_isp           111397  5 sensor_i2c,hi3518e_vpss,hi3518e_viu
hi3518e_region         66802  0 
hi3518e_tde           121887  1 hi3518e_region
hi3518e_sys            63731  3 hi3518e_vpss,hi3518e_viu
hi3518e_base           58704 18 hi3518e_adec,hi3518e_aenc,hi3518e_ao,hi3518e_ai,hi3518e_aio,acodec,hi3518e_jpege,hi3518e_h264e,hi3518e_chnl,hi3518e_venc,hi3518e_rc,hi3518e_vpss,hi3518e_viu,hi3518e_isp,hi3518e_region,hi3518e_tde,hi3518e_sys
hi_media                5801 23 hi_mipi,acodec,pwm,hi3518e_tde,hi3518e_base
mmz                    22641 10 hi_cipher,hi3518e_aenc,hi3518e_ao,hi3518e_jpege,hi3518e_h264e,hi3518e_viu,hi3518e_tde,hi3518e_sys,hi3518e_base
cpld_periph             6277  2 
hi_rtc                  6086  0 
exfat                  94437  0

shadow-1, 2017-07-15:
@yaddatrance 
Ah I was not aware. Xiaomi don't even have the firmware files for this camera on their website yet!
I will have to examine the differences between the 27US version and the 17CN version.

Interesting that they both share the same firmware filename and the same firmware version. I am thinking that the 27US version and the 17CN version are exactly the same in terms of the firmware.

I will check what are the differences between Yi Home 17CN and Yi Home 27US and create a fresh pair of firmware images if they should be exactly the same.

yaddatrance, 2017-07-15:
Awesome! Let me know if there's anything I can do to help.

arunshivaram, 2017-07-15:
@shadow-1 : I have a 27US model as well and the fritz-smh version did not work. I gathered your beta for 17CN was the closest possible hack we have for 27US (as they share the same chip hi3518ev200 as @yaddatrance  mentioned above). I will try the beta on my 27US and report back. Meanwhile,I believe you can find the firmware for the 27US on the xiaomi website here if you need it to compare with the 17CN version : http://download.us.xiaoyi.com/yifirmware/smarthomecam/1.8.7.0A_201702081101home

shadow-1, 2017-07-16:
@arunshivaram 
The link you provided appears to be for the original Yi Home and not for the Yi Home 27US. It appears that Xiaomi have not release a firmware update yet for the Yi Home 27US and everyone is still running on the initial firmware release.

@yaddatrance 
I had a look at the differences between the two firmwares and found that they are almost identical. The reason why WiFi fails on the firmware designed for the Yi Home 17CN is because the 27US version uses a different WiFi chip to the 17CN. The 17CN camera does not come with the WiFi driver that is compatible with the WiFi chip used on the 27US.

The other difference is the firmware version. The backup file you provided me is firmware version 1.8.7.0A whilst the Yi Home 17CN is up to firmware version 1.8.7.0C.

Considering 1.8.7.0C is available for download from the Xiaomi website (although from their Chinese page) and it uses the same firmware filename as the 27US. I have created a combined firmware that is hopefully compatible with both the Yi Home 17CN and Yi Home 27US.

Please try my test firmware. It would be a great help if you could provide feedback as it is a bit modified to the original. I'm not sure if the 27US camera will work properly when loaded with the 1.8.7.0C base firmware. However WiFi should be working along with telnet server, ftp server and web server.

Please bear in mind that I developed this firmware primarily to get around region locking limitations imposed on the Chinese version of the camera. The camera will not complete the pairing process until ProxyChains-ng is disabled in the web interface. For further details, refer to the getting started guide on the main GitHub page. https://github.com/shadow-1/yi-hack-v3#getting-started---step-by-step-guide

home: https://www.dropbox.com/s/lxwxvm1haydkqmb/home_y18?dl=0
rootfs: https://www.dropbox.com/s/w37nkc7bz0isv2o/rootfs_y18?dl=0

I have also created recovery images to go back to stock firmware:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

retyre, 2017-07-16:
Thanks, @shadow-1. WiFi is working now along with telnet/ftp/web server. No video through RTSP, though (tinycam picks up the device through scan and adds it), but there's no video. Nothing through VLC either.

Any suggestions?

shadow-1, 2017-07-16:
@retyre 
Good to hear that the test build works. Does the official app work with the camera?

At the moment, RTSP support has not been developed. It is currently a work in progress. I am not aware of any RTSP implementation that works flawlessly on any Yi camera based on Hi3518eV200 chipset.

retyre, 2017-07-16:
@shadow-1 Yes, the official app works with the camera. It pops up a message about a new firmware being available (1.8.7.0A), but who needs that.

Yes, I read about RTSP support being a WIP. No problem. Thank you for getting the firmware up and running so quickly.

shadow-1, 2017-07-16:
@retyre 
Just for your information, it is safe to perform firmware updates on this release of the firmware. :smile:

arunshivaram, 2017-07-17:
Thanks @shadow-1. I can also confirm that the test firmware you linked above worked as intended on my 27US version - telnet, http, ftp all up and running.

retyre, 2017-07-17:
@shadow-1 Do you have any pointers on how to experiment with rtsp? I saw your post (on another thread) about killing watch_process and others and trying one of the two rtsp builds you had linked. How does one do the latter? Editing equip_test.sh and running it with the correct call to the rtspsvr? Will equip_test run automatically (or manually) with your current firmware? Thanks.

yaddatrance, 2017-07-17:
Yay it works for me as well! Thanks!

mrbobo555, 2017-07-17:
@shadow-1, it works for me too, thanks! I wanna ask, does yi-hack-v3 support "no more cloud feature (nothing goes out of your local network)"(as it was in fritz-smh's version)?

shadow-1, 2017-07-18:
@arunshivaram @retyre @yaddatrance @mrbobo555 
Glad that the initial test firmware works as expected. In the next firmware release, support for the 27US version of the camera will be officially added.

@retyre @mrbobo555 
As mentioned in the other thread, it is possible to experiment with the initial RTSP release on this project:
https://github.com/xmflsct/yi-hack-1080p

It is not completely stable yet and you will have to disable cloud features on the camera to make it work.
When I have a bit of free time, I will create a simple script that will kill the Xiaomi programs running on the camera and launch this program.

I am currently in the process of developing this to be more stable and to be able to work in conjunction with the official Xiaomi programs running on the camera. However I need to work out how the official programs work which will take me a bit of time. However I do have some ideas.

Anyone who is a keen developer is welcome to help. 😄

yaddatrance, 2017-07-19:
So is the idea that if I kill those factory processes that the rtsp2301 binary will grab the correct video device and allow streaming rtsp?

Minims, 2017-07-19:
@shadow-1 Thanks for you work, i have restored my 27US cam with you firmware. The App works great.
I'm looking forward to the version that will support the stream video.

retyre, 2017-07-19:
@yaddatrance That's what rtsp2301 is supposed to do. It needs libsns_f22.so and/or libstdc++.so.6 in /home/lib. I found the binaries and the sources here: https://github.com/xmflsct/yi-hack-1080p/issues/5.

When I run rtsp2301, it reports a couple of errors but appears to stream; however,VLC picks up only a blank video. Could be a codec issue.

Please update with your findings. Thanks.

shadow-1, 2017-07-19:
@yaddatrance 
That is correct, you will need to kill the Xiaomi processes for that implementation to work.

@retyre 
I have not tested this RTSP implementation. Remember you will have to kill the Xiaomi programs running on the camera.

This RTSP implementation has only been tested on the Yi 1080p Home.
What could be the issue is the image sensor. From looking at the libraries, it appears that the 1080p versions of the camera use a different image sensor to the 720p versions of the camera.

It appears that the 1080p sensor is: Silicon Optronics JX-F22
It appears that the 720p sensor is: OmniVision OV9732

This RTSP implementation has only been compiled for use with the 1080p image sensor.
Unfortunately that means it needs to be recompiled to work with the 720p image sensor.

yaddatrance, 2017-07-21:
Same discovery here, it streams a clever blend of zeros.

retyre, 2017-07-21:
I was able to recompile rtsp2301 for OV9732. The new binary initiates the correct sensor, but with errors: 

[image]

Clearly, there are leftovers from the old code (likely this): 
[main.c.txt](https://github.com/shadow-1/yi-hack-v3/files/1164539/main.c.txt)

shadow-1, 2017-07-22:
@retyre 
Looks like you are on the right track! Some further tweaks look to be required for it to work with the 720p sensor.

kls629, 2017-07-24:
you guys are doing great work

shadow-1, 2017-07-25:
@kls629 
Thanks for the positive feedback. I have been trying to learn how the official app works, so I can create an RTSP implementation which will work with both the official app and RTSP at the same time.

I have been having a lot of trouble cross compiling various debug tools to examine the official programs. I have been progressing, however it is slow going for the time being.

0sm3l, 2017-08-29:
Just got two 27US cameras and got the same issue connecting but solved it using the firmware posted above. Following this thread for RTSP support which was the only reason I bought them (not knowing it was not supported 😭 ).

bmyphillic, 2017-09-05:
@shadow-1  Great work! Thank you for your support.
Just got four 27US cameras and Looking forward for RTSP support. :+1:

kruschman, 2017-10-24:
@shadow-1 Thanks for all of your work on these. Do you happen to have any updates on the RTSP support for the Yi Home 27US? I used Fritz's in the past but he is no longer developing and have newer cameras.

shadow-1, 2017-10-24:
@kruschman 
No update on RTSP support just yet. I have not had the time to develop the firmware lately.

I have done a bit of work for Yi Home camera support (the original 720p version) and that should be ready for testing/pre-release soon.

shadow-1, 2017-11-03:
@bartaspoz 
I can include the RTSP server developed by others with the firmware.

However be aware that the camera will not work with the smartphone app and the RTSP implementation at the same time. I am also not sure whether RTSP works with the newer firmware versions.

shadow-1, 2017-11-08:
@svtcontour 
If you have 27US version of the camera. You need to leave the filename home_y18 and rootfs_y18.
Remember you need to flash both files (home and rootfs).

If something has gone wrong, you can go back to stock firmware before flashing yi-hack-v3 again.

Also try a different microSD card. I have had issues with some particular cards which could not be resolved.

kruschman, 2017-11-08:
@shadow-1 Personally I would prefer the rack with RTSP support and no smartphone app support. Just like the fritz hack, I want to keep all video local only and not going out to Chinese servers.

bartaspoz, 2017-11-08:
@shadow-1 and I prefer to leave the App control. Had no Idea that adding RTSP will break that support. Maybe you could release two versions of firmware - one with RTSP and one without it.

shadow-1, 2017-11-10:
@kruschman @bartaspoz 
For RTSP support that I am planning, it should support RTSP and the app at the same time.
So if someone wants to disable the Yi Cloud support (which will also disable Xiaomi app support), it will be possible. If someone still wants to use the Xiaomi smartphone app, this will also be possible.

However at the moment, all RTSP implementations conflict with the Xiaomi programs running on the camera and consequently smartphone app support needs to be disabled.

kruschman, 2017-11-10:
@shadow-1 
Is there a firmware version out now that allows that RTSP support that you know of? I was under the impression that none currently existed for the new(er) 720p versions.

shadow-1, 2017-11-10:
@kruschman 
You are correct that no RTSP implementations have been developed for the newer 720p versions of the camera.

However an initial RTSP support has been developed for the Yi 1080p Home by others. The source code has been made available. So it is possible to modify the code slightly and recompile so that it will work with the 720p version of the camera.

There are many limitations with this implementation. However it is a working example.

emersonvier, 2017-12-10:
Hi I have one YHS-113-HR FW 1.8.7.0D_201708091510 / 27USYLB3LD.

I dont want brick my camera, so I prefer ask, I need to access this device by RTSP to see by VLC

I Need this file below from https://github.com/shadow-1/yi-hack-v3/releases/tag/0.1.4-beta2

home_y18 and rootfs_y18

Than follow instruction on this page https://github.com/shadow-1/yi-hack-v3

After what I need to do?

shadow-1, 2017-12-10:
@emersonvier
There is almost no risk in bricking a camera as I have created Recovery images to go back to stock firmware.

Unfortunetly RTSP support is still a work in progress. Refer to the following post: https://github.com/shadow-1/yi-hack-v3/issues/15

emersonvier, 2017-12-29:
@shadow-1 Thank you very munch, do you have any idea when the RTSP are will available?

shadow-1, 2017-12-30:
@emersonvier 
Unfortunately I am not willing to give a solid timeline (it is as if every timeline I have provided gets extended by some distance). I am only developing this firmware as a side project in my free time. RTSP support will take a bit of effort to implement.

However it is definitely on my radar.

emersonvier, 2018-01-05:
I am looking to buy the previous model 22US, if someone know please share.

Jesse75, 2018-03-18:
Sorry guys. I've been reading and reading and trying to figure this all out. I'm just confused. Does this work on the US27 firmware 1.8.7.0D_201708091510? Trying to get RTSP server running.

I also have 2 Dome. 13US? Firmware 1.9.1.0H_201708091518.

What options do I have for RTSP for either of these? Will the older yi hack work?

lance36, 2018-06-12:
> US27

where you able to use RSTP on US27? if so, how?

DarthMadeHer, 2018-07-29:
Thanks to shadow-1, I was able to get my US27 720p Home cam working thru the Yi app. There were a couple of quirks in my specific case (YMMV). 1st, I flashed the Recovry OEM firmware he keeps.
1) It took several resets before the cam would connect to wifi with the root_y18, home_y18 files (not sure if I used the beta or his latest). I checked the router for the IP address it got and thru the browser disabled ng-proxy.
2) For some reason I could not get my iPad to connect to the SSID after the cam joined the wifi network. When the cam was unplugged, the iPad could connect, but never both at the same time using a 2.4GHz network only.
3) I substituted a spare (newer) router as a test network and was finally able to pair them.

In my router I can see traffic going to an internet proxy server that just echoes "Hello World" back. So starting the camera up still requires web access?  I would be nice to cut that need off completely.

lance36 - a kludgie RSTP may be possible by adding Necromix's files onto the SD card with shadow-1's firmware substitute. It looks like the last 2 lines on Necromix's Github instructions might have to be run manually through telnet to start the stream (probably best, since it may not be ready for PrimeTime, so better to not have it auto-run in the shell script.)
Haven't tried the RTSP yet, since I just now got the shadow-1 hack working.

tommasoconti98, 2018-09-05:
Any news for us for the 27US version and the RTSP?

johnpdowling, 2018-09-24:
Here is what I have going with my 27US and Home Assistant, using just downloads across github between this hack and the 1080p hack. It's not finished yet, but there's almost light at the end of the tunnel for my instance.

I use the standard yi-hack-v3 from here. Its standard init calls /tmp/sd/yi-hack-v3/startup.sh so I created one to launch my apps. In testing I started killing everything with it like here (https://github.com/shadow-1/yi-hack-v3/issues/92#issuecomment-354794958), but now I just keep the part where only local traffic is allowed and launch sd card applications.

Within startup.sh, I call a variant of the alarm script (https://github.com/shadow-1/yi-hack-v3/issues/79#issuecomment-348161594) that upon motion detect/clear by the standard software, a MQTT message (https://github.com/shadow-1/yi-hack-v3/issues/130) is sent to HA.

After that I call a script to launch andy2301's rtsp2303 server. In testing I followed this 1080p rtsp thread: https://github.com/xmflsct/yi-hack-1080p/issues/5#issuecomment-298093437 and used andy's latest code in that thread to try various magic numbers for settings and see what would happen. To find starting numbers I wiki'd h.264 headers and hexdump-ed /tmp/view (the circular buffer file holding the 720p and 360p streams) and noticed a blank space around address 0x81b40. So calling his executable with a starting offset of 0, and playing around with the ending ("repeat") offset around 531264 and playing with the other variables I'm able to get an RTSP stream that mostly works. There are artifacts in the stream and it doesn't last forever, but since it's piggybacking on the standard firmware things like the camera switching to IR mode in low light still work.

It isn't perfect so there's still work for somebody to do. I haven't even attempted opening and building andy's code (just copied the /bin folder contents), but it sounds like some inroads were made into how yi packages the two streams into /tmp/view. I've even been able to pull out the 360p stream for a bit using that software, but that's even more unstable than the main stream.

Sygu00, 2018-10-29:
Hi @johnpdowling ;

couldo you share the scritp you use to comunicate via MQTT with HA? I'm struggling with this but cant move along.

regards.

Sergio

Arkady23, 2021-03-09:
> Any news for us for the 27US version and the RTSP?

Can dear roleoroleo tell you?