shadow-1/yi-hack-v3 #41: Does this hack work for my YI Home cameras?

Luis2u4u, 2017-07-21:
Hi. I have two YI Home 720p cameras, with those serial numbers (they're pretty old, I bought them 1-2 years ago):
1. 12CNBJOX18
2. 2FAPHFLRD8

Will yi-hack work for those cameras? For the moment I keep the 1.8.6.1B official firmware on both of them because I think that both are Chinese versions and I'm afraid that if I update the firmware they will no longer work. Thanks.

shadow-1, 2017-07-22:
@Luis2u4u 
At the moment, this hack doesn't support the original Yi Home. I have hosted a whole suite of firmware images for the original Yi Home. So you can always downgrade back to an older version if you wish.
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

Xiaomi only (relatively) recently started region locking Chinese cameras. Since your camera is quite old, there is a very good chance that it is not region locked despite being the Chinese version.

Luis2u4u, 2017-07-22:
So I can update to a newer official firmware without risking the camera not to work anymore? And if the new firmware does not work, are you sure that I can go back to the older working firmware? I ask you that because I read that some people who updated their cameras to new firmware versions could not revert back to older firmwares or even if they reverted the firmware it still gave the error about not working outside China.

And is there any chance you could someday make the yi-hack compatible with this type of camera? I have a USB To RS232 TTL adapter, can I use that to find out information about camera that you would need to make your firmware compatible with it? Thanks.

shadow-1, 2017-07-23:
@Luis2u4u 
I can not guarantee that the downgrade would work. However looking at the firmware files, it is very likely that there will be no issues.

I did a little bit of reading and it appears that Xiaomi started implementing region locks since firmware version 1.8.5.1L. Since you are already on a newer version and you are not having issues. It appears likely that your cameras are not region locked.

Yes, I can make my custom firmware compatible with this camera. If you have serial port access to the camera, you may be able to help me.

bachulator, 2017-10-24:
Another "lucky" user for older version of Yi ants (11CN, china version, not regionally locked). Any hope we could see firmware for these "dinosaurs"?  Have an access to the camera via telnet.

shadow-1, 2017-10-24:
@bachulator
Support for the original Yi Home is still a work in progress. It is not very far off from a pre-release.

mastix, 2017-11-08:
@shadow-1 Can you update this issue as soon as something testable is available? Just subscribed! +1