shadow-1/yi-hack-v3 #42: What cause slow "Retrieving Pairing Status" ?

genz10, 2017-07-23:
What cause slow "Retrieving Pairing Status" ? Even if we got nice working proxy, just for info I use beta-1 version. 

Thanks!

shadow-1, 2017-07-25:
@genz10 
Slow pairing process is likely due to slow proxy server or one of your proxy servers does not work properly (so there are lots of retries).
I have done further testing and it appears that there are reliability issues with HTTP proxy servers. Even if they pass testing, it is not guaranteed that it will work with the Xiaomi servers.

In the next beta release, I will probably include additional tests to be performed on the test page to make sure that the proxy server will work with the Xiaomi servers.

I have found SOCKS proxy servers to be much more reliable.

genz10, 2017-07-25:
@shadow-1 

I have high reliability SOCKS proxy server with static ip and port, maybe it can be useful for your development phase, please kindly send me an email at herzbazt@gmail.com and i'll send it to you.