shadow-1/yi-hack-v3 #44: Is It possible to support the latest firmware 2.1.0.0A_201703071456home_y20m for Yi 1080p Home?

skysider, 2017-08-05:
I work on the latest firmware of yi camera 1080p home recently and I'm not quite good at hardware hacking. Since the firmware is encrypted, I can't extract the filesystem from the firmware. If you have the extracted version of 2.1.0.0A_201703071456home_y20m, can you share it? Thanks a lot.

shadow-1, 2017-08-05:
@skysider 
yi-hack-v3 for the Yi 1080p Home camera is already based on version 2.1.0.0A.
The official firmware file is not actually a filesystem, it is an encrypted 7zip archive. There is a script that runs automatically upon boot that extracts this 7zip archive and replaces the appropriate files on the camera (so the official firmware upgrades do not actually replace the filesystem, they simply replace files on the camera).

I have provided a simple ZIP file which contains the contents of the official firmware file here under the directory `Official Firmware --> Yi 1080p Home Camera`:
https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

What have you been working on with these cameras?

skysider, 2017-08-05:
@shadow-1 ，Thank you. I work on these cameras to find vulnerabilities behind the logic or code implementation of the functionality to improve security of such devices.

shadow-1, 2017-08-05:
@skysider 
This sounds very interesting. Please share your findings when you discover interesting. I believe it will benefit the whole community that uses these cameras.

How do you plan on examining the programs running on the camera?

I have been trying to reverse engineer certain aspects of the official Xiaomi cameras to learn how they work. Unfortunately I am having trouble cross compiling many debugging tools. I could debug in Assembly. However I am not that strong in Arm Assembly programming/hacking.

skysider, 2017-08-05:
@shadow-1  I usually use IDA pro to do static analysis  and use gdb to do dynamic debugging. The hex rays, which is one of IDA pro's plugins, can disassemble arm assembly to c, so you can understand the function easily although it's not 100% correct. 
As for debugging tools, there are already prebuilt static linked gdbserver binaries like [https://github.com/mzpqnxow/gdb-static-cross/tree/master/prebuilt-static](https://github.com/mzpqnxow/gdb-static-cross/tree/master/prebuilt-static) and you just need to compile a gdb for arm arch to connect to gdb server. Network traffic monitor also helps the analysis as well as corresponding apps. It surely takes much time.

shadow-1, 2017-08-07:
@skysider 
Thanks for the tips. I have already cross-compiled gdb and gdbserver for this camera. The SDK actually provides the source and Makefile for building gdb for this platform. I do not have much assembly experience, I tried Hex Rays which comes with IDA Pro. I never realised that it could produce pseudocode from a disassembled program. I have also been able to utilise gdbserver with IDA Pro, although for some reason I have to kill gdbserver and restart it every time I want to debug a program through IDA Pro.

The pseudocode that IDA Pro produces is much more readable for me (I have done a lot of C/C++ programming but very little assembly programming). I guess I have to be mindful of certain patterns which look complex in the pseudocode but represent something very simple and for the occasional incorrect interpretation of the assembly code.

I was hoping of making use of some higher level tools. I have successfully cross-compiled strace and this has provided me with some insight without going too deep into the coding. I was hoping to cross compile ltrace. However I am having a lot of trouble cross compiling it. In particular, I am having trouble with the libelf dependency. This would have provided me with a fair bit on insight without diving into the code.

I guess I just need to find a solid amount of time (rather than short periods of time) to understand how the Xiaomi programs work through a disassembler so I can further develop the firmware.

imtiajmeah, 2017-12-07:
The firmware looks like it's been updated again to Version:2.1.0.0B_201710091352

shadow-1, 2017-12-08:
@imtiakmeah
I should release a new version soon with the work I have done thus far.

dvv, 2018-04-14:
@shadow-1 please share the disassembly for `/home/app/mp4record`. tia