shadow-1/yi-hack-v3 #45: 403 Forbidden  File is protected couple hours after crack

wangbourne, 2017-08-08:
I'm using Yi Dome, purchased from China and now is using in Australia.
It worked perfectly after strictly following instructions. However, when back after couple hours, i could not connect  it any more
1. On Android App(Yi Home), it shows: **Offline at xxxx**
2.  On PC, trying to open it's configure url (192.168.1.xxx?), it comes following error page: **403 Forbidden
File is protected.**

Appreciate anyone could help me out.

shadow-1, 2017-08-09:
@wangbourne 
This is a very strange issue.
Does it occur all the time, only occasionally, or only once?

This is very strange because it means that the camera has crashed altogether. There are various Xiaomi programs running on the camera. One of them is a watchdog. The watchdog monitors and restarts the official Xiaomi programs if they crash. It also automatically restarts the camera on certain errors.

It is meant to stop this kind of issue.

wangbourne, 2017-08-09:
it always works for a while (less than 10 minutes) after a restart, then always offline again (i can hear some little noise from camera when it becomes offline). 
It happens same before, so I hacked it again days ago.

firmware version: 1.9.1.0H_201706221046 (latest)
Android app: download from this website (3 days ago) 

Is there any way i can help to figure out the root cause?

shadow-1, 2017-08-10:
@wangbourne
I might need your help in diagnosing this issue.
Unfortunately it is best to diagnose it through Serial connection.

Through the serial connection, you can see everything that is happening on the camera and we may see the error that is causing the camera to crash.

I have exactly the same camera as you and it works perfectly.

You can try flashing to an older version of the firmware such as 0.1.3. Just be careful not to perform the Xiaomi updates on the older versions of the firmware. If you do, you will need to reflash the custom firmware.

ikzjfr0, 2017-08-10:
thanks. The oldest one is yi-hack-v3 0.1.1, where can i find out the version of 0.1.3 that you said above?

shadow-1, 2017-08-10:
All previous versions are available here.

https://github.com/shadow-1/yi-hack-v3/releases

ikzjfr0, 2017-08-10:
it works after flashing 0.1.3 version :)
Thanks

ikzjfr0, 2017-08-11:
Actually, it failed again, it worked only a night and a morning :(

shadow-1, 2017-08-14:
@ikzjfr0 
This is very strange. I can't think of a reason why it will not work.
I know that the Yi Dome and potentially the Yi Home 17CN/27US versions have less RAM than the 1080p versions of the camera.

Maybe you can try disabling a few features like Telnet, FTP, or web server to save RAM and see if that helps.

What is strange is that I am using the Yi Dome which has les RAM and it is working fine.

Are you sure that both rootfs and home partitions are being flashed correctly?
Per haps you can try upgrading both files one at a time and check whether both files are being flashed (blinking yellow light upon boot).

The most common fix for flashing issues is to downgrade to an older version before upgrading again to the latest.

ikzjfr0, 2017-08-16:
After disabling "Telnet, FTP, and web server", it seems working now. At least it works for 2 days without any errors. ( By the way, i also removed SD card for removing any potential failure reason)

will feedback if any new findings

Thanks so much for your help.

wangbourne, 2017-08-23:
:( after couple days' happiness, it goes back to the original issue. If I reboot the camera, it works couple minutes, then fails again.

batrarobin, 2017-08-23:
@wangbourne

Are you using the power adapter that came with the camera? Sometimes the devices reboot when the adapters can't supply the required current.

ikzjfr0, 2017-08-25:
@batrarobin  you are amazing. It works again after changing to an iPhone power adapter (I used the Yi orginal adapter, rather than third party's adapter).

I will feedback couple days later if anything wrong :)

ikzjfr0, 2017-09-25:
Coming back after almost 4 weeks.
Now, it works in most cases. During past 4 weeks, it is out of connection only twice. And after power off and power on again, it works again :)

shadow-1, 2017-09-30:
@ikzjfr0 
Good to hear that changing the power adaptor fixed your issue. Well mostly at least.

closed 2017-10-11 by shadow-1