shadow-1/yi-hack-v3 #46: I have a question

n3tranger, 2017-08-08:
Hi~
thank you for your development.

I'm not fluent English.
I'm using Yi Dome camera. (china version)

1. When I connect my internal proxychains-ng  setup page → and test proxy test page
some error occured.

ERROR : ERROR (1524) : Child process 31 ended abruptly Status Code : 23827736

ERROR : ERROR (10) : Could not start external process.

2. Using yi home app(android) and then I connect my camera, sometimes can't connect
(※ test config result is good)

shadow-1, 2017-08-09:
@n3tranger 
The error message you are receiving is a known issue. It only occurs from time to time and it does not crash the web server. Refreshing the page, or moving to another page and coming back to the test page should fix the issue.

The way I am handling processes in the web server is not ideal. However it does work nearly all the time.

---

With your issue that you sometimes can not connect to the camera. This is most likely caused at least one proxy server does not work properly. There is a known problem that some HTTP proxy servers pass the tests. However they still do not work with the Xiaomi servers. In the next release I will introduce an extra test to make sure the proxy server works.

For the time being, it is best to only use SOCKS proxy servers as they are much more reliable and don't require as much testing to ensure that they work.

n3tranger, 2017-08-10:
Thank you so much.
I'm looking forward to your next release.

When it comes to next version?

shadow-1, 2017-08-10:
@n3tranger 
I'm not sure when I will release the next version. However it should not be too long. It will be a minor update. Just adding an additional test to the test pages, officially adding support for Yi Home 27US and hopefully the original Yi Home as well.

RTSP is the long term goal. However I have not gotten very far just yet.

closed 2017-08-21 by shadow-1