shadow-1/yi-hack-v3 #49: How to fix bricked camera?

galadril, 2017-08-16:
HI

Got my camera in a bootloop / bricked after installing the latest yi-hack-v3 home_y18 hack. 
any ideas?

Best Regards
Mark

shadow-1, 2017-08-16:
@galadril 
The easiest fix is to downgrade to an older version of yi-hack-v3 before upgrading again back to the latest.
This should recover a bricked camera.

galadril, 2017-08-16:
sorry... but i think i've took the wrong firmware files (omg)..
found out that my camera was the US version.. so it's not supported by your hack yet./.

shadow-1, 2017-08-16:
@galadril 
The Yi 1080p Home camera uses the home_y18 and rootfs_y18 pair of firmware files. The US and the Chinese version of the camera use exactly the same firmware.

galadril, 2017-08-16:
Camera is working again with the use of your recovery Yi Home 27US files.. thanks!

galadril, 2017-08-16:
after a reinstall also the y18 files worked.. but i was not able to connect to wifi with the app.
it's possible now with the recovery files

shadow-1, 2017-08-16:
@galadril 
Just checked, yes the Yi Home 17CN and 27US versions use home_y18 and rootfs_y18 firmware files. I have not officially released a version compatible with Yi Home 27US yet.

I have provided a pre-release compatible with the Yi Home 27US in this post:
https://github.com/shadow-1/yi-hack-v3/issues/40#issuecomment-315593035

galadril, 2017-08-16:
ill try the pre-release 👍 thanks

galadril, 2017-08-16:
Works great thanks, thanks for the assistance! 🥇 
Looking forward to future features like RSTP :-)

closed 2017-08-16 by galadril

stn4stn, 2019-06-30:
How did you install the recovery files? Nothing happens If I copy the home_y18 and rootfs_y18 in the root directory of the SD.