shadow-1/yi-hack-v3 #50: When will this support YI Home Camera 2 1080p ?

ranaviran, 2017-08-17:
This is the 2nd version of the camera. When will you support this? Thank You..

shadow-1, 2017-08-18:
@ranaviran
Unfortunately the Yi Home 2 is not supported with this firmware.
The Yi Home 2 firmware uses a completely different chipset to the original Yi Home.

The Yi Home 2 makes use of the Ambarella S2LM chip.
This firmware only supports cameras based on the HiSilicon Hi3518eV200 chipset.

The firmware linked below supports the Yi Home 2. However it looks like it has not been developed for a little while.
https://github.com/niclet/yi-hack-v2

shadow-1, 2017-08-21:
@ranaviran 
It appears that @niclet has recently updated his firmware for the Yi Home 2.
This firmware now supports using the official Xiaomi app using a proxy server (using the same technique as my firmware).
https://github.com/niclet/yi-hack-v2

niclet, 2017-08-21:
Actually, I don't modify the firmware itself. Patches are applied dynamically and are not persistent. I plan to improve it a little bit more if I find some time ;)

closed 2017-10-25 by shadow-1