shadow-1/yi-hack-v3 #51: Camera access from internet

AndrewZAP1977, 2017-08-22:
Sorry about my english.
As far as I understand, access to camera WEB interface is on 80 port! 
Is it posible to change it on diffrent another? 
My router is on 80 port too by defauld and i have conflict between router and camera! I just can`t enter into camera`s WEB interface from internet!

shadow-1, 2017-08-22:
@AndrewZAP1977 
It is possible to change the port of the web server through Telnet or FTP manually. I don't have that feature built into the web interface.

To do what you want, I don't think changing the port will make a difference. You need to use Port Forwarding feature on your router. When a request comes into your router, it has no idea that it should forward that request to your camera. What Port Forwarding does is provide a rule to the router, if a request come into your router on a particular port, it should forward that request to a particular IP address on your network on a particular port.

---

Below I have a screenshot from Tomato firmware. Almost all routers have a similar interface.
On this type of interface, this is the rule you need to setup:
* Protocol: TCP
* Source Address: **blank** (Some routers require you to enter 0.0.0.0 or the setting doesn't exist)
* External Port: **Port you want to use to connect to the camera from the Internet** (Example: 8080)
* Internal Port: 80
* Internal Address: **IP Address of the Camera** (Example: 192.168.1.100)

[image]

AndrewZAP1977, 2017-08-22:
Thanks, **shadow-1**.
I've changed my router default port (on 999 for example), made port forwarding on my router for camera and it's work well. 
My previous camera (IPRobot2) was working through port 81, and it was enough for me to forward this port for the camera IP address, and now I have to change the settings of the router default port.
Thanks for answer!

AndrewZAP1977, 2017-08-22:
Could you explain in more detail how to change the port of the web server through Telnet or FTP manually?
Thanks.

AndrewZAP1977, 2017-08-22:
Is that enough if i just change port number in camera's "etc/services"?

shadow-1, 2017-08-23:
@AndrewZAP1977 
To change the web server port manually, you will need to modify the following configuration file on the camera: `/home/yi-hack-v3/etc/lwsws/conf.d/yi-hack-v3`

You will need to modify the following line:
```
"port": "80",
```

---

Using FTP, you can download the file to your local computer, edit the configuration file and then replace the configuration file.

Using Telnet, you can use a command line text editor such as `vi` to edit the configuration file directly on the camera.

You will need to reboot the camera for the changes to take effect.

AndrewZAP1977, 2017-08-23:
Thanks, **shadow-1**!
But i don't have /home/yi-hack-v3/etc/lwsws/conf.d/yi-hack-v3 in my camera root derictory!

My camera ROOT directory:
[image]

Or I do not understand you correctly?

AndrewZAP1977, 2017-08-23:
I've found this path **/home/yi-hack-v3/etc/lwsws/conf.d/yi-hack-v3** in your Source code (yi-hack-v3-0.1.4-beta2.zip) file! But what can i do with it?

shadow-1, 2017-08-23:
@AndrewZAP1977 
Based on the screenshots, you are not using the latest versions of yi-hack-v3. You must have version 0.1.3 or older installed at the moment.

In these older versions of the firmware I used a different web server. The documentation for httpd is a bit confusing. However it looks like the following line would need to be added to the end of `/home/app/httpd.conf`:

```
P:/:*:8080/
```
This should change the web server port to 8080.
However you can check the documentation available here:
http://wiki.chumby.com/index.php?title=Using_the_busybox_HTTP_server

AndrewZAP1977, 2017-08-23:
Yep! Everything is fine!

Thanks, **shadow-1**

closed 2017-10-25 by shadow-1