shadow-1/yi-hack-v3 #54: File transferred by ftp client (FileZilla) changed md5

Leelour, 2017-08-28:
I tried to upload a file to the camera via FileZilla, however, it seems the file md5 is changed after I upload. I suspect there may be something wrong with the ftp server running on the camera.

shadow-1, 2017-08-28:
@Leelour 
The camera has very very limited free space. It is very easy to exceed the free space. In this case, the transferred file will be truncated.

I recommend transferring files to the microSD card which is located in the directory: `/tmp/sd`

Leelour, 2017-08-31:
Is it even the case for very small file (less than 1KB)?

shadow-1, 2017-08-31:
@Leelour 
I have checked MD5 hash and everything appears to work correctly for me.
I used HashTab on Windows to get the MD5 hash of the file, transferred the file using FileZilla and ran `md5sum` on the camera and the MD5 hash matches.

How are you calculating the MD5 hash? As far as I know, there are multiple implementations.
Which operating system are you using?

Leelour, 2017-08-31:
I use 'md5' on mac and 'md5sum' on the camera. I can get the right md5 sum on camera when I host http server on my mac and download the exact same file using wget in camera. I will try to reproduce the problem again. I'll keep you updated.

shadow-1, 2017-09-01:
@Leelour
Can you post the command you use on your Mac?

Leelour, 2017-09-01:
Update: this seems caused by FileZilla. Here are more details including the commands you ask.
Here are the commands I use on the Mac:
```bash
$cp enc_key enc_key_1
$md5 enc_key
MD5 (enc_key) = 5d382a1154213e060768652473ec837f
$md5 enc_key_1 
MD5 (enc_key_1) = 5d382a1154213e060768652473ec837f
$cp enc_key_1 $HTTPSERVER/enc_key_2
$base64 enc_key_1 
jVa4zkbLgFVfKUyBqUDZds7AswE7s4Z+yejvBW7pH7zb2JB+nvVUg8/SywpFRLLQkgjlRTiscuOWFDpbY/NsBS41C+Qxf6bED/ipkSEc2u53DiC7+iVJ5wQcnSSIZht6RyfGewwz0MR4iBomwKSLK4LVAktAwqfkdQYCc7JUWI22gqDGCGEnF1MXqKATBa/I7xXh+B3Fono2ESIZai4BW1ehx6396/l7igxgqRd/TKYMNWCqrHBKJcfarrO/xMl6PHV0PfbQgoO0K2zTgJ9D6ciVDJuCwgKKAggAhXdU+MW7agQBLlQqtVNKP9HTtOp3PF1kHkbe0QR+A+wfDHcGKzpvxrie0Pks8QEGKn7gVPPkkFWJyCrNDs9WkM2jLcDm2llB2nhfs+VVWvu7ftcMNqZjthBmd4SLGqcswa/RAJ9o/YgaO/879QlJIDKiT9uytGv4hrb6MSM97jPwUZFNEnBaN/XRNrDivAmSrEsKordEwaEtrOi1j9JroESDEG4gbO1q6dltkVWKrXolwELVUL4s/0v+IkU0xNKNY+iHDS+IAZxtrrctE1cHIX2BFVX80easZEL/9qAYbfyR2BEkaHcmUbyNrr6IPjkusY7y7jjDNtgNPYhAz7evI29puNKuavGlOQ4AyDDBpR8YpCcNOfEhbHes3C9pkkYuLShQ93c5/xfypKwpayHfzlMh2XX1q70gr8+DfYBsUIKU9+awjzTrN/hMyc3ba73OrkVxyisMB510y0sapDH8fWoQLGMuXS41eGeewqSqQgReTvF30ysTPG3zGHoxlJ6X9Cbzp55qQ6jcLCJellfFUhCnytLiQvrDqeDTHWffK8aBdr9DN0RO5Kxck45jRYhtD5o+5JXlZSkXz25+LktM06dAfzbagH6v8khPrGnLCsrRyIjVmqLqavnUta1NixhiWw6AZcMPGpWk6ACAQXNepp8f0wcgOU+eq130XD4vJ6QBQQlqtwXpxTpyXSBACiabidQR/WKrSS7E7JI44Jfs3uJoMB/INRdYdvPmSmv8v8KhkfWf+auYK5lkPnUikBDnzJseg+T0huDYEVS+hxSRdzJQf417AUM0/i+jkOWuzeDtfDrlECCxVgBJJb0q6oGIpskFSdxZyloOBwgnZ4Pw64uusmnBQYoFZJq4Ml20trIIPCCXeeU5pOBcDgDQdl4dl1zazbe7wuchsSqYDn/MRaedxeI8Uz+uHDfsNg4wRtjHGBggan+ruyI6zRcpIIoYf4Q8ok0x+7xdPQL+4Ep79qR2eny5Y/C7KFtnG4WvtYBWud3VWXnWE4PB6bfz/R2NFeMg0Df+/puMVDKp3TR/0/o5QtMfhxb4kjPlY92Qi06eXvB/VHh8u3GsyIXkPgX+f76l1tGwSN8WXhNTzoZiyBAeZoifb4gLAIY/7u+9Di0ZxSR52HkK7Cya/9jAlVi1clfoQGEKkpHTOUrS4uLjUtmb162KggK0CAo0zmHLL4phvAAtZw63uBV1cFzDgKCDOqxuCF/Attk7608j4fsdnl7jSGNLUoM0fJlfTjVrsRCwsR3AZ+bp7HVQIeyPJGkIl3QJgrYS9ts+k/var9ki1Y+DABG7D3iHAYHLH+ZhCwDtAo3M4wUr7jJVM+hd3DckP9ybdxeiLLdemHfM27XTUQSqNo8Kv+Lk7FMyDwJE4lW4ys6V91lPR1s68EXQIMNo39Ty2Pt6U6EAAQrpkf6h0bQ42oQny6tduzrzREduZQmBQaaNhBvrcopKgL/FJ/unBBNB+uo3/qz2Yg6Va6VYQt3G3u7K
```
After uploading 'enc_key_1' to the camera /home directory via FileZilla, I ran following commands:
```bash
~ # cd /home/
/home # md5sum enc_key_1 
dd07bb1fb7cdbf0d957600440ee236db  enc_key_1
/home # wget http://{server-ip}/enc_key_2
/home # md5sum enc_key_2 
5d382a1154213e060768652473ec837f  enc_key_2
/home # base64 enc_key_1 
jVa4zkbLgFVfKUyBqUDZds7AswE7s4Z+yejvBW7pH7zb2JB+nvVUg8/Syw0KRUSy0JII5UU4rHLj
lhQ6W2PzbAUuNQvkMX+mxA/4qZEhHNrudw4gu/olSecEHJ0kiGYbekcnxnsMM9DEeIgaJsCkiyuC
1QJLQMKn5HUGAnOyVFiNtoKgxghhJxdTF6igEwWvyO8V4fgdxaJ6NhEiGWouAVtXocet/ev5e4oM
YKkXf0ymDDVgqqxwSiXH2q6zv8TJejx1dD320IKDtCts04CfQ+nIlQybgsICigIIAIV3VPjFu2oE
AS5UKrVTSj/R07TqdzxdZB5G3tEEfgPsHwx3Bis6b8a4ntD5LPEBBip+4FTz5JBVicgqzQ7PVpDN
oy3A5tpZQdp4X7PlVVr7u37XDDamY7YQZneEixqnLMGv0QCfaP2IGjv/O/UJSSAyok/bsrRr+Ia2
+jEjPe4z8FGRTRJwWjf10Taw4rwJkqxLDQqit0TBoS2s6LWP0mugRIMQbiBs7Wrp2W2RVYqteiXA
QtVQviz/S/4iRTTE0o1j6IcNL4gBnG2uty0TVwchfYEVVfzR5qxkQv/2oBht/JHYESRodyZRvI2u
vog+OS6xjvLuOMM22A09iEDPt68jb2m40q5q8aU5DgDIMMGlHxikJw058SFsd6zcL2mSRi4tKFD3
dzn/F/KkrClrId/OUyHZdfWrvSCvz4N9gGxQgpT35rCPNOs3+EzJzdtrvc6uRXHKKwwHnXTLSxqk
Mfx9ahAsYy5dLjV4Z57CpKpCBF5O8XfTKxM8bfMYejGUnpf0JvOnnmpDqNwsIl6WV8VSEKfK0uJC
+sOp4NMdZ98rxoF2v0M3RE7krFyTjmNFiG0Pmj7kleVlKRfPbn4uS0zTp0B/NtqAfq/ySE+sacsN
CsrRyIjVmqLqavnUta1NixhiWw6AZcMPGpWk6ACAQXNepp8f0wcgOU+eq130XD4vJ6QBQQlqtwXp
xTpyXSBADQomm4nUEf1iq0kuxOySOOCX7N7iaDAfyDUXWHbz5kpr/L/CoZH1n/mrmCuZZD51IpAQ
58ybHoPk9Ibg2BFUvocUkXcyUH+NewFDNP4vo5Dlrs3g7Xw65RAgsVYASSW9KuqBiKbJBUncWcpa
DgcIJ2eD8OuLrrJpwUGKBWSauDJdtLayCDwgl3nlOaTgXA4A0HZeHZdc2s23u8LnIbEqmA5/zEWn
ncXiPFM/rhw37DYOMEbYxxgYIGp/q7siOs0XKSCKGH+EPKJNMfu8XT0C/uBKe/akdnp8uWPwuyhb
ZxuFr7WAVrnd1Vl51hODwem38/0djRXjINA3/v6bjFQyqd00f9P6OULTH4cW+JIz5WPdkItOnl7w
f1R4fLtxrMiF5D4F/n++pdbRsEjfFl4TU86GYsgQHmaIn2+ICwCGP+7vvQ4tGcUkedh5DQrsLJr/
2MCVWLVyV+hAYQ0KkpHTOUrS4uLjUtmb162KggK0CA0KNM5hyy+KYbwALWcOt7gVdXBcw4Cggzqs
bghfwLbZO+tPI+H7HZ5e40hjS1KDNHyZX041a7EQsLEdwGfm6ex1UCHsjyRpCJd0CYK2EvbbPpP7
2q/ZItWPgwARuw94hwGByx/mYQsA7QKNzOMFK+4yVTPoXdw3JD/cm3cXoiy3Xph3zNu101EEqjaP
DQq/4uTsUzIPAkTiVbjKzpX3WU9HWzrwRdAgw2jf1PLY+3pToQABDQrpkf6h0bQ42oQny6tduzrz
REduZQmBQaaNhBvrcopKgL/FJ/unBBNB+uo3/qz2Yg6Va6VYQt3G3u7K
```
By comparing the base64 output, it become different on 58th character.
However, after using 'Transmit' (another FTP client), it seems md5 sum doesn't change. This means the problem is caused by FileZilla (3.26.1). And this problem persists on FileZilla's latest version 3.27.1.

shadow-1, 2017-10-11:
@Leelour 
Glad to hear that we got to the bottom of the issue.

closed 2017-10-11 by shadow-1

CrescendoFang, 2018-01-01:
set the file trf type in FileZilla to binary.... this was caused by ASCII trf type...