shadow-1/yi-hack-v3 #55: "This Camera Can Only Be Used Within China" even on running

genz10, 2017-09-02:
Hi,

Why the camera suddenly sounds "This camera can only be used within china" without any reason ? The camera is running and no connection issue. As describe before, I think the camera will only triggered region checking after inactivity for couple days.

shadow-1, 2017-09-07:
@genz10 
There is no reason why this should occur. The issue is most likely due to the proxy server that you are using.

The proxy server that you are using may not actually be located in Mainland China, it may be using some obfuscating technique to make it seem like it is coming from Mainland China whilst not actually being there.

I have not heard of this issue from other users.

genz10, 2017-09-12:
@shadow-1 

Sorry, seems there's trouble on my proxy server. Is there any plan launching the latest hack version ?