shadow-1/yi-hack-v3 #58: Yi dome camera, chinese version used in Europe

lionghan, 2017-09-17:
I have chinese version Yi Dome, and I am trying to use this in the Netherlands. I already installed the hack according to your instruction, but it stucks at step 9. Camera can not find the Wifi connection, blinking yellow light continuously.
Please help. Thx

shadow-1, 2017-09-17:
@lionghan
Are you sure you entered your password correctly? Does the blue light blink at all when you pair the camera to your phone in the app?

Remember that once Wi-Fi connection is established during the pairing process. You must setup the proxy servers on the camera. Then you can repair the camera and it will be successful and the Wi-Fi password will be remembered.

lionghan, 2017-09-17:
Yes the password is correct.
The blue light does not blink at all.

batrarobin, 2017-09-18:
@lionghan

I had a similar problem with one of the other cameras. You could probably try the following.. :

Turn off your home router
Borrow a spare Android phone with WiFi Hotspot
Set the hot-spot name exactly same as your home router SSID and also the same password.
Do the whole process as if you were connected to your home router.
If this works, then turn back on your home router and disconnect hot-spot.

lionghan, 2017-09-18:
@batrarobin 
I have just done what you described above, but the blue light does not blink at all.

batrarobin, 2017-09-18:
@lionghan

Are you sure you are scanning your phone screen correctly in front of the camera lens. You have to scan your phone screen when the camera says "Companion-cheeye". You have to keep it steady and at the same time move it far to closer while scanning. Make sure you phone screen is medium brightness.

lionghan, 2017-09-19:
Yes I am scanning the phone correcly and the screen has set to medium brightness.

Verzonden vanaf Samsung-tablet.

  Europe (#58)

shadow-1, 2017-09-20:
@lionghan
This is strange. As soon as the camera scans the QR code, the status light changes to blinking blue light to show that it is trying to connect to the Wi-Fi network.

As a last resort, you can try flashing an old version of the firmware before upgrading again to the latest.

Newbiesc, 2017-09-20:
1. Try to see your wifi ssid and password (make sure there is no space or other than letter and number in it), some times it makes trouble.
2. Make sure the wifi is not Microtik system.
if you still got trouble with it, try to make a video on pairing it and upload it to youtube, so we know the problem more specific.

shadow-1, can i have a private message: newbiesc.git@gmail.com

lionghan, 2017-09-24:
@Newbiesc: SSID and password only contain numbers and letters. No, I don't use Mikrotik.
@shadow-1: I have done the following steps:
1. Copied the firmware 1.9.1.0H_201708091518home_v200m (name changed to home_v200m) to my 16GB microSD card (FAT32 formatted).
2. Removed power from the camera, inserted the microSD card, turned the power back ON
3. The yellow light came ON and flashed for approx 45 seconds.
4. Chinese Yi app started and I tried to pair the camera a few times, but it did not read the QR. Camera says waiting to connect the whole time.
5. Then removed the power and the SD card and copied your two files (rootfs_v201 home_v201) to this micro SD card (I already tried using firmware yi-hack-v3 0.1.3 & Beta 1 and Beta 2).
6. Inserted the SD card to the camera and turned the power back ON
7. Again the yellow light came ON and flashed for approx 45 seconds.
8. Chinese Yi app started and I tried to pair the camera a few times, but it did not read the QR. Camera says waiting to connect the whole time. Blue light does not blink.

ykhandler, 2017-09-24:
Contact my email. I think i can help you...