shadow-1/yi-hack-v3 #59: security concerns

ervinch, 2017-09-20:
after crack the firmware, i can get the http access without any authentication. Will it be some of loop hole to hacker to hack my camera to be part of the DDOS attack zombie?

And as we need to use proxy server to connect back to xiaomi server, will be proxy be the middle man to hijack all my video images?

Thanks.

ver2go, 2017-09-20:
You're accessing the http server on a private ip network. If so those address do not route publicly.

Proxy to connect out to appear to Xiaomi servers as if yore in China.  Not reverse proxy to you at your edge.

shadow-1, 2017-09-21:
@ver2go 
You are absolutely correct.

@ervinch 
In addition to this, I can confirm that only some of the communication between the camera and the Xiaomi server goes through the proxy server. None of the video/audio data go through the proxy server.

ervinch, 2017-09-24:
great, thanks a lot for addressing my security concerns.

markusd1984, 2017-10-30:
While the local address doesn't route publicly without the router modified to do so and the video/audio data neither going through the proxy, could/will our routers external IP address be exposed that way?

If so then a hacker would first have to target our router to get to the camera right?

shadow-1, 2017-10-31:
@markusd1984 
The routers external IP address is exposed whenever a connection is established with any other server on the Internet. So it is theoretically possible for a hacker to find out your external/public IP address.

Those with ISPs that operate under a CGNAT are indirectly offered another layer of protection. A hacker will only theoretically have access to the ISPs external/public IP address and not the IP address of your router.

xvolte, 2017-11-13:
Hello,
how does the yi app open the camera feed ?
I did not open anything on my router, so i assume it asks for the yi cloud service to contact the camera and open a video stream ? if so, it means that upnp is required for the camera, and that it opens a port for "anyone ?" to see the feed ? as there is no authentication, how can we ensure we are the only ones to watch ?
Did you get a chance to see what traffic goes to yi cloud ?
thanks in advance

shadow-1, 2017-11-17:
@xvolte 
I'm not sure whether upnp is required by the camera or whether it uses a fixed port.

The camera makes use of a cloud service to work without the end user opening up anything on their router. It also works with CGNAT without issues because of this. The camera connects to the Xiaomi server and to the cloud service. The app then connects to the Xiaomi server. Once logged in, the app is able to connect to your camera through the cloud service connection.

All communication between the Xiaomi server and the cloud service is encrypted. The camera also authenticates with the Xiaomi server. The serial number and a hardware key embedded in the camera forms part of the authentication process along with a generated key (not sure how this authentication key is generated). If all three keys match what the Xiaomi server expects, the camera is authenticated to the Xiaomi server and cloud service.

I did get a chance to see what traffic gets sent to the Xiaomi server. However I have not been able to examine what is being sent to the cloud service.