shadow-1/yi-hack-v3 #60: Solution for the "new European" Yi Home 720p with 1.8.7.0A_201702081101?

davideyo, 2017-09-24:
Dear shadow-1, great work on the Yi cams!

If you can, keep up the good work!
:)

---------
I wonder whether you are planning any time soon  
to also support the "Yi Home 17 Int" Camera (720p)?

I have just bought the 720p model in Europe from amazon.de, 
(https://www.amazon.de/dp/B016F3M7OM/ref=pe_3044161_189395811_TE_dp_1)
and it is (probably) the Yi Home 17 International model, as it
uses the Intl Yi app and has the 1.8.7.0A_201702081101 firmware.

BTW: I read in https://github.com/fritz-smh/yi-hack/issues/174
(at the end of the thread) that "uraharakisuke12" 
successfully hacked his non-Chinese Yi Home 720 p model
with your solution fro the 17CN, quote:

" uraharakisuke12 commented 25 days ago • edited
@lastb0isct 27US? shadow-1/yi-hack-v3#40 (comment) this work for my cam
"
-------
Would this also work for my (European) 1.8.7.0A_201702081101 firmware?

If I try, do I risk bricking, or can I always downgrade using
your international Yi Home images?
(https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr/folder/32510114033)

Are you planning in the near future an "official"
solution for the European/International Yi Home 720p with 1.8.7.0A_201702081101?

These are now being sold in Europe (with the small white with red letter sticker on 
one of the backside screws).

Kind regards,
Dave

shadow-1, 2017-10-02:
@davideyo
yi-hack-v3 support for the regular Yi Home is coming soon.

At the moment, I have created versions compatible with Yi Home 17CN and Yi Home 27US as they are almost identical to the Yi 1080p Home and Yi Dome cameras which were already supported.

davideyo, 2017-10-04:
Great, shadow-1! 
Thanks for the good news regarding the the regular Yi Home (720p) model!

Will be watching this space :)
Dave

Berkinka, 2017-10-08:
Hi @shadow-1,

I have bought two Yi Home Camera 720p version. 

According to readme page, my camera is not supported yet. 

But according to your message in here, my camera is supported. I looked up to ID's of my cameras and they begin with 27US. 

Can you give further info about which file you provided should i use ?

Thank you,
Berkin

shadow-1, 2017-10-09:
@Berkinka 
I have created a pre-release version of the firmware which is compatible with Yi Home 27US.
It is available from this post: https://github.com/shadow-1/yi-hack-v3/issues/40#issuecomment-315593035

thewingit, 2017-12-13:
Hi,
Thank you very much for your engagement. 

I have recently bought an YI Home 720p, on the back of the camera under the QR-Code it says "27USY7..." and below it says "Model: YHS-113-IR".

The firmware that came with the cam was the "1.8.7.0A_201702081101", but unfortunately I upgraded with the smartphone app to version "1.8.7.0D_201708091510", before I got aware of this hack.

I'm not sure whether I bought the Yi Home 27US, because the app upgraded the firmware to the 1.8.7.0D, which according to your  should belong to the "newer" YI Home, for which support is still not granted.

Do you think that flashing the 1.8.7.0A on this cam (that was the firmware that was installed when I bought it) would still work?

Thanks again,
Michele

shadow-1, 2017-12-13:
@thewingit 
It looks like you have the 27US camera.

There are no issues using the pre-release firmware linked above.
Flashing the pre-release firmware will downgrade the firmware to an older version. However it is perfectly safe to upgrade it back to the latest 1.8.7.0D_201708091510.

fueledbyblueeyedblondes, 2018-02-11:
Hi,

First of all, thank you very much for you work. 

I am getting confused about all the different versions.

I have a Yi Home 720p that says Model: **YHS-113-IR** and the serial number is: **22EU**TX...161207. I am looking to integrate it into Home Assistant (https://home-assistant.io/components/camera.yi/).
Can I use the files for 27US / 47US perhaps? My firmware is called: "**1.8.7.0A_201702081101**".

Many thanks.

morphirr, 2019-01-18:
Hi thanks for all your hard work, totally new to this and I was able to apply the hack to the camera **47US** bought from Amazon uk with Firmware version **1.8.7.0F_201809191400** which it had upgraded to when I first connected the camera.
The Hack downgraded it to **1.8.7.0D_201708091510** even though I'm not exactly certain what it now allows me to do, I don't like the cloud service will there be a way to record locally, its still there thought this hack removed it and that the camera calls home and hope this can be stopped at some stage did I read correctly that now I can upgrade back to the latest Firmware or am I getting mixed up already??? Is there a way to change user and set passwords for FTP etc, are there any tutorials showing what I can now do with this hack, Any info much appreciated and keep up the good work thanks

racelife, 2019-02-13:
Where does one get started? I'm seeing lots of different instructions I have a yi model yhs-113-ir, 22us with Version=1.8.7.0D_201708091510 according to the log file on my sd card. I haven't used my camera since I changed my wifi access point last July so I don't want to power it up and have it update fw until I know I can fix it. All I really care about is having rtsp for my nvr to record. If someone could point me in the right direction that would be great! 
Thanks,
Race

shadow-1, 2019-02-20:
@racelife 
Not sure about RTSP support for the 22US version of the camera. This is the original 720p camera where the Fritz firmware should work to provide RTSP support.

No need to worry about firmware updates. I have hosted all the previous versions so you can downgrade at any time from this link; https://app.box.com/s/cibs7n1mgvhqaqjlidtveegu1uajt5yr

racelife, 2019-02-26:
@shadow-1 
How do I downgrade? Which firmware should I downgrade to? What Fritz firmware should I use and how do I flash it? Sorry to be a pest I've just been having a hard time locating this info on my own.

moistgun, 2019-03-25:
@shadow-1 is that statement saying us22 is supported because releases don't mention that model?

Teravatio, 2021-03-02:
ar Traductor de Bing
Hello I have installed the Yi Home app from Play Store for my cameras, I have the following problem: one of them is not installed says that it is not on the correct server is a Xiaomi YI YHS-113-IR 22US5BT601. You could tell me what firmware I can install to be recognized by YI HOME from the Play Store in Europe.Thank you

graham30rad, 2021-04-03:
@shadow-1 great job and thanks for sharing you're work! I have a YHS-113-IR 720P running 1.8.7.0f_201809191400 firmware, what is my best option when it comes to unlocking it? And if it doesn't work out, is the original firmware available to put it back to stock? Thanks.