shadow-1/yi-hack-v3 #62: 28US new 1080p model

ver2go, 2017-09-30:
My first time using this firmware was on a Yi 1080 with 18US on the QR code.
https://www.amazon.com/gp/product/B01CW4AR9K

Just purchased another one but this firmware does not seem to be flashing on boot.  Confirmed by looking at DHCP clients and going to http://<address> .  Not listening on 80

This one has 28US on the QR.  

Any instructions on what to capture?  I have TTL to RS232 adapters.  Can solder decently, but would like to be less intrusive if possible.

shadow-1, 2017-10-02:
@ver2go
I am not aware of the 28US version of the Yi 1080p Home camera. As I do not have this camera, I will need someone to take a backup of all the partitions on the camera and it would also be very helpful to get the serial port output during bootup.

The difference in the firmware could be similar to the differences between the regular Yi Home 17CN and the Yi Home 27US versions.

If you are willing to solder a serial port connection to your camera. I can post what commands to run to take a backup of the partitions. The backup will have to be written to a microSD card as the camera does not have enough free space and it makes it easy accessing the backup files on a PC.

ver2go, 2017-10-02:
It looks to use the same official firmware as the 18US, but this is just my assumptions based on the version it came with and the version it wants to update to.

I have to travel this week, but will work on it this weekend. Please provide the commands, I guess just dd the internal partitions to the sd? I'll need to the block devices and if you want to skip any of the leading blocks.

Is the console just TX/RX and GND?  

I guess they didn't put a tty on the USB?

shadow-1, 2017-10-02:
@ver2go
Yes you are correct. We just need to dd the internal partitions to the microSD card.
Assuming the partition structure is similar (like all the other cameras based on Hi3518eV200 chipset), these are the commands:
```
dd if=/dev/mtd0 of=/tmp/sd/mtd0
dd if=/dev/mtd1 of=/tmp/sd/mtd1
dd if=/dev/mtd2 of=/tmp/sd/mtd2
dd if=/dev/mtd3 of=/tmp/sd/mtd3
dd if=/dev/mtd4 of=/tmp/sd/mtd4
dd if=/dev/mtd5 of=/tmp/sd/mtd5
dd if=/dev/mtd6 of=/tmp/sd/mtd6
dd if=/dev/mtd7 of=/tmp/sd/mtd7
```

The console is just Rx and Tx. You do not need to connect GND or VCC (+5V).

No they did not put tty on the USB. On the other cameras I have seen, there are pads available to solder a fly lead which you can connect to a USB to TTL converter. You can then access the console through the serial connection.

ver2go, 2017-10-06:
@shadow-1 I'm not getting a login prompt and can't' seem to break out of auto boot.
Tried pressing any key and Control C

https://pastebin.com/0XikiAMH

shadow-1, 2017-10-06:
@ver2go 
To get to the login screen, you just need to press Enter a few times. What is very annoying is that all the Xiaomi programs which are running in the background write to the console. So it is difficult to actually see the commands that you will be typing.

There is no easy way to break auto boot and go to the u-boot command prompt. The time delay to break auto boot has been set to 0 seconds. Notice in your log this:
```
Hit any key to stop autoboot:  0
```

There are ways to get to the u-boot command prompt if you wish to experiment with u-boot. However it is not required for this.

ver2go, 2017-10-07:
@shadow-1 
There may not be shell on the console.  I've keep hitting enter after boot up. Tried a different TTL converter (one FTDI and one SiLabs).  I even tried running both with one on the solder points and the other on the JST (it was labeled with TX/RX/VCC/GND) but nothing on that port.

As far as I can tell, my wiring up to the camera is good.  TX to RX on the camera should be good.

I just keep getting CONSOLE logged to my tty.  I can see local echo when I type.  I also keep typing "ls" just to make sure

Anything else I can try?

shadow-1, 2017-10-07:
@ver2go 
This is a little strange. From what you describe, the Tx connection does not seem like it is connected properly. However you did say that you see the local echo when you type.

Unfortunately `ls` command won't show anything by default. The default directory is `/root` which is empty by default. It does contain at least one hidden file though.

Upgrading the firmware should not affect the serial connection at all. The firmware updates simply replaces some files in `/home`

---

From what I understand, the JST connector does not work for everyone. Most videos I see on Youtube utilise the solder pads just behind the connector.

---

My current setup is connecting Rx and Tx to a FTDI based USB-TTL Serial converter (I actually use an old Arduino).

The main tip that I have is to make sure the camera is powered up through the USB port of the same computer that you are connecting the serial converter to. It does not work properly for me if the camera is powered externally. Perhaps this would not be an issue if I connected the GND connection.

ver2go, 2017-10-09:
@shadow-1 You can close this issue.  
I'm not sure what finally got console and the hack working.  I did too many changes at once.  I believe it was after I allowed it to upgrade that it started working.

Even after upgrade to 2.1.0.A it wanted to upgrade to 2.1.0.B.  So maybe there is "weird" version that comes out of the box on this model.

closed 2017-10-10 by ver2go