shadow-1/yi-hack-v3 #63: Add MLSD command to FTP server

bachya, 2017-10-07:
Hi there – currently, the FTP server only supports the `LIST` command for directory listings. Although this is fine, it’s a difficult format for some libraries to parse. The `MLSD` command, on the other hand, provides a standardized, easily-dealt-with response that would help library development.

Info on the command: [https://tools.ietf.org/html/rfc3659#page-23](https://tools.ietf.org/html/rfc3659#page-23)

shadow-1, 2017-10-07:
@bachya 
The FTP Server running on the camera is just the BusyBox implementation of ftpd. From a quick Google search, it appears to support `MLSD` command.

I will have look into it and see if anything needs to be done to enable support.

bachya, 2017-10-07:
I tried this morning: `MLSD` throws an exception using Python’s `ftplib`, but `LIST` does not.

Appreciate you taking a look!

shadow-1, 2017-11-10:
@bachya
I finally got around to checking this. It appears that the ftp server ftpd included as part of BusyBox does not support the `MLSD` command.

I may be able to include an alternative ftp server as long as the footprint is very small and is lean. These cameras have very limited free space and memory.

Do you have any suggestions?

bachya, 2017-11-10:
@shadow-1 – understood. What about [Pure-FTPd](https://www.pureftpd.org)? Is that small enough?

shadow-1, 2017-11-11:
@bachya 
I cross-compiled Pure-FTPd. With the default options, it comes out to be roughly 348kb which is on the large side.

There is a minimal compilation flag --with-minimal which makes the filesize much better at 90kb.
However this does not include additional FTP commands like MLSD which is what you are after.

I have tried the default configuration whilst disabling a few features. However I only saved 4kb (344kb).

bachya, 2017-11-11:
Got it. Isn’t worth it in the end, then. Thanks for considering!

shadow-1, 2017-12-19:
@bachya 
I have been looking into providing SSH support for the firmware. I was wondering whether SFTP support would help for your development?

bachya, 2017-12-19:
@shadow-1 Thanks for reaching out. `MLSD` has to do with structuring the output of listings, so SFTP won't do much for me there. That said, I think it's a great addition nonetheless!

shadow-1, 2017-12-20:
@bachya 
I understand that `MLSD` is specific to the FTP standard. I guess SFTP has a bit of a confusing name as it is not at all like FTP.

My understanding is that SFTP is an extension to SSH protocol and utilises SCP to transfer files. Therefore the specific FTP commands do not have much meaning in this context.

I guess what I was wondering is whether utilising SFTP alleviate some of the issues you are facing. You would not be bound to the FTP `LIST` command. However perhaps SFTP equivalent has he same issue (I am not particularly familiar with either FTP nor SFTP).

bachya, 2017-12-20:
@shadow-1 I don't believe so. I think the underlying FTP architecture – commands, etc. – remains the same whether the traffic occurs over SSL or not. That said, I've had people ask if my "[thing](https://home-assistant.io/components/camera.yi/)" can support SFTP, so I still think it's a good idea.

shadow-1, 2017-12-24:
@bachya 
I have managed to significantly reduce the filesize of Pure-FTPd. I have just published a new release og yi-hack-v3 which includes Pure-FTPd. Unfortunately I could not include TLS support as it would take up far too much space on the camera. The OpenSSL library which is dependency for TLS support would have to reside on the microSD card.

I could not get SFTP working for some reason. I never received any error messages, however it just did not work. So I left that out for the time being. However SCP now works, so you can try using this for secure file transfers.

bachya, 2017-12-28:
Great, thank you!

shadow-1, 2017-12-30:
@bachya 
No problems.

closed 2017-12-30 by shadow-1