shadow-1/yi-hack-v3 #64: QR code invalid

blueyoghurt, 2017-10-07:
I'm on Yi 1080p Home Camera 18CNY. I've followed all the steps and it seems to work but I am stuck at step 18. Everything was in English until step 18 when it connects through a CN proxy.

The camera tells me (in Chinese) the WiFi connection is successful and it starts the synchronization on Yi Home app. But before it completes the synchronization, the camera says "QR code is invalid. Please reboot camera" (in Chinese) and remains on blue light. I would then power down the camera and when I turn it back on, it'll be an orange blinking light waiting for a connection.

Does anyone else face this issue? Am I doing any step wrongly?

Any advice is greatly appreciated! Thank you!

blueyoghurt, 2017-10-07:
Here's what i got from Test Config page

[image]

blueyoghurt, 2017-10-07:
Hi, I read through the instructions again in detail and found the cause. I was using the **Mi Home App** instead of the **Yi Home App** for my previous cameras.

After downloading and connecting via the Yi Home App, its working fine now.

Can I also ask if there's any plan for it to be compatiable with the Mi Home App? It's has all the Xiaomi devices under one roof hence I thought it was the preferred app.

shadow-1, 2017-10-07:
@blueyoghurt 
Glad to hear that you worked it out.

These mobile apps are developed by Xiaomi. I do not have any input into them.

I have heard the Mi Home app is quite convenient. However it is designed for the Mijia range of devices. I believe a plugin was created for the original Yi Home camera and it works with the Mi Home app. However Xiaomi have not developed plugins for the Yi 1080p Home and Yi Dome cameras at this stage.

closed 2017-10-11 by shadow-1